sparkfun/SparkFun_RTK_Firmware #317: NTRIP client can't connect to caster

pyrog, 2022-08-31:
Same as #285 ?

### Your workbench
* [RTK_Surveyor_Firmware_v2_4.bin](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/download/v2.4/RTK_Surveyor_Firmware_v2_4.bin)
* WIFI AP and USB connection to check device status.

### Steps to reproduce
Start device. NCY is the selected profile (`caster.centipede.fr` on port `2101`. Mountpoint `NCY`)

On the AP of the smartphone I see 3 or 4 quick connections.

```
Profile ‘NCY’ loaded

MAX17048 configuration complete
Batt (93%): Voltage: 4.13V Discharging: -1.04%/hr Green

GNSS configuration complete
System time set to: Wednesday, August 31 2022 19:12:55
Bluetooth broadcasting as: Surveyor Rover-0CE2
*NTRIP_CLIENT_OFF
NTRIP Client start
NTRIP_CLIENT_ON
State: Rover - No Fix, 2022-08-31 19:12:58.371
Batt (93%): Voltage: 4.13V Discharging: -1.04%/hr Green

Log file created: SFE_Surveyor_220831_191258.ubx

UBX file size: 102 - Generation rate: 0.0kB/s - Write speed: 24.8kB/s
WIFI_NOTCONNECTED
WiFi connecting to Pyrog
NTRIP_CLIENT_WIFI_CONNECTING
State: Rover - Fix, 2022-08-31 19:12:58.500
Rover Accuracy (m): 3.6030
Rover Accuracy (m): 2.3448
WIFI_CONNECTED
WiFi IP address: 192.168.43.204 RSSI: -37

NTRIP_CLIENT_WIFI_CONNECTED
WIFI_OFF
NTRIP_CLIENT_ON
WIFI_NOTCONNECTED
WiFi connecting to Pyrog
NTRIP_CLIENT_WIFI_CONNECTING
WIFI_CONNECTED
WiFi IP address: 192.168.43.204 RSSI: -39

NTRIP_CLIENT_WIFI_CONNECTED
Rover Accuracy (m): 1.8554
WIFI_OFF
NTRIP_CLIENT_ON
WIFI_NOTCONNECTED
WiFi connecting to Pyrog
NTRIP_CLIENT_WIFI_CONNECTING
Batt (93%): Voltage: 4.13V Discharging: -1.04%/hr Green

UBX file size: 31722 - Generation rate: 6.3kB/s - Write speed: 7469.9kB/s
WIFI_CONNECTED
WiFi IP address: 192.168.43.204 RSSI: -36

NTRIP_CLIENT_WIFI_CONNECTED
WIFI_OFF
NTRIP_CLIENT_ON
Rover Accuracy (m): 1.5459
WIFI_NOTCONNECTED
WiFi connecting to Pyrog
NTRIP_CLIENT_WIFI_CONNECTING
WIFI_CONNECTED
WiFi IP address: 192.168.43.204 RSSI: -38

NTRIP_CLIENT_WIFI_CONNECTED
WIFI_OFF
NTRIP_CLIENT_ON
NTRIP Client connection attempts exceeded!
NTRIP_CLIENT_OFF
NTRIP Client caster failed to connect. Do you have your caster address and port correct?

```

nseidle, 2022-09-01:
Did you successfully connect to your cellphone AP in previous firmware versions to connect to NTRIP Caster? I have seen this on my own cellphone AP, I assumed it was my mobile carrier (T-Mobile) blocking port 2101 but perhaps it is a firmware issue.

nseidle, 2022-09-01:
12:41:06.168 -> [ 11622][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
    12:41:06.216 -> [ 11654][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
    12:41:06.216 -> [ 11654][D][WiFiGeneric.cpp:880] _eventCallback(): STA IP: 192.168.101.133, MASK: 255.255.255.0, GW: 192.168.101.25
    12:41:06.216 -> WIFI_CONNECTED
    12:41:06.216 -> WiFi IP address: 192.168.101.133 RSSI: -59
    12:41:06.216 -> NTRIP_CLIENT_WIFI_CONNECTED
    12:41:06.403 -> NTRIP Client connected to rtk2go.com:2101
    12:41:06.403 -> NTRIP Client sending credentials: test@test.com:
    12:41:06.403 -> NTRIP Client serverRequest size: 120 of 515 bytes available
    12:41:06.449 -> NTRIP Client sending server request: 
    12:41:06.449 -> GET /bldr_SparkFun1 HTTP/1.0
    12:41:06.449 -> User-Agent: NTRIP SparkFun_RTK_Express_v2.5
    12:41:06.449 -> Authorization: Basic dGVzdEB0ZXN0LmNvbTo=
    12:41:06.449 -> 
    12:41:06.449 -> 
    12:41:06.449 -> NTRIP_CLIENT_CONNECTING
    12:41:06.495 -> UBX file size: 16561 - Generation rate: 3.3kB/s - Write speed: 1988.0kB/s
    12:41:06.495 -> [ 11951][D][NtripClient.ino:397] ntripClientUpdate(): NTRIP Client connected to caster
    12:41:06.588 -> NTRIP_CLIENT_CONNECTED
    12:41:06.634 -> [ 12082][D][WiFiClient.cpp:528] connected(): Disconnected: RES: -1, ERR: 104
    12:41:06.634 -> NTRIP Client connection dropped
    12:41:06.634 -> WIFI_OFF

Hmm, we're connected then disconnected. Let's look at the response:

    12:47:25.511 -> Caster responded with: HTTP/1.1 200 OK
    12:47:25.511 -> Date: Thu, 01 September 2022 18:47:25 UTC
    12:47:25.511 -> Server: SubCarrier Systems Corp SNIP simpleNTRIP_Caster_[wPRO]R3.06.12
    12:47:25.557 -> Last-Modified: Thu, 01 September 2022 18:47:25 UTC
    12:47:25.557 -> Content-Length: 716
    12:47:25.557 -> Content-Type: text/html
    12:47:25.557 -> Connection: Closed
    12:47:25.557 -> 
    12:47:25.557 -> <html> <body> <h1>The NTRIP Caster has banned your IP:</h1>
    12:47:25.557 -> <h3>The IP address <b>4.31.10.254</b> has been <b>Banned</b> from connecting to the Caster.</h3>
    12:47:25.557 -> <p>This IP has failed to connect over <b>500</b> times in a row without success. <br>Once the ban pe

Oh dear. I don't know exactly how IP addresses are assigned via cell phone companies but I'm not too surprised that the aggregate of rtk2go users across the t-mobile domain have done bad things to their caster, so they are banning the global IP. 

I'll scan for and turn on additional messages so that the user can see when the caster connects but then refuses to provide data.

nseidle, 2022-09-01:
More lessons learned: Cellphone APs buffer incoming data and make RTCM lumpy. For example, below is NTRIP Client using Emlid because RTK2go banned my phones IP, connected to local WiFi:

    13:57:10.236 -> [262051][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 141 RTCM bytes, pushed to ZED
    13:57:10.283 -> [262068][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 58 RTCM bytes, pushed to ZED
    13:57:10.283 -> [262095][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 56 RTCM bytes, pushed to ZED
    13:57:10.329 -> [262120][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 33 RTCM bytes, pushed to ZED
    13:57:10.376 -> [262157][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 56 RTCM bytes, pushed to ZED
    13:57:10.376 -> [262184][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 40 RTCM bytes, pushed to ZED
    13:57:10.424 -> [262220][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 53 RTCM bytes, pushed to ZED
    13:57:10.470 -> [262253][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 61 RTCM bytes, pushed to ZED
    13:57:10.470 -> [262276][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 4 RTCM bytes, pushed to ZED
    13:57:10.612 -> [262388][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 25 RTCM bytes, pushed to ZED
    13:57:11.263 -> [263068][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 128 RTCM bytes, pushed to ZED
    13:57:11.309 -> [263085][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 53 RTCM bytes, pushed to ZED
    13:57:11.309 -> [263102][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 57 RTCM bytes, pushed to ZED
    13:57:11.355 -> [263137][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 34 RTCM bytes, pushed to ZED
    13:57:11.355 -> [263164][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 55 RTCM bytes, pushed to ZED
    13:57:11.402 -> [263201][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 57 RTCM bytes, pushed to ZED
    13:57:11.449 -> [263226][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 35 RTCM bytes, pushed to ZED
    13:57:11.449 -> [263259][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 55 RTCM bytes, pushed to ZED
    13:57:11.495 -> Batt (89%): Voltage: 4.11V Discharging: 0.00%/hr Green
    13:57:11.495 -> 
    [263287][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 28 RTCM bytes, pushed to ZED
    13:57:11.542 -> [263319][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 25 RTCM bytes, pushed to ZED
    13:57:11.635 -> UBX file size: 824555 - Generation rate: 3.2kB/s - Write speed: 1931.6kB/s
    13:57:11.729 -> Rover Accuracy (m): 0.0141

You can see bytes arriving every few milliseconds.

Below is NTRIP Client over Google Pixel 4a AP:

    13:58:55.524 -> [367327][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 256 RTCM bytes, pushed to ZED
    13:58:55.756 -> [367550][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 47 RTCM bytes, pushed to ZED
    13:58:56.641 -> Batt (89%): Voltage: 4.11V Discharging: 0.00%/hr Green
    13:58:56.641 -> 
    UBX file size: 1163769 - Generation rate: 3.2kB/s - Write speed: 1959.7kB/s
    13:58:57.479 -> Rover Accuracy (m): 0.0141
    13:58:59.480 -> Rover Accuracy (m): 0.0141
    13:59:01.347 -> WiFi IP address: 192.168.101.133 RSSI: -54
    13:59:01.347 -> 
    Rover Accuracy (m): 0.0145
    13:59:01.676 -> Batt (89%): Voltage: 4.11V Discharging: 0.00%/hr Green
    13:59:01.676 -> 
    UBX file size: 1179846 - Generation rate: 3.2kB/s - Write speed: 1778.4kB/s
    13:59:03.495 -> Rover Accuracy (m): 0.0151
    13:59:05.502 -> Rover Accuracy (m): 0.0164
    13:59:06.673 -> Batt (89%): Voltage: 4.11V Discharging: 0.00%/hr Green
    13:59:06.673 -> 
    UBX file size: 1196004 - Generation rate: 3.2kB/s - Write speed: 1987.2kB/s
    13:59:07.513 -> Rover Accuracy (m): 0.0141
    13:59:08.485 -> [380287][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 233 RTCM bytes, pushed to ZED
    13:59:09.562 -> Rover Accuracy (m): 0.0141
    13:59:11.572 -> Rover Accuracy (m): 0.0143
    13:59:11.666 -> Batt (89%): Voltage: 4.11V Discharging: 0.00%/hr Green
    13:59:11.666 -> 
    UBX file size: 1212219 - Generation rate: 3.2kB/s - Write speed: 1995.4kB/s
    13:59:13.342 -> WiFi IP address: 192.168.101.133 RSSI: -55
    13:59:13.342 -> 
    Rover Accuracy (m): 0.0150
    13:59:15.626 -> Rover Accuracy (m): 0.0156
    13:59:16.749 -> Batt (89%): Voltage: 4.11V Discharging: 0.00%/hr Green
    13:59:16.749 -> 
    UBX file size: 1228401 - Generation rate: 3.2kB/s - Write speed: 1979.0kB/s
    13:59:17.682 -> Rover Accuracy (m): 0.0165
    13:59:19.733 -> [391503][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 2048 RTCM bytes, pushed to ZED
    13:59:19.733 -> Rover Accuracy (m): 0.0174
    13:59:20.016 -> [391816][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 2048 RTCM bytes, pushed to ZED
    13:59:20.296 -> [392077][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 2048 RTCM bytes, pushed to ZED
    13:59:20.576 -> [392389][D][NtripClient.ino:481] ntripClientUpdate(): NTRIP Client received 2048 RTCM bytes, pushed to ZED
[... cut ...]
Note the 13 second and 9 second delay between RTCM dumps. RTK is possible over an AP like this but it may loose precision (say from 14mm to 25mm) as the lack of available RTCM increases.

The current RC has a RTCM timeout of 30s so I don't think this lumpiness is your disconnect issue. I wonder if your cellular provider is banned by your caster?

In the next RC it will print entire caster response if we detect the word 'banned'. Please continue to test and try the RCs (opposed to firmware releases) as they have debug output turned on and will help us troubleshoot your issue.

nseidle, 2022-09-01:
[RC ready for testing](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5RC).

pyrog, 2022-09-02:
> Did you successfully connect to your cellphone AP in previous firmware versions to connect to NTRIP Caster?

Yes 😃 

>  I wonder if your cellular provider is banned by your caster?

No, NTRIP client on my smartphone is working well.
[caster.centipede.fr ](http://caster.centipede.fr) is a an "open" source caster ([BKG caster](https://igs.bkg.bund.de/ntrip/bkgcaster)) operated by the [Centipede community](https://centipede.fr).

pyrog, 2022-09-02:
RC on test 😃 

Sorry, the caster server name was wrong 🤪
```
NTRIP_CLIENT_WIFI_CONNECTED
[ 10044][E][WiFiGeneric.cpp:1230] hostByName(): DNS Failed for http://caster.centipede.fr
[ 10044][D][NtripClient.ino:351] ntripClientUpdate(): NTRIP Client caster failed to connect. Trying again.
[ 10051][D][Form.ino:159] stopWebServer(): Web Server Stopped
WIFI_OFF
```

With `caster.centipede.fr` that's fine:
```
WiFi connecting to Pyrog
NTRIP_CLIENT_WIFI_CONNECTING
State: Rover - Fix, 2022-09-02 10:15:04.526
Rover Accuracy (m): 0.7613
[  9059][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  9088][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  9088][D][WiFiGeneric.cpp:880] _eventCallback(): STA IP: 192.168.43.204, MASK: 255.255.255.0, GW: 192.168.43.46
WIFI_CONNECTED
WiFi IP address: 192.168.43.204 RSSI: -26

NTRIP_CLIENT_WIFI_CONNECTED
NTRIP Client connected to caster.centipede.fr:2101

NTRIP Client sending credentials: pyrog:*******
NTRIP Client serverRequest size: 110 of 515 bytes available
NTRIP Client sending server request: 
GET /NCY HTTP/1.0
User-Agent: NTRIP SparkFun_RTK_Surveyor_v2.5
Authorization: Basic ************

NTRIP_CLIENT_CONNECTING
Rover Accuracy (m): 0.7534
[ 10494][D][NtripClient.ino:386] ntripClientUpdate(): Caster Response: ICY 200 OK

[ 10494][D][NtripClient.ino:407] ntripClientUpdate(): NTRIP Client connected to caster
NTRIP_CLIENT_CONNECTED
Rover Accuracy (m): 0.7426
```

Thanks for displaying the caster name and port in debug ❤️ 
Could you display it on the next releases?

nseidle, 2022-09-02:
Agreed, we'll print the caster and port.

Annoyingly and incredibly, ESP32 returns only [1 or 0](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiClient.cpp#L213) on connect so we can't print the failure reason. 

Addtionally, we can cut off http:// or https:// from user input that causes DNS failures.

pyrog, 2022-09-02:
> Addtionally, we can cut off http:// or https:// from user input that causes DNS failures.

Good idea (copying and pasting is a source of errors for the distracteous 😅)

nseidle, 2022-09-21:
This has been fixed in [v2.5](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5). Please open a new issue if the problem persists.

closed 2022-09-21 by nseidle