sparkfun/SparkFun_RTK_Firmware #321: Display NTRIP state in system status

pyrog, 2022-09-03:
Could you display something like:

```
NTRIP state : client connected
NTRIP mountpoint : caster.centipede.fr:2101/NCY
NTRIP uptime : 0 00:10:50 (2 reconnexions)
```

or for NTRIP server:
```
NTRIP state : server connected
NTRIP mountpoint : caster.centipede.fr:2101/MYBASE
NTRIP uptime : 0 00:10:50 (0 reconnexions)
```

### Actual behavior
```
Menu: System Menu
[256188][W][Wire.cpp:204] begin(): Bus already started in Master Mode.
GNSS: Online - ZED-F9P firmware: HPG 1.13

SIV: 28, HPA (m): 0.015, Lat: 47.097260186, Lon: 5.493810494, Altitude (m): 277.5
Display: Offline
Accelerometer: Offline
Fuel Gauge: Online - Batt (93%) / Voltage: 4.21V
microSD: Online
Bluetooth (0CE2): Online
WiFi MAC Address: AC-67-B2-1B-0C-E0
WiFi IP address: 192.168.43.204 RSSI: -49

Uptime: 0 00:04:16.223 (Resets: 0)
f) Display microSD Files
d) Configure Debug
z) Set time zone offset: 00:00:00
b) Set Bluetooth Mode: Classic
r) Reset all settings to default
B) Switch to Base mode
R) Switch to Rover mode
W) Switch to WiFi Config mode
S) Shut down
x) Exit

```

### Expected behavior
```
Menu: System Menu
[336721][W][Wire.cpp:204] begin(): Bus already started in Master Mode.
GNSS: Online - ZED-F9P firmware: HPG 1.13

SIV: 29, HPA (m): 0.019, Lat: 47.097260279, Lon: 5.493810805, Altitude (m): 277.5
Display: Offline
Accelerometer: Offline
Fuel Gauge: Online - Batt (93%) / Voltage: 4.21V
microSD: Online
Bluetooth (0CE2): Online
WiFi MAC Address: AC-67-B2-1B-0C-E0
WiFi IP address: 192.168.43.204 RSSI: -47
NTRIP: caster.centipede.fr:2101/NCY
NTRIP mountpoint : caster.centipede.fr:2101/MYBASE
NTRIP uptime : 0 00:10:50 (0 reconnexions)

Uptime: 0 00:05:36.755 (Resets: 0)
f) Display microSD Files
d) Configure Debug
z) Set time zone offset: 00:00:00
b) Set Bluetooth Mode: Classic
r) Reset all settings to default
B) Switch to Base mode
R) Switch to Rover mode
W) Switch to WiFi Config mode
S) Shut down
x) Exit
```

nseidle, 2022-09-09:
A fine idea. Thanks for the mockup. I'd like to prevent the menu from growing as much as possible so I've combined it into one line. Please let me know if it's sufficient.

    WiFi MAC Address: B8-D6-1A-0D-B0-20
    WiFi IP address: 10.8.252.93 RSSI: -62
    System Uptime: 0 00:04:43.094 (Resets: 0)
    NTRIP Client Connected - rtk2go.com/bldr_SparkFun1:2101 Uptime: 0 00:03:21.026 (Reconnects: 0)
    f) Display microSD Files
    d) Configure Debug

nseidle, 2022-09-21:
This has been implemented in [v2.5](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5).

closed 2022-09-21 by nseidle