sparkfun/SparkFun_RTK_Firmware #322: Show coordinates Ecef and LLH at the beginning of the WIFI configuration menu.

amlago, 2022-09-04:
Sparkfun facet v2.3
I would like a little information when configuring by wifi.
Along with the firmware data I would like the Ecef and LLH coordinates to be displayed. They would be useful for me to quickly enter them as Approximate Fixed Base, for example.
When configured by usb serial this information is provided.
I think that making this small modification is possible and can be useful.

tonycanike, 2022-09-08:
Would a browser refresh get updated coordinates?  That might be sufficient.

nseidle, 2022-09-08:
Yes it would.

amlago, 2022-09-08:
Excellent, same thing.

amlago, 2022-09-09:
When will this be for download?

nseidle, 2022-09-09:
RC just pushed [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5RC).

amlago, 2022-09-09:
Well, I just tried the updated v2.5.
1) I see the coordinates correctly in the wifi configuration
2) I can assign the coordinates to the fixed base with the new button.
3) I can assign baton height and phase center offset
Now:
a) Hitting save does not save the new settings and restarts only with the old data
b) If I go back to the Wi-Fi configuration in the base coordinates, your default coordinates appear.
c) By serial configuration if you change and save the configuration

Go back to v2.4 because tomorrow I have to go to work.
I will keep trying
Cheers

amlago, 2022-09-09:
I forgot. Sending fixes with fixed base ntrip, it seems that message 1005 is not being sent. At least Rtk2go doesn't show it to me. With previous versions it looks perfect.

nseidle, 2022-09-11:
@amlago - My apologies, the firmware had a bug that caused the stack to explode (I am a bad programmer!). New binary with fix is ready for testing: [RTK_Surveyor_Firmware_RC-Sep_11_2022](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5RC).

I'm sure there will be more bugs so please keep reporting.

amlago, 2022-09-11:
Thank you very much. Programming is one of my loves from my time as a student, but I just started to know Arduino, so for now I support you with the tests.
I will keep reporting.

amlago, 2022-09-11:
In the first tests with v2.5 on 9/11 there are good and bad news.
The base values ​​entered with the button work and when saving does not restart. Those are the good ones.
When changing the profile from the Wi-Fi configuration, the Facet was restarted and of course the communication was cut off. A bad.
Enter the wifi configuration again and it seemed that everything was fine to use in fixed base mode.
Ok it starts issuing fixes and tries to connect 2 times to wifi and it disconnects forever. The corrections seem to keep increasing.
Then I tried with a mobile base, with survey in and the same thing happened.
So it will be reviewed what happens with the ntrip corrections by wifi.
The other bad news.
With version 2.5 prior to 6/9, it connected to Wi-Fi and transmitted the corrections.
I keep reporting. We are on the good way.
To make you happy, the Facet has been working professionally in moving and fixed base mode, writing down the base coordinates on a piece of paper and then entering them by hand and it has worked very well.
Cheers

nseidle, 2022-09-12:
> When changing the profile from the Wi-Fi configuration, the Facet was restarted and of course the communication was cut off. A bad.

Thanks! I will look into this.

> Ok it starts issuing fixes and tries to connect 2 times to wifi and it disconnects forever.

The way the Server currently works is this:

* If an SSID is available, connect to it
* If the connection is lost, wait 15s timeout before attempting reconnection. If SSID is not found, increase to 5 minutes. If SSID not found, increase to 10 minutes, 15 minutes, etc.

Your unit should attempt once to connect to SSID, and again after a timeout period (15s, 5 minutes, 10 minutes, etc). This will continue up to 38 hours before it will give up connecting to an SSID.

I have been testing this with my phones hot spot (Android, Google Pixel 4a) and it is working well. I can 'fake' a disconnect and the device correctly waits 15 seconds before attempting to reconnect. Please let me know what you are seeing.

amlago, 2022-09-13:
did some quick tests and got it to issue rtcm corrections.
I failed a couple of times entering and exiting the wifi configuration.
I'll keep trying. Once started it doesn't seem to fail.

nseidle, 2022-09-14:
I'm seeing AP disconnects as well, but this issue seems to have been fixed. Please start new issues if you discover new problems.

amlago, 2022-09-15:
Hello, I have been working with Facet v2.5 and it has helped me a lot to visualize the coordinates at the beginning of the WIFI configuration.
A query.
Can you add any more decimal places to improve accuracy?
  When passing the coordinates obtained with survey in to a fixed base with the new button, I think I lose some decimal figures.
I know that I am working in a relative and not absolute way with respect to the world, but so that the repetition of the work the next day is more exact, I would like some more decimal figure.
In my data controller I see the base coordinates with more figures and I have to manually adjust them in the Facet to match them.
We are already very close to having the best simple and productive software.
Cheers

nseidle, 2022-09-16:
Hi Amlago - Please start new issues for new issues so that we can track them. I will move this one to [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/325).

nseidle, 2022-09-21:
This has been implemented in [v2.5](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5).

closed 2022-09-21 by nseidle