sparkfun/SparkFun_RTK_Firmware #323: Copy ubx files by usb cable or FTP?

amlago, 2022-09-09:
I want to copy the ubx files without having to remove the micro sd.
Would it be possible to copy ubx files by usb cable?
Or implement an ftp over wifi just to access these ubx files?
Either method would be great and would allow easy access to ubx files for post processing.

tonycanike, 2022-09-09:
This is a good idea. 

Deleting files from the SD card would be useful to clean up the card. 

Sending files to the card would be useful too.  Would help for updating profiles (after editing on a computer) and saving a new firmware version to the card 

This would save a lot of fussing trying to get the card in and out of the Facet.  There's not a lot of room for fingers and those cards are small. The card sometimes misses the card reader and goes into the Facet case, where it rattles around until I take the Facet apart.

nseidle, 2022-09-09:
A fine idea. It will take us a bit to implement. We may use [ZModem](https://en.wikipedia.org/wiki/ZMODEM) as we've used it before on other products.

nseidle, 2022-09-09:
Ah, we also have file transfer over WiFi in the works.

amlago, 2022-09-11:
Good news, we keep in touch.

pyrog, 2022-09-28:
> The card sometimes misses the card reader and goes into the Facet case, where it rattles around until I take the Facet apart.

Same issue with the Surveyor and Express. The slot could be redesigned see #342.

> We may use [ZModem](https://en.wikipedia.org/wiki/ZMODEM) as we've used it before on other products.
> Ah, we also have file transfer over WiFi in the works.

SD card file server over HTTP is highly anticipated 😃 

Instead of using ZModem over USB, could you implement one of the following protocols?

-  [USB mass storage device class](https://en.wikipedia.org/wiki/USB_mass_storage_device_class)
- [OBEX](https://en.wikipedia.org/wiki/OBject_EXchange) (It could be used over BT too)

It could be more simple for end user?

amlago, 2022-12-23:
Fantastic. This is really good.
Nice job.

Merry Christmas!

amlago, 2022-12-26:
Facet v2.7 12/26
The file explorer is an excellent idea.

1) Downloading files from a Win 10 notebook went smoothly.
2) From an Android cell phone I got different results depending on the web browser used.
With Chrome the download is almost always successful.
With other browsers like Dolphin Brownser the download never happened.

amlago, 2022-12-26:
@nseidle
  I know it's a lot of work and time consuming but adding 2 buttons to the explorer, one to select all and one to deselect all, would be great.
Thank you for your great work.

nseidle, 2022-12-28:
> adding 2 buttons to the explorer, one to select all and one to deselect all, would be great.

Ah! A great idea. I will add.

nseidle, 2022-12-28:
Global checkbox added:

[image]

amlago, 2022-12-29:
Excelente

nseidle, 2023-01-19:
This has been implemented in [v3.0 firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.0).

closed 2023-01-19 by nseidle