sparkfun/SparkFun_RTK_Firmware #325: More decimals of precision for base entry

nseidle, 2022-09-16:
amlago:

> Can you add any more decimal places to improve accuracy? When passing the coordinates obtained with survey in to a fixed base with the new button, I think I lose some decimal figures. I know that I am working in a relative and not absolute way with respect to the world, but so that the repetition of the work the next day is more exact, I would like some more decimal figure. In my data controller I see the base coordinates with more figures and I have to manually adjust them in the Facet to match them.

amlago, 2022-09-16:
8 digits is enough. We are in the order of mm.
As you say, it was reported up to the seventh figure or so it seemed to me.
finger up.

nseidle, 2022-09-16:
Awesome! Thanks

nseidle, 2022-09-21:
I believe this has been fixed in [v2.5](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5). Please open a new issue if the problem persists.

closed 2022-09-21 by nseidle