sparkfun/SparkFun_RTK_Firmware #326: Make current coordinates the default coordinates

nseidle, 2022-09-16:
amlago:

> The default values of the base coordinates, which I believe are the ones corresponding to the Sparkfun Company location, should then be replaced by the values obtained at the current base location. In other words, if a base location was never obtained, the coordinates will be the default coordinates, but when locating a base, its coordinates should replace the default ones.

nseidle, 2022-09-16:
Hi @amlago - 

>  when locating a base, its coordinates should replace the default ones.

Is this in the AP Config page only? Or in the serial menu? Or both?

[image]

In the above image, when you click on on 'Fixed', then 'Geodetic', the fields are populated with the default SparkFun coordinates. Are you asking that the fields be populated with the current lat/long as shown at the [top of the page](https://user-images.githubusercontent.com/117102/190699919-00daf68b-9f45-473a-84e6-9a76cc6ef3ac.png)?

That's fine, but why not use the 'Use Current Coordinate' button? 

If we implement your feature, should we get rid of the 'Use Current Coordinate' button?

amlago, 2022-09-16:
I will try to explain myself better.
1) The use current coordinate button is great, don't even think about deleting it. ha ha
2) It has happened to me on some occasions with the Wi-Fi configuration that when I return to configure the base, the default coordinates appear again and not those of the previous station, which I entered.
I honestly don't know what I did to make that happen.

My comment was to try to ensure that after entering fixed base coordinates, they are not altered.
It may have happened that I made a mistake.

Thank you

nseidle, 2022-09-16:
1) Ok no problem :)

> It has happened to me on some occasions with the Wi-Fi configuration that when I return to configure the base, the default coordinates appear again and not those of the previous station, which I entered.

Ah! That is the crux of the issue. We do not need a new feature, we need to fix the problem!

When the AP Config page loads it should contain all the settings that are stored on that profile. If that profile uses a fixed base, the coordinates should also load. If the coordinates are not loading, please let me know.

We just discovered and fixed a bug that was, at certain times, corrupting the profile when exiting the AP Config page. This would show up as what you describe: by accident the fixed base coordinates would be cleared and reset to defaults. Please continue to test (I'll get an RC posted shortly) and let me know if you still see this issue.

nseidle, 2022-09-16:
Ok, many stack explosions later, we've located most of the problems. Please give [Sept 16 RC](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5RC) a try.

 The profiles now switch and behave as expected. Please try using a new Profile for each location you visit. The base coordinates should be assigned and stored. And when you move to a new location, change to a new profile.

nseidle, 2022-09-21:
I believe this has been fixed in [v2.5](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.5). Please open a new issue if the problem persists.

closed 2022-09-21 by nseidle