sparkfun/SparkFun_RTK_Firmware #327: Request for another change in base coordinate entry related to #269.

amlago, 2022-09-16:
Trying to improve and simplify the entry of base coordinates, I pass the following for your consideration:

1) When obtaining base coordinates with survey in, you could add the height of the rod and the difference in the phase center and save the coordinates of the mark in a file on the SD.
The corrections are sent from the coordinates of the phase center but we save the coordinates of the mark for future use. (It would be necessary to subtract from the height of the antenna the height of the pole and the difference in the center of phase)
2) The next day, I park on the mark and I would like to access the list of saved stations, by means of, for example, Wi-Fi configuration.
I indicate which point I want to use as a base and of course the height of the pole and the difference in the phase center will be indicated, in order to determine the correct height of the antenna.
 3) The list could be seen on the Base page in the wifi configuration.
4) If I want to manually enter known base coordinates, I will do it as usual.

I leave you my ideas.
Cheers

amlago, 2022-09-16:
To access the list of base points, a button would have to be added that allows the list to be displayed, as is done with the nmea messages, etc.

amlago, 2022-09-16:
I would love for this crazy idea to be possible.

amlago, 2022-09-19:
To simplify the programming a fixed amount of base points could be saved, for example 5 or 10 as fixed message data nmea is saved.  As there is little data, it would not be necessary to save a points file on the sd.  I would like you to give your ideas that are always very successful.

amlago, 2022-09-19:
Base points can be identified as Base1, Base 2, etc or by using the date and time as a name.

tonycanike, 2022-09-20:
This is a fine idea; we've discussed this in the past in some of the long and intertwined conversations on various issues.

I am not opposed to this, it would help me, especially if the file was in a CSV .txt format similar to that input and output from survey data controllers and desktop survey packages (I use Carlson SurvPC on my data collector and Carlson Survey on my desktop.)

I think there was a conversation that resulted in we should put the coordinates of the base in a profile settings file, which I am doing, but it is getting tedious to edit the profile settings files.

On another note, I'm having some success in using my data collector (with SurvPC) to set the base coordinates of the Facet.  But that's another story.

Though in this sentence, I think you meant " **subtract** the height of the rod and the difference in the phase center" to calculate the coordinates of the mark.

>     1. When obtaining base coordinates with survey in, you could add the height of the rod and the difference in the phase center and save the coordinates of the mark in a file on the SD.

amlago, 2022-09-21:
We agree on everything.

1) I said a while ago that using profiles to load different base coordinates was not useful for me.
2) If you could save the data to a txt or csv file that would be great. I am content at the moment with saving a few base points in any format or in the most practical way for programming.
3) Regarding the height of the mark you are right.
     mark height = antenna ellipsoidal height - pole height - phase center difference
    

I apologize and remind you that I communicate through Google translator.
If I do a reverse translation back to Spanish, it is still understood, I am not sure what you receive.
We will continue exchanging ideas. Thanks for sharing ideas.

amlago, 2022-09-22:
In addition, if fixed base coordinates are entered manually, we should add a button to add the coordinates to the list of base points.
So the procedure would be:
1) If the base is obtained with survey in , the coordinates are automatically added to the list of base points
2) If you add a fixed base point manually, touch the new button and add it to the list.

I keep dreaming....

nseidle, 2022-10-10:
Ok, so 

* The RTK device needs to store a list of coordinates in a separate file 'StationCoordinates.csv' or similar to SD and LittleFS
* We will need StationCoordinates-Geodetic.csv and StationCoordinates-ECEF.csv
* The list of coordinates needs to be presented any time the user might be entering coordinates. The user should be able to select a group of coordinates to fill in the various boxes.
* The list needs to be editable, but perhaps not at the time the user is selecting coordinates.
* The user should be able to copy the current ECEF or Geodetic data to the list of coordinates

Do we need to store a unique pole length and ARP for each StationCoordinate-Geodetic.csv entry?

nseidle, 2022-10-10:
Here's a bad markup, but hopefully it helps. Is this kind of what you are looking for?

[image]

The 'Copy Coordinates To List' would add a record to the list of station coordinates. The 'Load' button would copy the selected record to the ECEF boxes. The Delete button would remove a record from the list.

If a nickname matched a record already in the list, it would be overwritten.

amlago, 2022-10-10:
What is presented is a very beginning of a solution to what has been proposed. Excellent. The idea of ​​being able to add a nickname is very good.

1) For Ecef coordinates I think it would be enough, since it would be used with really fixed bases.

2) For geodesics, the idea is to save the coordinates of the mark so that the height to be saved must be obtained by subtracting the height of the pole and ARP.
mark height = antenna ellipsoidal height - pole height - phase center difference
The height of the antenna is not saved, but the height of the mark.
I don't think it's necessary to save stick height and ARP.

3) When I go to the field another day, I choose the base I need in Geodesics by one of the authorized methods and enter the new height of the pole and ARP as it is already being done.
Method 1 Manual entry
Method 2 Use current coordinates with the button
Method 3 The list of coordinates is displayed

My idea is that this method 3 is another base coordinate entry method in addition to the existing ones and what has already been programmed to follow like this or similar.
In other words, the button to use current coordinates is very useful in the field.

You should be able to check option a) use current coordinates and use the button or option b) choose from the list and use this new thing you have programmed.

Keep in touch.
Cheers

gdt, 2022-10-11:
Just to emphasize, near Boston, the actual geoid height (EGM2008) is about -29, but the model in the F9P and many other receivers is about -33.   My best guess is that receivers are using a 15 degree version of EGM96.   15 **degrees**, not minutes.  So this builtin geoid model absolutely needs to be backed out to h, always.

amlago, 2022-10-11:
Any ideas.
1) All gnss equipment takes raw data with respect to some ellipsoid, which by default is generally WGS84.

2) The corrections regarding some geoid are generally carried out by the controller or control program, whether professional or free, and we must enter the geoid model that suits our needs.
3) In short, to locate the base station it would not be necessary to embark on issues such as orthometric height corrections.
That each user in his software solves it, because personal needs are infinite.
  This is how I currently do it on my controller.

There are many programs that go from ellipsoid heights to orthometric heights, which everyone can run in the office.

Summary: to locate the base station, it is enough to work on data and heights with respect to the ellipsoid.

They are ideas...

amlago, 2022-10-11:
Anyway, I really like the discussion on the subject.
We continue exchanging ideas to improve team productivity.
Cheers

nseidle, 2022-10-11:
Thanks for continuing to educate me! Help me get the correct data to where it needs to be. The ZED-F9x reports WGS84 based coordinates. The module itself outputs altitudes in both *height above ellipsoid* and *height above mean sea level*.

The NMEA GGA message that is reported by the module and transmitted over Bluetooth to any data collector is **altitude above mean sea level** (as shown below):
[image]

However, note in the above image the geoid separation number.

The altitude used within the RTK firmware (so the LLh reported at the WiFi config page, on the system status menu, etc) is obtained from the UBX-NAV-PVT message which is **height above ellipsoid** (UBX-NAV-PVT message shown below):

[image]

So when a user sees LLh on the AP config page, and sets the coordinates for the base in LLh, the system and module will be setup for whatever 'h' they enter. Any error in h will be transmitted in the RTCM. The Bluetooth data will continue to report GGA data (predominately MSL) but I suspect the data collector will use the geoid separation number to change/correct the MSL to HAE. But I cannot say with certainty.

I believe we are ok with regards to using HAE and recording HAE, I just wanted everyone to be aware.

nseidle, 2022-10-11:
So @tonycanike:

    So depending on where the firmware determines the ZED coordinates, it might be necessary to

    1. add crude geoid height (N)
    2. subtract ARP-APC offset
    3. subtract instrument height (rod length)

I don't think need step 1 because the LLh from the ZED as it is in HAE. Correct?

amlago, 2022-10-11:
1) I insist on working in coordinates on ellipsoid because they are the most basic raw data. And we all know the WGS84 in this case.
Mark coordinates must be saved with the corresponding height. (LLh)
mark height = antenna ellipsoidal height - pole height - phase center difference

2) Regarding orthometric height corrections on geoid models, there are as many solutions as there are countries.
Almost all countries adjust a geoid model that best suits their area. I don't know if it is feasible or easy to enter geoid models in Sparkfun.
In my case, when I need orthometric heights, I add the geoid model to my controller (HCE320 from CHCnav) and it takes care of transforming the heights.
In my data controller, to use a geoid model you must enter a geoid model and check an option enabling its use. In general, we work with heights h on an ellipsoid and they are corrected to orthometrics according to the user's needs.
But my antenna keeps sending raw data on ellipsoid.

3) As I said before, if the user needs orthometric heights, transform them after going to the field in the office with some software or in his data controller.

Summary: In my case, when I need orthometric heights, I add the geoid model to my controller (HCE320 from CHCnav) and it takes care of transforming the heights,
It is not necessary for the gnss antenna to do any height transformation.
I think it is the easiest solution to implement.

tonycanike, 2022-10-11:
@nseidle believe you are correct, you don't need step 1 above.  I've never used the UBX-NAV-PVT message but per the docs it appears you are correct and you made a good choice of message to use.  Good to know!

@amlago   I believe we all 100% agree, the RTK firmware should work in heights above ellipsoid.

So, given that
   _height_   is the height above ellipsoid from the UBX_NAV_PVT messge,

the height above ellipsoid of the mark would be
    HAEmark =  _height_  -  APC_ARF_Offset  -  rodHeight 

-

gdt, 2022-10-11:
Discussion seems basically right and I agree with almost ever
ything, but a few fine points:
  1) NMEA spec predates precise height from GNSS and talks about sea level heights, I believe vaguely.
  2) In the context of WGS84, the proper term is "WGS84 Orthometric Height", which is defined as  relative to some specific equipotential and realized/accessed via HAE and the EGM2008 model.  When people say MSL, this is the more precise thing they would say, if they understand the details (which take a vast vast amount of reading).
  3) I agree with @nseidle that using the UBX message with HAE is best.   It takes all the "geoid model is bad" and round-tripping through it off the table.
  4) It is not entirely right to say that outputs are in WGS84.   They are computed as if they are in WGS84, but for RTK they are really in the frame of the reference station, which in the US is often NAD83(2011) 2010.0.    And, when using GNSS, if surveying in with SBAS, they are not in WGS84 but in ITRF2008 or ITRF2014, but that's essentially the same.
  5) Yes, countries/regions have geoid models, but those are for different purposes than EGM2008.  EGM2008 is from WGS84 to WGS84 Orthometric Height.   In the US, there is GEOID18 which transforms from NAD83 HAE to NAVD88, which is a classical height datum.   Notably NAVD88 0 surface is not an equipotential.   It is just wrong to use GEOID18 with WGS84, or EGM2008 with NAD83.   I suspect some countries have models for iTRF to some equipotential which may be equal to WGS84's, and those are "better/different versions of EGM2008".  But other countries will have models from their regional crust-fixed modern frame (GDA2020, ETRS89, etc.) to their regional height system too.  This is way beyond the scope of the ESP32!!!
  6) overall, the data collector has to be told what datum the coordinates are in, by knowing about the reference network.  I don't think this is really in the RTCM stream.

amlago, 2022-10-12:
I understand that @gdt wants to clarify theoretical terms that I think deserve a separate talk or discussion due to the depth of concepts that must be handled. If we spin fine on some spilled concepts I have some subtle discrepancies.
Maybe I didn't understand well in the translation; I remind you that English is not my original language. My apologies.

I just think that on the basics we agree and that is very important.

We will then work with geodetic coordinates with ellipsoidal heights (LLh) and let each user solve the change to orthometric height according to the geoid model that they need in their data controller or on their PC in the office.
Cheers

gdt, 2022-10-12:
I certainly agree that the proposed path forward is correct.

nseidle, 2022-10-13:
A list of common coordinates for ECEF and Geodetic have been added. It looks like this:

[image]

Responsive displays (ie, cell phones) render the item list into a single unit then present the user with the full list when its clicked. 

[image]

I think this works well enough.

[image]

If SD is available, the coordinates are recorded to SD. If not, it's recorded to internal flash. SD gets the priority (ie, if the coordinates on SD are different than LFS, SD wins).

The max number of stations is 50 which is probably *way* too many but it's easily adjustable.

nseidle, 2022-10-13:
WRT ARP and pole height: I have not implemented any math behind the input boxes because this thread had gotten a bit long. Would someone mind starting a new issue describing what needs to be done?

tonycanike, 2022-10-13:
Woah nice!

#347 submitted!   Though #330 could also apply.

amlago, 2022-10-13:
Really good job, excellent.
With these improvements, positioning in base mode is a luxury.
Congratulations Nahtan.

nseidle, 2022-10-13:
Thank you! It's always good to hear when something helps! Please test it out and let me know what needs changing.

amlago, 2022-10-13:
Regarding the math behind Baton Height and ARP.

1) Some time ago I asked Naithan to allow negative values to be assigned to staff height and ARP. This is already possible, since it was programmed a long time ago.
If we assign negative values, the heights of the marks will be saved.
That doesn't require any extra math in programming.

2) The ideal would be that the height of the mark is automatically calculated and saved in the file as we have spoken.
Mark coordinates must be saved with the corresponding height. (LLh)
mark height = antenna ellipsoidal height - pole height - phase center difference

I await your opinions.

But we have taken a giant step in the use of base mode.
Cheers

nseidle, 2022-10-13:
@amlago - I do not want to discuss ARP and staff heights on this issue. I think Tony's got a good start over on #347.

Just a note: The list of Commonly Used Coordinates (Geodetic and ECEF) is not accessible via serial menu. Nor do I think it is needed. Let me know if anyone needs a serial menu interface.

amlago, 2022-10-13:
Got it, we continue the topic at #347

nseidle, 2022-12-19:
In release v2.6.

closed 2022-12-19 by nseidle