sparkfun/SparkFun_RTK_Firmware #328: uSD card from macOS (maybe formatted on, maybe just having been mounted on) fails

gdt, 2022-09-17:
# Subject of the issue
A 32G uSD card that previously worked, and worked again after `fsck_msdos` on NetBSD, fails to be recognized.

# Your workbench
Observed on 2.4 firmware, with wifi NTRIP.
uSD probably formatted on NetBSD, and later mounted on a mac.

# Expected behavior
It should work like it does for everybody else.

# Actual behavior
As shown in #303, the firmware rejects the filesystem.

# Meta
This issue is too vague, but serves as a placeholder for others having problems.

nseidle, 2022-09-21:
Thanks for documenting for future readers. I'm going to close the issue because there's nothing to fix at the moment.

closed 2022-09-21 by nseidle

gdt, 2022-09-21:
I don't understand "nothing to fix"; it really seems like there's a bug.  I guess I'll interpret that as "too vague to be allowed to be open" and that I should file a proper issue with a repro recipe.

nseidle, 2022-09-26:
Sorry, I should have been a little more verbose: We use the [sdfat](https://github.com/greiman/SdFat) library. While it's good for users to know there may be issues with RTK products and SD cards formatted on Mac, the issue is likely with sdfat which we are not easily able to replace or modify.