sparkfun/SparkFun_RTK_Firmware #331: EOL characters change order part way through settings file  (LFCR, then CRLF)

tonycanike, 2022-09-20:
### Subject of the issue
Halfway through the settings files, the end of line characters change order, creating the appearance of extra blank lines in the first half of the file.  This is annoying when editing the files to enter coordinates for the base.

In the first half of the file, the order is LF-CR.
In the second half, it's CR-LF.

The LF-CR order gives the appearance of extra blank lines on Windows boxes.

### Your workbench
v2.5-Sep 19 2022

### Steps to reproduce
It wasn't always this way.   I found an old settings file on my PC (v1.10-Feb 21 2022) that had CR-LF through the whole file.

### Expected behavior
Line endings should be CR-LF through the whole file.

gdt, 2022-09-21:
I'm seeing problems, which I think are the same, but I hadn't chased it down enough to file an issue.

tonycanike, 2022-09-21:
For me this isn't a problem, just an annoyance.

gdt, 2022-09-21:
It's a bug whether or not it bothers you :-)  But seriously, all of these things could be processed by scripts etc. and wrong content is maybe going to cause trouble, and not going to do any good.

nseidle, 2022-09-21:
I agree it's Not Right®. Should be easy enough to fix.

tonycanike, 2022-09-21:
> But seriously, all of these things could be processed by scripts

Good point.

nseidle, 2022-09-21:
[image]

In next RC.

tonycanike, 2022-09-21:
Awesome, thanks!

commit 1b0f93d6 (2022-09-21): Change line endings to \r\n

Fix #331
closed 2022-09-21 by nseidle