sparkfun/SparkFun_RTK_Firmware #334: Connection problems in sending corrections rtcm wifi and bluetooth connection to SWmaps simultaneously.

amlago, 2022-09-22:
Sparkfun Facet v2.5
Sometimes I have used since v2.4, the sending of rtcm corrections via wifi and at the same time the connection via bluettoth to SWmaps to see the coordinates of the base.
In the final v2.5, the simultaneous Wi-Fi and Bluetooth connection is obtained but the sending of rtcm corrections via Wi-Fi is cut off after a few seconds.
If you are still connected to SWmaps via bluetooth.
I assume it must be something simple to check, because on previous occasions it worked.
Cheers

Nightglobe, 2022-09-23:
The only stable firmware was 2.5 Dev from September 11, the rest even the stable 2.5 posted, frequently restarts GPS-19984 the first variant without a bag, when activated ntrip client and you try to connect on Bluetouh, the variant of TCP / IP and port Implemented for Vespuci does not work on my Surpad program, even if I select from the terminal menu S to the status c) and n) the connection is refused. I returned to 2.4, version 2.5 is unstable and this is, maybe we are more fortunate

nseidle, 2022-09-26:
Thanks for reporting. I am having problems replicating the problem. Here's what I've tried:

* Boot unit with SD card logging
* Connect unit to local WiFi and download NTRIP data from RTK2Go
* Connect unit over BT to SW Maps to view position

Unit works as expected.

> , the simultaneous Wi-Fi and Bluetooth connection is obtained but the sending of rtcm corrections via Wi-Fi is cut off after a few seconds.

Ok, so you have a unit in base mode? 

* Boot unit with SD card logging
* Put unit into base mode
* Connect unit to local WiFi and upload NTRIP data to RTK2Go
* Connect unit over BT to SW Maps to view static base position

Unit works as expected.

Are you logging to SD card?

Why are you connecting to a base over Bluetooth? Just to view the base's location?

Does the unit reset all the time or just some of the time?

> GPS-19984 the first variant without a bag

To be clear, that is the Facet and the 'first variant' is [here](https://www.sparkfun.com/products/retired/19029). Both items are identical. 19984 just added a carrying case. I don't believe this is hardware variable but it's good to know.

Nightglobe, 2022-09-26:
I do not use SD card.I do not use their base, rover ,  I use GPS-19029, 2 units bought from  berrybase  Germany  and one from mouser (US) , the classic case use the settings made from Terratem, activate ntrip client, enter the logging data to The NTRIP service, I set WiFi and the access password to them, give the status and then get out of the menu on the Rover and and this is accessed via Bluetooth , The unit begins to restart  or randomly interrupts the NTRIP connection of GPS. If the Ntrip client  is not used and the classic bluetouth connection is used  my program work( Surpad , https://esurvey-gnss.com/product/software/surpad) . Surpad  allows connections on NMEA using Bluetouth or TCP / IP, I also use it on 2 GPS from Emlid, and honestlyFact is better , but 2. 4 work 2.5 Dev from September 11, it works, I do not understand why the last does not work and . By the way the last facet was bought by me from Germany 3 weekly a go , I update to firmware at 1.32 from 1.13 ,  with 2.4 firmware  with 2.5 Dev 11 goes with the end 2.5 no All the torment with the client in the face was that I want to take the survey master to take over the corrections on the face, my ntrip provider has implemented the ownership messages and if I use the Ntrip customer of the Master Surve does not initiate the connection, if the face initiates the NTRIP connection this goes, The problem is that in 2.5 everything is cabbage only goes in the surpad with a ntrip connection and corrections nmea bluetouh

nseidle, 2022-09-26:
The firmware generated before September 11 used ESP32 core v2.0.2. After that, the core was upgraded to v2.0.4. The 'new' core uses more RAM and may be leading to unit resets.

If possible, please turn on heap printing:

* Open serial menu
* Press 's' for system
* Press 'd' for debug
* Press '2' to enable Heap Reporting

You will start to see reports like this:

    FreeHeap: 19896 / HeapLowestPoint: 1196 / LargestBlock: 8052

We can see the current heap is ~20,000 bytes of RAM. If this ever drops below ~1000 bytes, the unit will likely reset. We watch these values as new features are added.

Would you please try enabling heap reporting and give us some feedback about what your unit's heap is before a reset?

nseidle, 2022-09-29:
Got it. I'm starting to see similar behavior. Sorry this wasn't caught sooner.

My testing notes:

v2.0.2 with BT connected transmitting to RTK2Go:

    FreeHeap: 41788 / HeapLowestPoint: 16476 / LargestBlock: 24564

v2.0.4 (v2.5 firmware) with BT connected to RTK2Go:

    FreeHeap: 18344 / HeapLowestPoint: 216 / LargestBlock: 5364

v2.0.5 (test firmware) 

    FreeHeap: 16176 / HeapLowestPoint: 252 / LargestBlock: 3188

The ESP32 core is only getting heavier on its RAM usage (not really surprising). I also think this is related to BT congestion and heap hits that we recently removed in order to avoid a custom BT library. We may bring it back. The short term solution is to revert to v2.0.2 of the ESP32 core.

I'll revert the compilation actions and get a release candidate with larger heap which should prevent unit resets.

nseidle, 2022-09-30:
Ok, please give the [release candidate](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.6RC) of v2.6 a try.

gdt, 2022-11-15:
Is the above RC still what I should be trying to test with?  My aim is NTRIP over WiFi and NMEA over Bluetooth SPP to QFIeld.

Nightglobe, 2022-11-15:
You don't have to worry, it works well and yes, I also tested it on tcp/ip and bluetooth, not in Qfield, and I haven't noticed any major deficiencies so far

closed 2022-12-19 by nseidle

nseidle, 2022-12-22:
To document some recent changes, v2.7 moves the Webserver to a pointer to free up additional RAM.

With ESP32 core v2.0.2, BT connected, and NTRIP over WiFi connected we see:

    FreeHeap: 48196 / HeapLowestPoint: 25740 / LargestBlock: 29684

This is good and an improvement over previous versions.

PR #365 (2022-12-19): Release candidate
**Note:** If you are currently using version 1.x firmware you will need to load v2.x firmware using a USB connection (SD and WiFi are not supported). We have created a Windows GUI to ease this process. Please see [Updating Firmware from GUI](https://sparkfun.github.io/SparkFun_RTK_Firmware/firmware_update/#updating-firmware-from-gui) for a step by step process. Once you've updated to v2.x firmware, you will be able to load future firmware using any method (SD, WiFi, USB, or GUI).

Changes in this release:

* (Bug fix #345) Decrease UART FIFO to avoid buffer overruns
* (Bug fi [... cut]