sparkfun/SparkFun_RTK_Firmware #335: Choose rover or base mode in Ap Config and reset.

amlago, 2022-09-24:
Most of the time I configure my Facet in the field.

1) I would like to choose the rover or base mode and that when exiting and resetting it is placed in the chosen mode.

Something similar can already be done in the serial configuration on the PC.
This makes it faster to start working and avoids pressing the power button to extend its useful life.
The truth is that the useful life of the power button is something that worries me since if it fails I am thousands of km from Colorado.
Cheers

nseidle, 2022-09-26:
Ok, we'll add.

> The truth is that the useful life of the power button is something that worries me since if it fails I am thousands of km from Colorado.

Hah! The buttons have a rated life of >1 million cycles. I'm not sure if I will believe the switch manufacturer, but If you break the button, I will send you 2 new overlays for repair :)

tonycanike, 2022-09-27:
Very nice, thanks,this is a good idea. 

I'm getting used to simply bringing up the WiFi AP config every time I use my Facets.  After a few moments to connect my phone to the Facet's WiFi AP, it's as good as any mobile app.

nseidle, 2022-10-10:
[image]

Added

amlago, 2022-10-10:
It looks great, it will be very practical in the field.

nseidle, 2022-12-19:
In release v2.6.

closed 2022-12-19 by nseidle