sparkfun/SparkFun_RTK_Firmware #336: Increase the limit to send the NMEA, NAV, RXM and MON messages using WIfi Configuration

IngridBerru, 2022-09-24:
Hello All,
There are cases where the survey is for a long period of time such as months, so sending the NEMEA messages or another type of message every 1 second or 20 seconds it is too much data to post-process. Using U-blox and TERATERM we can set the message to be sent every 60 seconds, however when I opened the wifi configuration, the system does not allow us to set up the message to 60, the maximum number is 20, and a warning message appears. I was wondering if you can increase the number at least to 60.

Best Regards,

nseidle, 2022-09-26:
Thanks for reporting! u-blox has a limit of 255. We've increased the limits to 255.

nseidle, 2022-12-19:
In release v2.6.

closed 2022-12-19 by nseidle