sparkfun/SparkFun_RTK_Firmware #337: RTK Surveyor log only $GNTXT

pyrog, 2022-09-26:
### Subject of the issue
All my last logs files are smaller than 500 bytes.
The device should log NMEA default messages.

### Your workbench
```
rtkFirmwareVersion=v2.5-Sep 21 2022
zedFirmwareVersion=HPG 1.13
```

### Steps to reproduce
One profile.
```
sizeOfSettings=4504
rtkIdentifier=37
rtkFirmwareVersion=v2.5-Sep 21 2022
zedFirmwareVersion=HPG 1.13
printDebugMessages=0
enableSD=1
enableDisplay=1
maxLogTime_minutes=1440
maxLogLength_minutes=1440
observationSeconds=60
observationPositionAccuracy=5.00
fixedBase=0
fixedBaseCoordinateType=0
fixedEcefX=-1280206.568
fixedEcefY=-4716804.403
fixedEcefZ=4086665.484
fixedLat=40.090294790
fixedLong=-105.185057610
fixedAltitude=1560.0890
dataPortBaud=460800
radioPortBaud=57600
surveyInStartingAccuracy=1.0
measurementRate=250
navigationRate=1
enableI2Cdebug=0
enableHeapReport=0
enableTaskReports=0
dataPortChannel=0
spiFrequency=16
sppRxQueueSize=2048
sppTxQueueSize=512
dynamicModel=0
lastState=0
enableSensorFusion=0
autoIMUmountAlignment=1
enableResetDisplay=0
enableExternalPulse=1
externalPulseTimeBetweenPulse_us=900000
externalPulseLength_us=100000
externalPulsePolarity=1
enableExternalHardwareEventLogging=0
profileName=LLS1
enableNtripServer=0
ntripServer_StartAtSurveyIn=0
ntripServer_CasterHost=rtk2go.com
ntripServer_CasterPort=2101
ntripServer_CasterUser=
ntripServer_CasterUserPW=
ntripServer_MountPoint=bldr_dwntwn2
ntripServer_MountPointPW=WR5wRo4H
ntripServer_wifiSSID=TRex
ntripServer_wifiPW=parachutes
enableNtripClient=1
ntripClient_CasterHost=caster.centipede.fr
ntripClient_CasterPort=2101
ntripClient_CasterUser=pyrog
ntripClient_CasterUserPW=secret
ntripClient_MountPoint=LLS1
ntripClient_MountPointPW=
ntripClient_wifiSSID=Pyrog
ntripClient_wifiPW=secret
ntripClient_TransmitGGA=1
serialTimeoutGNSS=1
pointPerfectDeviceProfileToken=
enablePointPerfectCorrections=1
home_wifiSSID=
home_wifiPW=
autoKeyRenewal=1
pointPerfectClientID=
pointPerfectBrokerHost=
pointPerfectLBandTopic=
pointPerfectCurrentKey=
pointPerfectCurrentKeyDuration=0
pointPerfectCurrentKeyStart=0
pointPerfectNextKey=
pointPerfectNextKeyDuration=0
pointPerfectNextKeyStart=0
lastKeyAttempt=0
updateZEDSettings=1
LBandFreq=1556290000
enableLogging=1
timeZoneHours=0
timeZoneMinutes=0
timeZoneSeconds=0
enablePrintWifiIpAddress=1
enablePrintState=1
enablePrintWifiState=0
enablePrintNtripClientState=1
enablePrintNtripServerState=0
enablePrintPosition=1
enableMarksFile=0
enablePrintBatteryMessages=1
enablePrintRoverAccuracy=1
enablePrintBadMessages=1
enablePrintLogFileMessages=0
enablePrintLogFileStatus=1
enablePrintRingBufferOffsets=0
enablePrintNtripServerRtcm=0
enablePrintNtripClientRtcm=0
enablePrintStates=1
enablePrintDuplicateStates=0
radioType=0
espnowPeerCount=0
enableRtcmMessageChecking=0
bluetoothRadioType=0
enableNmeaClient=1
enableNmeaServer=0
espnowBroadcast=1
antennaHeight=0
antennaReferencePoint=0.00
constellation.GPS=1
constellation.SBAS=0
constellation.Galileo=1
constellation.BeiDou=1
constellation.QZSS=1
constellation.GLONASS=1
message.UBX_NMEA_DTM.msgRate=0
message.UBX_NMEA_GBS.msgRate=0
message.UBX_NMEA_GGA.msgRate=1
message.UBX_NMEA_GLL.msgRate=0
message.UBX_NMEA_GNS.msgRate=0
message.UBX_NMEA_GRS.msgRate=0
message.UBX_NMEA_GSA.msgRate=1
message.UBX_NMEA_GST.msgRate=1
message.UBX_NMEA_GSV.msgRate=4
message.UBX_NMEA_RMC.msgRate=1
message.UBX_NMEA_VLW.msgRate=0
message.UBX_NMEA_VTG.msgRate=0
message.UBX_NMEA_ZDA.msgRate=0
message.UBX_NAV_ATT.msgRate=0
message.UBX_NAV_CLOCK.msgRate=0
message.UBX_NAV_DOP.msgRate=0
message.UBX_NAV_EOE.msgRate=0
message.UBX_NAV_GEOFENCE.msgRate=0
message.UBX_NAV_HPPOSECEF.msgRate=0
message.UBX_NAV_HPPOSLLH.msgRate=0
message.UBX_NAV_ODO.msgRate=0
message.UBX_NAV_ORB.msgRate=0
message.UBX_NAV_POSECEF.msgRate=0
message.UBX_NAV_POSLLH.msgRate=0
message.UBX_NAV_PVT.msgRate=0
message.UBX_NAV_RELPOSNED.msgRate=0
message.UBX_NAV_SAT.msgRate=0
message.UBX_NAV_SIG.msgRate=0
message.UBX_NAV_STATUS.msgRate=0
message.UBX_NAV_SVIN.msgRate=0
message.UBX_NAV_TIMEBDS.msgRate=0
message.UBX_NAV_TIMEGAL.msgRate=0
message.UBX_NAV_TIMEGLO.msgRate=0
message.UBX_NAV_TIMEGPS.msgRate=0
message.UBX_NAV_TIMELS.msgRate=0
message.UBX_NAV_TIMEUTC.msgRate=0
message.UBX_NAV_VELECEF.msgRate=0
message.UBX_NAV_VELNED.msgRate=0
message.UBX_RXM_MEASX.msgRate=0
message.UBX_RXM_RAWX.msgRate=1
message.UBX_RXM_RLM.msgRate=0
message.UBX_RXM_RTCM.msgRate=0
message.UBX_RXM_SFRBX.msgRate=1
message.UBX_MON_COMMS.msgRate=0
message.UBX_MON_HW2.msgRate=0
message.UBX_MON_HW3.msgRate=0
message.UBX_MON_HW.msgRate=0
message.UBX_MON_IO.msgRate=0
message.UBX_MON_MSGPP.msgRate=0
message.UBX_MON_RF.msgRate=0
message.UBX_MON_RXBUF.msgRate=0
message.UBX_MON_RXR.msgRate=0
message.UBX_MON_TXBUF.msgRate=0
message.UBX_TIM_TM2.msgRate=0
message.UBX_TIM_TP.msgRate=0
message.UBX_TIM_VRFY.msgRate=0
message.UBX_RTCM_1005.msgRate=0
message.UBX_RTCM_1074.msgRate=0
message.UBX_RTCM_1077.msgRate=0
message.UBX_RTCM_1084.msgRate=0
message.UBX_RTCM_1087.msgRate=0
message.UBX_RTCM_1094.msgRate=0
message.UBX_RTCM_1097.msgRate=0
message.UBX_RTCM_1124.msgRate=0
message.UBX_RTCM_1127.msgRate=0
message.UBX_RTCM_1230.msgRate=0
message.UBX_RTCM_4072_0.msgRate=0
message.UBX_RTCM_4072_1.msgRate=0
message.UBX_ESF_MEAS.msgRate=0
message.UBX_ESF_RAW.msgRate=0
message.UBX_ESF_STATUS.msgRate=0
message.UBX_ESF_ALG.msgRate=0
message.UBX_ESF_INS.msgRate=0
```
### Actual behavior
Sample log file:
```
$GNTXT,01,01,00,ESP_RST_POWERON*1C
$GNTXT,01,01,03,v2.5-Sep 21 2022*67
$GNTXT,01,01,04,HPG 1.13*37
$GNTXT,01,01,00,ESP_RST_SW*46
$GNTXT,01,01,03,v2.5-Sep 21 2022*67
$GNTXT,01,01,04,HPG 1.13*37
$GNTXT,01,01,00,ESP_RST_SW*46
$GNTXT,01,01,03,v2.5-Sep 21 2022*67
$GNTXT,01,01,04,HPG 1.13*37
$GNTXT,01,01,00,ESP_RST_SW*46
$GNTXT,01,01,03,v2.5-Sep 21 2022*67
$GNTXT,01,01,04,HPG 1.13*37
$GNTXT,01,01,00,ESP_RST_SW*46
$GNTXT,01,01,03,v2.5-Sep 21 2022*67
$GNTXT,01,01,04,HPG 1.13*37

```

pyrog, 2022-09-26:
Console display `UBX file size: 102 - Generation rate: 0.0kB/s - Write speed: 0.0kB/s` 😠

pyrog, 2022-09-26:
Set a new profile with default settings.
Console display `UBX file size: 199 - Generation rate: 0.0kB/s - Write speed: 0.0kB/s`

nseidle, 2022-09-26:
Are you able to obtain any information over Bluetooth? I ask because any data sent from ZED to ESP is transmitted to BT and well as logged. If you've got the RTK Surveyor, be sure both internal switches are in the upwards position:

[image]

If that's not the problem:

What size microSD card are you using? 

Have you successfully used this card before with the RTK products?

How was the card formatted? We recommend the [SD Card Formatter](https://www.sdcard.org/downloads/formatter/sd-memory-card-formatter-for-windows-download/).

pyrog, 2022-09-26:
> you've got the RTK Surveyor, be sure both internal switches are in the upwards position:

Before / After
[image] [image]

`UBX file size: 623134 - Generation rate: 2.9kB/s - Write speed: 1610.0kB/s` 🥳 

Thank you so much Nathan 😎

closed 2022-09-26 by pyrog

nseidle, 2022-09-26:
My sincerest apologies for my bad design :) I designed in those switches thinking it would be a nice feature. They are too easily bumped.

I recommend hot glue or other tape to make sure you don't get caught in the field without BT.

pyrog, 2022-09-26:
I opened the case to get the SD card that had fallen inside 😉 
Note: make a chamfer on the slot with a cutter help to remove the card.

> I recommend hot glue or other tape to make sure you don't get caught in the field without ~BT~ anything.

Good idea ❤️