sparkfun/SparkFun_RTK_Firmware #338: Setting NTRIP client from USB terminal don't save all parameters

pyrog, 2022-09-26:
### Subject of the issue
After resetting the device, I set the NTRIP client parameters manually from terminal.
Some parameters weren't edited.

### Your workbench
* Firmwares
```
rtkFirmwareVersion=v2.5-Sep 21 2022
zedFirmwareVersion=HPG 1.13
```
* Device was reseted `(Main Menu > System Menu > Reset all settings to default)`

### Steps to reproduce
1. Main Menu
2. GNSS Menu
3. Set WiFi SSID: 6
Enter local WiFi SSID: Pyrog
4. Set WiFi PW: 7
Enter password for WiFi network Pyrog: secret
5. Set Caster Address: 8
Enter new Caster Address: caster.centipede.fr
6. Set Caster User Name: 10
Enter user name for : pyrog
7. Set Caster User Password: 11
Enter user password for : centipede
secret

Some parameters are missing in the menu:
```
Menu: GNSS Menu
1) Set measurement rate in Hz: 4.0000
2) Set measurement rate in seconds between measurements: 0.25
3) Set dynamic model: Portable
4) Set Constellations
5) Toggle NTRIP Client: Enabled
6) Set WiFi SSID: Pyrog
7) Set WiFi PW: 
8) Set Caster Address: 
9) Set Caster Port: 2101
10) Set Caster User Name: pyrog
11) Set Caster User Password: 
12) Set Mountpoint: NCY
13) Set Mountpoint PW: 
14) Toggle sending GGA Location to Caster: Enabled
x) Exit
```
### Actual behavior
Following parameters weren't edited and saved in setting file.

```
ntripClient_CasterHost=
ntripClient_CasterUserPW=
ntripClient_wifiPW=
```

nseidle, 2022-09-26:
This baffles me. I cannot replicate at all. What terminal program are you using to do the serial menus?

[image]

Above, Transmits are terminated with CR. 

    Menu: GNSS Menu
    1) Set measurement rate in Hz: 4.0000
    2) Set measurement rate in seconds between measurements: 0.25
    3) Set dynamic model: Portable
    4) Set Constellations
    5) Toggle NTRIP Client: Enabled
    6) Set WiFi SSID: Pyrog
    7) Set WiFi PW: testsecret
    8) Set Caster Address: caster.centipede.fr
    9) Set Caster Port: 2101
    10) Set Caster User Name: pyrog
    11) Set Caster User Password: secret
    12) Set Mountpoint: bldr_SparkFun1
    13) Set Mountpoint PW:
    14) Toggle sending GGA Location to Caster: Enabled
    x) Exit

I can enter and save settings without issue. 

Do you have an SD card inserted?

nseidle, 2022-10-10:
@pyrog - I'm going to close this since I cannot replicate. If you continue to see anything problems like this, please let me know.

closed 2022-10-10 by nseidle