sparkfun/SparkFun_RTK_Firmware #339: Inconsistent command echo and command termination in Serial Menu

tonycanike, 2022-09-27:
### Subject of the issue
Terminal commands are not consistently echoed in the Serial Menu.
Terminal commands are inconsistent if they are immediate (enter one character) or require entering the command character followed by a CR.

I noticed this when reading #338 and saw @nseidle 's screen shot of his Tera Term terminal configuration setup.  He had "local echo" on in Tera Term, so I wondered if _This Is The Way_ and decided to try it.  

### Workbench
Firmware 2.5 release
Facet (the one before cases and L band corrections)
Tera Term 4.106

### Observed Behavior
I didn't exhaustively examine all the menus.  

I noted that the firmware does NOT echo the entered commands in the Main Menu but DOES echo the commands in the GNSS Menu.  

I noted that the Main Menu commands do not required a CR to terminate the command; entering one character immediately begins the command.  The GNSS Menu does require a CR.

### With Local Echo ON in Tera Term

#SparkFun RTK Facet v2.5-Sep 21 2022
** Bluetooth broadcasting as: Facet Base-4DAE **
Menu: Main Menu
1) Configure GNSS Receiver
2) Configure GNSS Messages
3) Configure Base
4) Configure Ports
5) Configure Logging
p) Configure Profiles
r) Configure Radios
s) System Status
x) Exit
1
Menu: GNSS Menu
1) Set measurement rate in Hz: 4.0000
2) Set measurement rate in seconds between measurements: 0.25
3) Set dynamic model: Portable
4) Set Constellations
5) Toggle NTRIP Client: Disabled
x) Exit
11x
SparkFun RTK Facet v2.5-Sep 21 2022
** Bluetooth broadcasting as: Facet Base-4DAE **
Menu: Main Menu
1) Configure GNSS Receiver
2) Configure GNSS Messages
3) Configure Base
4) Configure Ports
5) Configure Logging
p) Configure Profiles
r) Configure Radios
s) System Status
x) Exit
1
Menu: GNSS Menu
1) Set measurement rate in Hz: 4.0000
2) Set measurement rate in seconds between measurements: 0.25
3) Set dynamic model: Portable
4) Set Constellations
5) Toggle NTRIP Client: Disabled
x) Exit
11
Enter GNSS measurement rate in Hz:

### With Local Echo OFF in Tera Term

SparkFun RTK Facet v2.5-Sep 21 2022
** Bluetooth broadcasting as: Facet Base-4DAE **
Menu: Main Menu
1) Configure GNSS Receiver
2) Configure GNSS Messages
3) Configure Base
4) Configure Ports
5) Configure Logging
p) Configure Profiles
r) Configure Radios
s) System Status
x) Exit
*NTRIP_CLIENT_OFF
State: Rover - No Fix, 2022-09-27 12:18:53.698
Batt (100%): Voltage: 4.19V Discharging: 0.00%/hr Green

SparkFun RTK Facet v2.5-Sep 21 2022
** Bluetooth broadcasting as: Facet Base-4DAE **
Menu: Main Menu
1) Configure GNSS Receiver
2) Configure GNSS Messages
3) Configure Base
4) Configure Ports
5) Configure Logging
p) Configure Profiles
r) Configure Radios
s) System Status
x) Exit

Menu: GNSS Menu
1) Set measurement rate in Hz: 4.0000
2) Set measurement rate in seconds between measurements: 0.25
3) Set dynamic model: Portable
4) Set Constellations
5) Toggle NTRIP Client: Disabled
x) Exit

SparkFun RTK Facet v2.5-Sep 21 2022
** Bluetooth broadcasting as: Facet Base-4DAE **
Menu: Main Menu
1) Configure GNSS Receiver
2) Configure GNSS Messages
3) Configure Base
4) Configure Ports
5) Configure Logging
p) Configure Profiles
r) Configure Radios
s) System Status
x) Exit

Menu: GNSS Menu
1) Set measurement rate in Hz: 4.0000
2) Set measurement rate in seconds between measurements: 0.25
3) Set dynamic model: Portable
4) Set Constellations
5) Toggle NTRIP Client: Disabled
x) Exit
1
Enter GNSS measurement rate in Hz:

nseidle, 2022-10-01:
Well that was a sweater thread I wish I hadn't of pulled on!

Menus and data entry have been overhauled.

New rules:
* Data entry is echoed at all times, unless you turn it off (via the system menu).
* If you press enter or 'x' on a menu, it will exit that menu and return to the previous menu.
* If you only press enter or only 'x' during data entry, it will exit without changing that value and return to the previous menu.
* More filtering of illegal values

Please give everything a try on the [v2.6RC](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.6RC. There were a few gremlins but I believe all menus and data entry spots are now acting uniformly.

tonycanike, 2022-10-03:
Thank you sir!  Looks good on the 10/1 RC! Appreciate it.

closed 2022-10-03 by tonycanike