sparkfun/SparkFun_RTK_Firmware #340: Inconsistent Serial Menu names

tonycanike, 2022-09-27:
### Subject of the issue
Describe your issue here. Additionally, screenshots are easy to paste into github.

### Your workbench
Facet
Firmware V2.5 release

I noted the menu choise to "Configure GNSS Receiver" is titled the "GNSS Menu".   
The menu to "Configure GNSS Messages" is titled the "Messages Menu".

Minor issue to be sure, it's probably only a concern when folks like us are reporting issues :)

I did not exhaustively check all the menus for consistent naming agreement between the menu title displayed when one is in a menu, and the wording of the higher-level menu selection that takes one into the menu.

pyrog, 2022-09-28:
I didn't check too, but labels should be consistent with Wifi AP settings.

tonycanike, 2022-10-03:
Looks awesome, thanks!  Appreciate it @nseidle !

commit 737e7769 (2022-12-19): Make menu titles consistent

Fix #340
closed 2022-10-03 by tonycanike