sparkfun/SparkFun_RTK_Firmware #345: UBX Log corruption with V2.5

tonycanike, 2022-10-03:
### Subject of the issue
I'm seeing some log file corruption again.   Four hour files gathered with V2.5 firmware, all constellations, 1 Hz.

Most of the corruption was early in the log file during the time I was using WiFi AP to configure the Facet, but UBX_Integrity_Checker.py reported a few resyncs much later in the log file.

I''ve loaded the 10/1/22 RC and will attempt again today.  I'll update as I learn more.

### Your workbench

V2.5
Facets

### Steps to reproduce
Tell us how to reproduce this issue. Please post any log files from serial output that may have been generated.

### Expected behavior
Tell us what should happen

### Actual behavior
Tell us what happens instead

tonycanike, 2022-10-03:
Results using the 10/1/22 firmware.  Looks like issues were early in the file.

C:\Users\canike\Carlson Projects\Jobs\P1014 Facet PPP\2022-10-03 Roost Table>python UBX_Integrity_Checker.py SFE_Facet_221003_131234.ubx
UBX Integrity Checker

Could this file contain any NMEA messages? (Y/n):
Do you want to repair the file? (y/N):

Processing SFE_Facet_221003_131234.ubx

NMEA $ found at byte 0. Attempting to process the message
Panic!! UBX checksum error!
Sync lost at byte 7719. Attemting to re-sync.
Sync has been lost. Currently processing byte 7719. Rewinding to byte 7677
NMEA $ found at byte 7691. Attempting to process the message
Sync successfully re-established at byte 7742. The NMEA message started at byte 7691

Panic!! NMEA checksum error!
Sync lost at byte 7893. Attemting to re-sync
Sync has been lost. Currently processing byte 7893. Rewinding to byte 7860
NMEA $ found at byte 7864. Attempting to process the message
Sync successfully re-established at byte 7895. The NMEA message started at byte 7864

Panic!! NMEA checksum error!
Sync lost at byte 12145. Attemting to re-sync
Sync has been lost. Currently processing byte 12145. Rewinding to byte 12061
UBX Sync Char 1 (0xB5) found at byte 12109. Checking for Sync Char 2
Sync successfully re-established at byte 12156. The UBX message started at byte 12109

Panic!! UBX checksum error!
Sync lost at byte 13448. Attemting to re-sync.
Sync has been lost. Currently processing byte 13448. Rewinding to byte 12214
NMEA $ found at byte 12217. Attempting to process the message
Panic!! Excessive NMEA message length!
Sync lost at byte 12346. Attemting to re-sync
NMEA $ found at byte 12417. Attempting to process the message
Panic!! NMEA checksum error!
Sync lost at byte 12526. Attemting to re-sync
NMEA $ found at byte 12697. Attempting to process the message
Panic!! NMEA checksum error!
Sync lost at byte 12761. Attemting to re-sync
UBX Sync Char 1 (0xB5) found at byte 12787. Checking for Sync Char 2
Panic!! Was expecting Sync Char 2 (0x62) but did not receive one!
Sync lost at byte 12788. Attemting to re-sync
NMEA $ found at byte 12947. Attempting to process the message
Sync successfully re-established at byte 12990. The NMEA message started at byte 12947

NMEA $ found at byte 13075. Attempting to process the message
Panic!! NMEA checksum error!
Sync lost at byte 215552. Attemting to re-sync
Sync has been lost. Currently processing byte 215552. Rewinding to byte 215492
NMEA $ found at byte 215636. Attempting to process the message
Panic!! Excessive NMEA message length!
Sync lost at byte 215765. Attemting to re-sync
NMEA $ found at byte 215868. Attempting to process the message
Sync successfully re-established at byte 215898. The NMEA message started at byte 215868

Processed 2407768 bytes
File size was 2407768
Longest valid UBX message was 1240 bytes
Longest valid NMEA message was 83 characters
Message types and totals were:
Message type: GNTXT   Total: 10
Message type: 0x02 0x15   Total: 1556
Message type: GNRMC   Total: 1556
Message type: GNGGA   Total: 1555
Message type: GNGSA   Total: 4687
Message type: GPGSV   Total: 2493
Message type: GLGSV   Total: 1726
Message type: GAGSV   Total: 31
Message type: GBGSV   Total: 22
Message type: GQGSV   Total: 392
Message type: GNGST   Total: 1556
Message type: 0x02 0x13   Total: 9506
Number of successful resyncs: 5

Bye!

C:\Users\canike\Carlson Projects\Jobs\P1014 Facet PPP\2022-10-03 Roost Table>

tonycanike, 2022-10-03:
Today I got what appears to be a 4-hour uncorrupted log file.   

I did this by setting the measurement rate to 1 Hz, only collecting GPS and GLONASS data, and not using the WiFi AP config.

I'll paste in the config file and UBX Integrity Checker output below.

### -----------------------------------------

UBX Integrity Checker

Could this file contain any NMEA messages? (Y/n):
Do you want to repair the file? (y/N):

Processing C:\Users\canike\Carlson Projects\Jobs\P1017 Eckle Static\20221003 Eckle Logging Clearing\SFE_Facet_221003_145712.ubx

NMEA $ found at byte 0. Attempting to process the message

Processed 32382018 bytes
File size was 32382018
Longest valid UBX message was 1272 bytes
Longest valid NMEA message was 83 characters
Message types and totals were:
Message type: GNTXT   Total: 3
Message type: 0x02 0x15   Total: 16198
Message type: GNRMC   Total: 16198
Message type: GNGGA   Total: 16198
Message type: GNGSA   Total: 48594
Message type: GPGSV   Total: 24377
Message type: GLGSV   Total: 22532
Message type: GQGSV   Total: 4050
Message type: GNGST   Total: 16198
Message type: 0x02 0x13   Total: 153078

Bye!

### --------------------------------------------------

sizeOfSettings=4520
rtkIdentifier=38
rtkFirmwareVersion=v2.6-Oct  1 2022
zedFirmwareVersion=HPG 1.32
printDebugMessages=0
enableSD=1
enableDisplay=1
maxLogTime_minutes=1440
maxLogLength_minutes=1440
observationSeconds=60
observationPositionAccuracy=5.00
fixedBase=0
fixedBaseCoordinateType=0
fixedEcefX=-1280206.568
fixedEcefY=-4716804.403
fixedEcefZ=4086665.484
fixedLat=40.090294790
fixedLong=-105.185057610
fixedAltitude=1560.0890
dataPortBaud=460800
radioPortBaud=57600
surveyInStartingAccuracy=1.0
measurementRate=1000
navigationRate=1
enableI2Cdebug=0
enableHeapReport=0
enableTaskReports=0
dataPortChannel=0
spiFrequency=16
sppRxQueueSize=2048
sppTxQueueSize=512
dynamicModel=0
lastState=0
[... 134 lines cut ...]
message.UBX_MON_RXBUF.msgRate=0
message.UBX_MON_RXR.msgRate=0
message.UBX_MON_TXBUF.msgRate=0
message.UBX_TIM_TM2.msgRate=0
message.UBX_TIM_TP.msgRate=0
message.UBX_TIM_VRFY.msgRate=0
message.UBX_RTCM_1005.msgRate=0
message.UBX_RTCM_1074.msgRate=0
message.UBX_RTCM_1077.msgRate=0
message.UBX_RTCM_1084.msgRate=0
message.UBX_RTCM_1087.msgRate=0
message.UBX_RTCM_1094.msgRate=0
message.UBX_RTCM_1097.msgRate=0
message.UBX_RTCM_1124.msgRate=0
message.UBX_RTCM_1127.msgRate=0
message.UBX_RTCM_1230.msgRate=0
message.UBX_RTCM_4072_0.msgRate=0
message.UBX_RTCM_4072_1.msgRate=0
message.UBX_ESF_MEAS.msgRate=0
message.UBX_ESF_RAW.msgRate=0
message.UBX_ESF_STATUS.msgRate=0
message.UBX_ESF_ALG.msgRate=0
message.UBX_ESF_INS.msgRate=0

### -------------------------------------

Eric-FR, 2022-10-12:
I have the same issue with the RTK Express. And checking back my files, it seems not being so new.
v1.11-Mar 14 2022 No problem
v1.11-Apr  5 2022 Problem
v2.2 Problem
v2.5 Problem
(although I may have changed acquisition parameters at the same time...).

Attached file faced 78 resync with UBX Integrity Checker:

[SFE_Express_220730_135955.zip](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/9768824/SFE_Express_220730_135955.zip)

nseidle, 2022-10-13:
@tonycanike - As always, thanks for reporting

@Eric-FR - This is *very* good feedback as it helps us pinpoint, via the changelog, what we broke. The only relevant change I see is a change to serialTimeoutGNSS from 0 to 1ms in [v1.12](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.12). 

There are many variables that we need to test so I welcome help from any reader. This is not the first time this issue has been reported. @pyrog - I know you've been reporting this type of error over the months and releases. We (at SparkFun) have compiled a tremendous amount of testing data and have insights but I worry my preconceived notions will only hinder finding what's causing this issue.

How to run a test:
* Use whatever SD card you have available but note its capacity and any concerns you have about its quality. I believe we will be able to support nearly all grades of SD cards except for the obvious "[1TB for $1](https://www.amazon.com/Memory-1024GB-Adapter-Bluetooth-Computers/dp/B0BCHG383X)" cards.
* Format an SD card with [SD Card Formatter](https://www.sdcard.org/downloads/formatter/) or other.
* Setup a unit in rover mode, logging data to an SD card.
* Configure the test settings: load the test firmware, set the logging rate, messages, constellations, etc.
* Capture 15 minutes to >4 hours of data
* Pull the log from the SD card.
* Run the [UBX Integrity Checker](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/Utils/UBX_Integrity_Checker.py) against the log file to see if there are errors. **Note:** the specific errors are not important, but the number of errors is. We are hoping for zero errors but if we see a trend to fewer errors, that may be helpful.

Tests that need to be done (I welcome others):
* Does firmware v1.11 (full release) have this issue?
* On firmware v2.6RC, does changing the serialTimeoutGNSS back to 0 fix the issue? (This setting is available under 's'->'d'->'9')
* On firmware v2.6RC, does reducing the update rate to 1Hz fix the issue?

I believe if a corruption is going to happen, it happens within ~15 minutes so I recommend running short tests before running an overnight test.

My office test setup is an RTK Express with an SMA extension running to a fixed antenna on the roof. If you have a Facet, my apologies but this will be tedious (at least you'll get your steps in!). If anyone needs an roof antenna or SMA extension, please email me - I'm happy to provide equipment.

Eric-FR, 2022-10-14:
I never buy memory cards online but always go to physical well known shops near me, to avoid fake cards.

I will start some testing this week-end.

pyrog, 2022-10-16:
### Workbench

- [RTK_Surveyor_Firmware_RC-Oct_13_2022.bin](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/download/v2.6RC/RTK_Surveyor_Firmware_RC-Oct_13_2022.bin)
- NMEA only at 4Hz
- Rover mode with **internal NTRIP server**
- `Panic!! NMEA checksum error!`

File | Size (bytes) | Resync
--| -- | --
SFE_Surveyor_221016_075333.ubx | 2349025 | 3
SFE_Surveyor_221016_092833.ubx | 13913100 | 75
SFE_Surveyor_221016_112701.ubx | 3532900 | 22
SFE_Surveyor_221016_114843.ubx | 12873467 | 73

pyrog, 2022-10-16:
- GNSS Serial Timeout: 0

File | Size (bytes) | Resync
--| -- | --
SFE_Surveyor_221016_164048.ubx | 3945046 | 23

tonycanike, 2022-10-17:
I ran a couple tests with the 10/13/ RC

The first test was about 4.5 hours.  Receiver running autonomously (no RTCM.)  Got two resyncs very early on in the UBX file. 

The second test was about 5.5 hours.  RTK rover getting NTRIP from public servers.  Usually fixed solution.

Got five resyncs and a number of messages well throughout the file.  Perhaps receiving the NTRIP feed consumed extra resources.

### first test (autonomous)

Start of log file:
$GNTXT,01,01,00,ESP_RST_POWERON*1C
$GNTXT,01,01,03,v2.6-Oct  1 2022*68
$GNTXT,01,01,04,HPG 1.32*34

Analysis output:

C:\....>python UBX_Integrity_Checker.py SFE_Facet_221015_172738.ubx
UBX Integrity Checker

Could this file contain any NMEA messages? (Y/n):
Do you want to repair the file? (y/N):

Processing SFE_Facet_221015_172738.ubx

NMEA $ found at byte 0. Attempting to process the message
Panic!! Excessive NMEA message length!
Sync lost at byte 43107. Attemting to re-sync
Sync has been lost. Currently processing byte 43108. Rewinding to byte 42979
NMEA $ found at byte 43415. Attempting to process the message
Sync successfully re-established at byte 43445. The NMEA message started at byte 43415

Panic!! UBX checksum error!
Sync lost at byte 96257. Attemting to re-sync.
Sync has been lost. Currently processing byte 96257. Rewinding to byte 96215
NMEA $ found at byte 96310. Attempting to process the message
Sync successfully re-established at byte 96340. The NMEA message started at byte 96310

Processed 52667387 bytes
File size was 52667387
Longest valid UBX message was 1944 bytes
Longest valid NMEA message was 83 characters
Message types and totals were:
Message type: GNTXT   Total: 23
Message type: 0x02 0x15   Total: 18584
Message type: GNRMC   Total: 18584
Message type: GNGGA   Total: 18584
Message type: GNGSA   Total: 92850
Message type: GNGST   Total: 18583
Message type: 0x02 0x13   Total: 257000
Message type: GPGSV   Total: 27021
Message type: GLGSV   Total: 22674
Message type: GQGSV   Total: 4645
Message type: GAGSV   Total: 25247
Message type: GBGSV   Total: 17377
Number of successful resyncs: 2

Bye!

### second test (RTK rover)

Start of log file:
$GNTXT,01,01,00,ESP_RST_POWERON*1C
$GNTXT,01,01,03,v2.6-Oct 13 2022*7B
$GNTXT,01,01,04,HPG 1.32*34
$GNTXT,01,01,07,4DAE*22

Analysis output:

C:\....>python UBX_Integrity_Checker.py SFE_Facet_221017_125247.ubx
[... 99 lines cut ...]
Message type: GNRMC   Total: 261
Message type: GNGGA   Total: 261
Message type: GNGSA   Total: 1305
Message type: GPGSV   Total: 431
Message type: GLGSV   Total: 305
Message type: GAGSV   Total: 319
Message type: GBGSV   Total: 149
Message type: GQGSV   Total: 66
Message type: GNGST   Total: 262
Message type: 0x06 0x13   Total: 13
Message type: 0x05 0x01   Total: 679
Message type: 0x0A 0x06   Total: 2
Message type: 0x05 0x00   Total: 30
Message type: 0x06 0x8B   Total: 64
Message type: 0x06 0x00   Total: 16
Message type: 0x0A 0x04   Total: 2
Message type: 0x06 0x71   Total: 2
Message type: 0x01 0x04   Total: 2241
Message type: 0x01 0x07   Total: 2235
Message type: 0x01 0x14   Total: 2210
Message type: 0x01 0x35   Total: 1119
Message type: 0x02 0x32   Total: 562
Number of successful resyncs: 5

Bye!

Eric-FR, 2022-10-17:
Yesterday, I made a mountain bike trip of 4 hours with the RTK Express and last firmware update : 
`$GNTXT,01,01,03,v2.6-Oct 13 2022*7B`

I got a lot of errors:

[image]

```
Processing SFE_Express_221016_102210.ubx

NMEA $ found at byte 0. Attempting to process the message
Panic!! UBX checksum error!
Sync lost at byte 185544. Attemting to re-sync.
Sync has been lost. Currently processing byte 185544. Rewinding to byte 184910
UBX Sync Char 1 (0xB5) found at byte 185103. Checking for Sync Char 2
Panic!! Was expecting Sync Char 2 (0x62) but did not receive one!
Sync lost at byte 185104. Attemting to re-sync
NMEA $ found at byte 185220. Attempting to process the message
Panic!! NMEA checksum error!
Sync lost at byte 185319. Attemting to re-sync
UBX Sync Char 1 (0xB5) found at byte 185331. Checking for Sync Char 2
Panic!! Was expecting Sync Char 2 (0x62) but did not receive one!
Sync lost at byte 185332. Attemting to re-sync
UBX Sync Char 1 (0xB5) found at byte 185504. Checking for Sync Char 2
Sync successfully re-established at byte 185559. The UBX message started at byte 185504
[...]
Panic! Maximum rewind attempts reached! Aborting...

Processed 3066688 bytes
File size was 69441916
FILE SIZE MISMATCH!!
Longest valid UBX message was 1816 bytes
Longest valid NMEA message was 86 characters
Message types and totals were:
Message type: GNTXT   Total: 4
Message type: 0x02 0x15   Total: 622
Message type: 0x01 0x07   Total: 652
Message type: 0x01 0x02   Total: 654
Message type: GNRMC   Total: 654
Message type: GNGGA   Total: 653
Message type: GNGSA   Total: 3269
Message type: GPGSV   Total: 4367
Message type: GLGSV   Total: 3207
Message type: GAGSV   Total: 3197
Message type: GBGSV   Total: 3412
Message type: GQGSV   Total: 656
Message type: GNGST   Total: 654
Message type: 0x01 0x35   Total: 640
Message type: 0x02 0x13   Total: 8216
Number of successful resyncs: 101

Bye!

```

Settings and full report attached:

[SFE_Express_Settings_2.txt](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/9804840/SFE_Express_Settings_2.txt)
[murier_integrity.txt](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/9804841/murier_integrity.txt)

nseidle, 2022-10-18:
Thanks all! There is no need to copy paste the output from the UBX checker; all we need to know is what test you attempted (ex: v1.11 firmware shows errors) and a summary of the integrity checker (success or number or resyncs) will suffice.

From my own notes:

* v2.6 - Reducing to 0ms, log errors (4Hz, 7 msgs)
* v2.6 - Replace task with v1.11 task, log errors
* v1.1 - 4Hz/7msgs produces errors.
* v2.6 - with v1.11 task. Reduce to 1Hz. No errors.

Currently I don't see a firmware that *doesn't* have errors. v1.11 produces errors similar to v2.6. However, reducing to 1Hz allows error free logging (a few others have seen this as well). But I have a few more ideas to try to get fewer (hopefully zero) errors.

pyrog, 2022-10-19:
Cause of the log corruption seem [data race](https://en.wikipedia.org/wiki/Race_condition#Data_race) for me.

Does reading and writing to F9P buffer are atomic?
Same for writing to SD card?

tonycanike, 2022-10-19:
> **Note:** This issue is related to log corruption, regardless of firmware version and message format.
> 

Agreed, I updated the issue title.

nseidle, 2022-10-19:
@pyrog - Thanks! You're asking a nerd about his code, I'd love to share! :) I appreciate your thoughts. Have a look at the [older version of the code](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/8127a7e2ca2fa383ef84fb6a0de583890824ddaa/Firmware/RTK_Surveyor/Tasks.ino#L72) (as it is a lot simpler to follow):

    //Attempt to write to file system. This avoids collisions with file writing from other functions like recordSystemSettingsToFile()
    if (xSemaphoreTake(sdCardSemaphore, fatSemaphore_shortWait_ms) == pdPASS)
    {
      ubxFile.write(rBuffer, s);

      xSemaphoreGive(sdCardSemaphore);
    } //End sdCardSemaphore
    else
    {
      log_d("sdCardSemaphore failed to yield, %s line %d\r\n", __FILE__, __LINE__);
    }

Any access to the SD card is through a semaphore meaning only one thing can talk to the SD card at a time, thus I believe all writes are atomic. The ESP32 is dual core and runs FreeRTOS (real time operating system). In short, this means various tasks can run simultaneously and we don't know ahead if time (really) what task will be accessing which piece of hardware. The semaphore prevents other tasks from having the SPI/SD card open at the same time we trying to record log data. 

The firmware comes to a screeching halt when multiple things try to access the same hardware so we are confident the semaphore is working as expected.

The [current version of the code](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Firmware/RTK_Surveyor/Tasks.ino#L238) that writes to the SD card  is more complex because it implements a circular buffer to catch all incoming data from the ZED-F9P incase the semaphore cannot be unlocked in the given amount of time.

pyrog, 2022-10-19:
@nseidle (just after posting my previous message) I saw only one mutex, for gaining access to the SD card 😄 
When reading several articles about data race, I understood that the issue could also be located at the input buffer.
The buffer containing the F9P message could be overwrite (by another task or an interruption) when the buffer is currently parsed.

So even with a Semaphore on the SD Card, this "corrupted" buffer could be "safely" written in the log file.
I don't dig into your code, but I don't see another semaphore to protect the input buffer.
Even with a real time OS and a single core processor, data race could happened.

But maybe the issue in the new code is that the circular buffer is too small (or the code for logging is too slow) ?

pyrog, 2022-10-19:
Does accessing i.e. `online.logging` is atomic ?
What happened if `wifiNmeaData()` or `bluetoothWriteBytes()` are too long ?

ButtonCheckTask as the same priority than  F9PSerialReadTask or  F9PSerialWriteTask (1. 3 being the highest, and 0 being the lowest)
Does F9PSerialReadTask could have a higher priority than ButtonCheckTask?

Eric-FR, 2022-10-21:
RTK Express
$GNTXT,01,01,03,v2.6-Oct 13 2022*7B
$GNTXT,01,01,04,HPG 1.32*34
$GNTXT,01,01,07,181E*2B

New µSD card.
Factory reset.

5 default NMEA messages (4 Hz)

30 mn recording with RTK correction through BT: more than 100 rewind.
15 mn recording without RTK: integrity OK.
Again
15 mn recording with RTK/BT: more than 100 rewind.
15 mn recording without RTK: integrity OK.

Eric-FR, 2022-10-22:
New test this morning suggested by pyrog.
Same as previous. BT connexion with SW_Maps to get the position on the smartphone but without sending RTK corrections to the Express. 15 mn recording: integrity OK.

pyrog, 2022-10-24:
Recording 14 mn of NMEA at 4 Hz.
Internal NTRIP client on.
Wifi AP connected but smartphone disconnected from internet, so the device can't receive RTCM.

File | Size (bytes) | Resync
--| -- | --
SFE_Surveyor_221022_161511.ubx | 2504304 | 1

https://swairlearn.bluecover.pt/nmea_analyser report:
```
Bad header and termination: 44,44,10,29,053,46,3*76
Invalid Checksum: $GLGSV,2,2,06,79,38,041,49,88,09,028,29*4C

```

nseidle, 2022-10-25:
Just an update: We are working a lot on this issue. There's a lot learned and discovered but we don't have a flawless solution yet. 

@pyrog - Thank you for the comparisons! 

> Note: UBX_Integrity_Checker don't seem to detect Invalid NMEA Checksum.

I believe [it does](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/main/Utils/UBX_Integrity_Checker.py#L338).

> following messages aren't logged to the SD Card

Good! You are doing a lot of what we are: comparing the ZED's USB output to the SD card. They should be identical, but only if you have configured the USB interface on the ZED properly. This is not trivial. You can use u-center but you need to modify a lot of options. Below is an example: the GSV message is decimated to once per second on the UART interface, but is 4Hz on the USB interface.

[image]

This is complicated enough we wrote a helper function in the debug menu. 17 will duplicated the UART settings to the USB interface:

[image]

Only after this has been set should you compare USB output from the ZED to the SD logs. There's still discrepancies but you will  see far fewer:

[image]

pyrog, 2022-10-26:
> Note: UBX_Integrity_Checker don't seem to detect Invalid NMEA Checksum.

Oops! it doesn't detect "bad header and termination" (see test with SFE_Surveyor_221024_091005.ubx)
This is probably normal because this tool is used to parsed primarily UBX, not NMEA.

> They should be identical, but only if you have configured the USB interface on the ZED properly. 

You're right. I thought that the device configured automatically both F9P outputs.
At least for me, one interest of Sparkfun RTK device is to avoid to use u-Center for setting the F9P.
 
> This is not trivial. 

👍 

> You can use u-center but you need to modify a lot of options. Below is an example: the GSV message is decimated to once per second on the UART interface, but is 4Hz on the USB interface.

Manual settings with U-Center are useful for fine tuning the f9p, but with your product I thing we shouldn't use it.

Anyway, the Sparkfun device should log to the SD card exactly what it receive from the f9p.
>This is complicated enough we wrote a helper function in the debug menu. 17 will duplicated the UART settings to the USB interface:
>Only after this has been set should you compare USB output from the ZED to the SD logs

Okay, I will use this option in next tests 😃

tonycanike, 2022-10-28:
Thanks @nseidle!

I ran a 2 hour test with the 26-Oct RC at 1Hz and logging defaults (7 msgs.)  File looked clean according to the UBX integrity checker.  

I attempted a test at 4 Hz but ran into an unrelated problem.

Eric-FR, 2022-10-28:
Format of the µSD card you provided with official SD card software.
Firmware update to 26-Oct RC from the SD card.

First run for 15 min. as prepared (should be NMEA only). No error message.

Change to NMEA+UBX default messages. 15 min. with RTK correction. No error message.

Same as previous incl. RTK correction but with overnight (and day) running. A bunch of errors. The first occurred around 500 kB i.e. corresponding to 1 min. recording. UBX Integrity Checker quickly stopped after 100 resync.

[image]

Same as previous without RTK correction. Overnight running. Still errors but less:

[image]

So, with same parameters one may or may not have errors. Sending RTK corrections seems to worth the situation.

pyrog, 2022-10-30:
Two long tests with usual SD card (not reformatted).
Firmware v2.6-Oct 28 2022

**No errors detected** with UBX_Integrity_Checker 😎

With JOSM:
[image]
[image]

Message | SFE_Surveyor_221029_092539.ubx | SFE_Surveyor_221030_104606.ubx
-- | -- | --
File size | 19108900 | 24828676
Longest valid UBX message | 0 | 0
Longest valid NMEA message | 117 | 121
GNTXT | 4 | 7
GNRMC | 2551 | 15419
GNGGA | 2551 | 15419
GNGSA | 10204 | 61676
GNGST | 2551 | 15419
PUBX | 21829 | 28081
GPGSV | 30707 | 39684
GLGSV | 29672 | 38222
GAGSV | 28341 | 41657
GBGSV | 22781 | 29515
GPRMC | 19279 | 12662
GPGGA | 19279 | 12662
GPGSA | 77116 | 50648
GPGST | 19278 | 12662
GPTXT | 15 | 6

nseidle, 2022-10-31:
@pyrog - Thank you! Glad to hear it is working. I suspect the single NMEA checksum error is at the start of the log. We currently do a hard start to logging, meaning, whatever is in the buffer, we log. So it's more often than not that a partial message is recorded. We might be able to work around this. I will think on it.

@Eric-FR - Thank you for testing! Long term logging during an RTK fix is not something I had thought a user would need. How are you obtaining the RTCM corrections? BT or WiFi? I assume it is WiFi. I consider long term RTK logging a rare user case, but I will continue to test for it. Please let me know if there are other issues you are seeing.

tonycanike, 2022-10-31:
> Long term logging during an RTK fix is not something I had thought a user would need.

This is not uncommon at all.   PPK  - post processed kinematic - in addition to RTK has some advantages.   Surveyors sometimes uses a variation of PPK called "Stop and Go"  to distinguish it from full post processed kinematic (like the track of a drone.)   Stop and Go is PPK to determine the location of discrete survey points and not, say, the continuous track of a moving vehicle.

The advantages include:

1.  If you are doing a day RTK work and temporarily move out of range of your base-rover comm link (e.g. the 100mW Holybros aren't powerful enough), you can keep working even though you don't have an RTK fix for, say, an hour of your day.  Back in the office you can post process the raw logs from the rover and the base to get PPK solutions when the RTK comm link was down.
2. **Bad fixes with RTK -- 1 or 2 foot busts --  do happen** even when you see the magic .014 and the doublebulls eye -- do happen and _everyone needs to know that._  The rover gives you no warning whatsoever.  $20K Trimbles and the like have extensive internal processing to prevent this.  Sometimes the algorithm in the ZED just doesn't get it right.   You can post process the data (PPK) to verify the RTK findings and check for bad fixes.

Watch from about 2:10 - 3:22 in this video
https://www.youtube.com/watch?v=MLRFHn9tQZM&list=PLf5ROJWC2pbZcF3RUufD645znvgvPFgcv&index=9

The Carlson SurveyGNSS manual describes this ("Processing of Stop and Go Data") on page 40.   The inputs are the raw GNSS observations of the Base and the Rover (eg UBX, rinex, etc), and the raw data log (eg .RW5 file) from the data collector.

The raw data log (eg. RW5) file from the survey data collector contains the survey point observations (point name, point description, time, coordinates, etc.) that the surveyor collected.  The RW5 is like an automatic digital version of what a surveyor used to write down on paper.   You know what's in the two rinex files.    The processing uses the point names and times from the RW5 file to pick times to determine PPK location solutions from the two RINEX files.

https://www.carlsonsw.com/wordpress/wp-content/uploads/2014/09/Carlson-SurveyGNSS-UserGuide.pdf

Eric-FR, 2022-11-05:
Although my tests were quite long with more than 24 hours, a 12 hours run with car across France or Madagascar is quite common.

But, even for smaller trips, I find corrupted sentences. Last week-end:
- forward trip 2h30 without correction : 12 resyncs
- first in place move of 2 hours, with RTK corrections : integrity check stopped after 100 resyncs around 4 MB (file size is 120 MB)
- second in place move of 1h30, with RTK corrections : integrity check stopped after 100 resyncs around 20 MB (file size is 100 MB)
- backward trip 2h45 with RTK correction : integrity check stopped after 100 resyncs around 30 MB (file size is 150 MB)

Corrections are sent with BT from smartphone, with BT_GNSS in the above example.

BTW, I'm also interested by the Stop&Go idea. Will open an issue.

nseidle, 2022-12-28:
As of Dec 22, we have added a UBX/RTCM/NMEA parser to validate incoming messages. This helps reject any message that becomes mal-formed in flight from the ZED-F9x to ESP32. This parser has been particularly good at rejecting bad messages when entering/exiting the config menu when there is a large amount of file access and modification (as settings files get updated at the same time we need to be logging). Preliminary findings across ~60 hours of test logs shows no corruptions in the logs. Additionally, we're able to convert to RINEX and get CSRS reports just fine.

There are two places where you can view rejected message counts: 

1) The top of the debug menu
2) Bottom of a log

We've added the following custom NMEA text message to the end of every log:

    $GNTXT,01,01,08,1,0,0,*44

Where:

* $GNTXT, - NMEA message type
* 01, - Number of sentences in this message
* 01, - Sentence number
* 08, - Message ID (8 = CUSTOM_NMEA_TYPE_PARSER_STATS)
* 1, - Rejected NMEA messages
* 0, - Rejected RTCM messages
* 0, - Rejected UBX messages
* *44 - NMEA CRC

So in the above example, there was one message excluded from the log.

Please give [v2.7-Dec 22](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.7RC) or later release candidate a try and let me know.

nseidle, 2023-01-19:
We've got ~500+ hours of uncorrupted logs. I'm pretty sure it's fixed in [v3.0 firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.0). Please open a new issue if anyone has any more problems.

PR #365 (2022-12-19): Release candidate
**Note:** If you are currently using version 1.x firmware you will need to load v2.x firmware using a USB connection (SD and WiFi are not supported). We have created a Windows GUI to ease this process. Please see [Updating Firmware from GUI](https://sparkfun.github.io/SparkFun_RTK_Firmware/firmware_update/#updating-firmware-from-gui) for a step by step process. Once you've updated to v2.x firmware, you will be able to load future firmware using any method (SD, WiFi, USB, or GUI).

Changes in this release:

* (Bug fix #345) Decrease UART FIFO to avoid buffer overruns
* (Bug fi [... cut]
closed 2023-01-19 by nseidle