sparkfun/SparkFun_RTK_Firmware #346: Add Device Id to log file (last 4 of BT Mac in $GNTXT message?)

tonycanike, 2022-10-05:
### Subject of the issue
When running multiple static raw collections for PPP (OPUS, etc.), it would be useful to have a device identifier in the beginning of the log file.  Perhaps put the last 4 digits of the BT MAC in a $GNTXT message?

Also see #341 and #344 

When post-processing raw data from multiple collection sessions using multiple devices, it is necessary to be able to double-check which data file came from which occupation session.  Being able to correlate hand-written field notes about the collection session with each individual log file is very important.

My hand-written notes typically contain information such as the occupied survey marker, instrument height, start time, end time, and device id (the last 4 of the BT MAC as displayed on the OLED.)

Having the device ID (last 4 of BT MAC?  Or whole BT MAC?)  in the beginning of the log file would be very helpful.

### Your workbench
Facet with v2.6 of the firmware

nseidle, 2022-10-10:
[image]

Added

tonycanike, 2022-10-10:
Wonderful, thank you!

nseidle, 2022-12-19:
In version 2.6 release.

closed 2022-12-19 by nseidle