sparkfun/SparkFun_RTK_Firmware #347: Display HAE of APC and HAE of Mark in Base Geodetic Coordinate Configuration AP page

tonycanike, 2022-10-13:
### Subject of the issue
To be completely clear to the user and prevent confusion, the Base Geodetic Coordinate Configuration page should contain the following six entry fields:
Lat  (user entered, required)
Long  (user entered, required)
HAE_APC  (Height above ellipsoid of the Antenna Phase Center)   (display only)
Antenna_Height  (user entered optional, default to zero)
APC_Offset  (use entered, optional, default to zero)
HAE_Mark  (user entered, required)

This is a simplified request related to #330 

### Expected behavior

HAE_APC is sent and retrieved from the ZED.   It is never loaded or saved to the CSV file.

HAE_Mark is loaded or saved to the CSV file.   

HAE_APC and HAE_Mark are equal if Antenna_Height and APC_Offset are both zero.

When user copies coordinates from the ZED, the HAE from the ZED is loaded into HAE_APC.  the APC_Offest and Antenna_Height are subtracted from the HAE_APC and displayed in the HAE_Mark field.

When user types into the HAE_APC field,the APC_Offest and Antenna_Height are subtracted from the HAE_APC and displayed in the HAE_Mark field.

When user types into the HAE_Mark field,the APC_Offest and Antenna_Height aare added to HAE_Mark and the sum is loaded into the HAE_APC field.

When the user saves coordinates to the CSV file, the HAE_Mark field is saved.

When the user loads coordinates from the CSV file, the HAE_Mark field is loaded from the file.  The APC_Offset and Antenna_Height are added to HAE_Mark and the sum is loaded into the HAE_APC field.

It's a toss-up which direction to calculate  (HAE_Mark --> HAE_APC, or  HAE_APC --> HAE_Mark)   when the user updates the Antenna_Height and/or APC_Offset field.  

I work in HAE_Mark coordinates so I'd say always go HAE_Mark --> HAE_APC.   That is, if Antenna_Height or APC_Offset are updated, recalculate HAE_APC.

amlago, 2022-10-13:
Good summary.
I support what was raised by tonycanike.

"I work in HAE_Mark coordinates so I'd say always go HAE_Mark --> HAE_APC. That is, if Antenna_Height or APC_Offset are updated, recalculate HAE_APC."

amlago, 2022-11-02:
One question: This topic has been added to the last RC?
It's just that I've been away from the discussions and I don't know what we agreed on.
It doesn't have a label so I'm assuming it doesn't.
This is a very important part of what was advanced in #327 .

amlago, 2022-12-09:
Hello. I understand that on these dates time is scarce.
I tried the latest version 2.6 and the height correction for the base station save has not been added.
I would ask you, if possible, to consider what is requested in this topic so that the saving of base coordinates with the height of the mark is possible.
I have a job these days where what I requested would be very useful due to the high number of bases that I will use.
Thanks.

nseidle, 2022-12-30:
WIP while I fix the common coordinates box.

[image]

nseidle, 2022-12-30:
The current RC (Dec 30) has these boxes but we are not storing them within the commonly used coordinate list. Please let me know if we are headed in the right direction.

amlago, 2022-12-31:
We are on the right track regarding the APC HaE presentation.
What I see is that it is not being dynamically updated when entering the rest of the data.

amlago, 2022-12-31:
Now we can easily access the serial menu via bluetooth, so
In the serial menu it would be interesting to replicate this way of working and add a presentation of HAE APC

amlago, 2022-12-31:
The HAE APC calculation that you present in the screenshot of your menu is correct.
That is the expected value.
We are making progress.

amlago, 2023-01-01:
In principle, it is not necessary to save the HAE APC as this should be dynamically calculated from the data entered, each time the equipment is installed on a known point.
Either we know the base coordinates or we already have them saved in the list of base points.

amlago, 2023-01-09:
Facet v2.7 01/06/23
I am testing the improvements in the latest version and I propose some variants to the input of base coordinates

1) Now we have a visualization of the HAE APC
2) The suggestion is that when we paste coordinates with the PASTE button, the APC height is pasted in the HAE APC box and not in the mark height box as it currently happens
3) The HAE Mark Height should then be calculated using the entered AH and ARP data.
HAE Mark = HAE APC - AH-ARP

This is only when we paste data with the PASTE button.
For the other situations, everything is already working fine and the HAE APC is calculated dynamically as programmed.

nseidle, 2023-01-20:
> The suggestion is that when we paste coordinates with the PASTE button, the APC height is pasted in the HAE APC box and not in the mark height box as it currently happens

Good feedback. The HAE APC is now updated if the Paste button is pressed. We're working on storing the AH and ARP to the common coordinates record.

amlago, 2023-01-20:
Ok. In relation to point 3) how would it be?

3) The HAE Mark Height should then be calculated using the entered AH and ARP data.
HAE Mark = HAE APC - AH-ARP

Well, the height to save in the file is the height of the mark. (HAE Mark)
The good thing would be that it is calculated automatically when we paste coordinates with the Paste button.

amlago, 2023-01-20:
We must be careful because there are three different procedures as indicated at the beginning.
a) I enter data of mark coordinates manually and the HAE APC height is calculated
b) Ellipsoidal coordinates are copied with paste and HAE mark is calculated to save in the file
c) If I enter ellipsoidal antenna coordinates manually, I would have to enter them as Brand and do some trick to calculate the HAE Brand in order to save it on file.
Well, we would not have enabled manual entry of HAE APC.

amlago, 2023-01-20:
As we refine the input of coordinates new ideas appear, I apologize.

An ideal option would be to have a dropdown with the 3 different tasks that I listed before. And based on that, the necessary mathematics will be done.
A 4th entry option for the dropdown would be for Facet or any device to average the coordinates in a time that I decide, for example 10 seconds, 30 seconds, and then paste it into the corresponding boxes. They would be ellipsoidal coordinates and it would be necessary to calculate the HAE Mark to save.
I don't know if it's even possible.

amlago, 2023-01-20:
A more general idea would be to put all the basic configuration options in a drop-down menu, including the survey in, and from there each option is programmed.

nseidle, 2023-03-02:
I believe the original intent of this issue has been implemented (Display both HAE of APC and HAE of Mark in Base Geodetic Coordinate Configuration AP page). I would like to close this issue out. If there are other UI or improvements, please feel free to consolidate your ideas into a new request.

closed 2023-03-02 by nseidle