sparkfun/SparkFun_RTK_Firmware #349: Dynamic display of coordinates in wifi config menu

amlago, 2022-10-18:
Facet with v2.6 RC from 10/13

1) I have found that the coordinates presented in the Wi-Fi configuration menu are not updated dynamically, nor if the web page is refreshed.
I thought that doing a page refresh would update but I have not succeeded.

2) I would like the coordinates to be updated dynamically from time to time or to be updated if I refresh the page.

nseidle, 2022-12-19:
Added to RC V2.7.

amlago, 2022-12-21:
Facet v2.7
1) The dynamic presentation of coordinates works if the equipment was previously in rover mode.
If it was in base mode, it presents the coordinates of the last base and of course they are not dynamic.
2) The idea is that they are always dynamic, whether the team was previously in rover or base mode and that the current position is presented in the header.
3) The utility is to quickly configure a base in the field and to be able to copy the coordinates presented in the header.

nseidle, 2023-01-19:
This has been implemented in [v3.0 firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.0).

closed 2023-01-19 by nseidle