sparkfun/SparkFun_RTK_Firmware #351: GNSS Failed : ZED-F9R firmware 1.30 incompatible.

imkishan96, 2022-10-22:
### Subject of the issue

'GNSS failed' msg on the display. 
this happened when I updated the firmware on the ZED-F9R from 1.21 to 1.30
downgrading it back to 1.21  fixes it.  I want to use the latest ZED firmware. is there any workaround? 

### Your workbench
* SparkFun RTK Express Plus v2.6-RC-Oct 13 2022 ( I also tried with v2.5- same issue)
* ZED-F9R firmware version 1.30

```
[1337392][D][Begin.ino:241] beginSD(): SD card detected
microSD: Online
[1337439][D][NVM.ino:90] recordSystemSettingsToFileSD(): Removing from SD: /SFE_Express_Plus_Settings_0.txt
[1337475][D][NVM.ino:109] recordSystemSettingsToFileSD(): Settings recorded to SD: /SFE_Express_Plus_Settings_0.txt
microSD: Offline
[1337496][D][NVM.ino:136] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_Express_Plus_Settings_0.txt
[1337605][D][NVM.ino:148] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_Express_Plus_Settings_0.txt
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8

setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
Display started
ZED-F9R firmware: HPS 1.30
LittleFS Started
[   215][D][esp32-hal-adc.c:210] __analogReadMilliVolts(): eFuse Vref: Supported
[   215][I][esp32-hal-adc.c:240] __analogReadMilliVolts(): ADC1: Characterized using eFuse Vref: 1079

[   222][D][Begin.ino:10] beginBoard(): Board ADC ID: 142
SparkFun RTK Express Plus v2.6-Oct 13 2022
[   253][D][NVM.ino:1057] loadProfileNumber(): Using profile #0
Reset reason: ESP_RST_SW
[   469][D][Begin.ino:241] beginSD(): SD card detected
microSD: Online
[   699][D][NVM.ino:90] recordSystemSettingsToFileSD(): Removing from SD: /SFE_Express_Plus_Settings_0.txt
[   736][D][NVM.ino:109] recordSystemSettingsToFileSD(): Settings recorded to SD: /SFE_Express_Plus_Settings_0.txt
[   749][D][NVM.ino:136] recordSystemSettingsToFileLFS(): Removing LittleFS: /SFE_Express_Plus_Settings_0.txt
[   864][D][NVM.ino:148] recordSystemSettingsToFileLFS(): Settings recorded to LittleFS: /SFE_Express_Plus_Settings_0.txt
[  1084][E][vfs_api.cpp:102] open(): /littlefs/SFE_Express_Plus_Settings_1.txt does not exist, no permits for creation
[  1086][D][NVM.ino:375] loadSystemSettingsFromFileSD(): File /SFE_Express_Plus_Settings_1.txt not found
[  1096][E][vfs_api.cpp:102] open(): /littlefs/SFE_Express_Plus_Settings_2.txt does not exist, no permits for creation
[  1105][D][NVM.ino:375] loadSystemSettingsFromFileSD(): File /SFE_Express_Plus_Settings_2.txt not found
[  1115][E][vfs_api.cpp:102] open(): /littlefs/SFE_Express_Plus_Settings_3.txt does not exist, no permits for creation
[  1125][D][NVM.ino:375] loadSystemSettingsFromFileSD(): File /SFE_Express_Plus_Settings_3.txt not found
[  1135][E][vfs_api.cpp:102] open(): /littlefs/SFE_Express_Plus_Settings_4.txt does not exist, no permits for creation
[  1144][D][NVM.ino:375] loadSystemSettingsFromFileSD(): File /SFE_Express_Plus_Settings_4.txt not found
[  1155][E][vfs_api.cpp:102] open(): /littlefs/SFE_Express_Plus_Settings_5.txt does not exist, no permits for creation
[  1164][D][NVM.ino:375] loadSystemSettingsFromFileSD(): File /SFE_Express_Plus_Settings_5.txt not found
[  1174][E][vfs_api.cpp:102] open(): /littlefs/SFE_Express_Plus_Settings_6.txt does not exist, no permits for creation
[  1183][D][NVM.ino:375] loadSystemSettingsFromFileSD(): File /SFE_Express_Plus_Settings_6.txt not found
[  1194][E][vfs_api.cpp:102] open(): /littlefs/SFE_Express_Plus_Settings_7.txt does not exist, no permits for creation
[  1203][D][NVM.ino:375] loadSystemSettingsFromFileSD(): File /SFE_Express_Plus_Settings_7.txt not found
Profile 'Profile1' loaded
MAX17048 configuration complete
Batt (100%): Voltage: 4.19V Discharging: 0.00%/hr Green
Set rate failed
Get Constellation failed
Set Constellation failed
Get Constellation failed
Set Constellation failed
Get Constellation failed
Set Constellation failed
Get Constellation failed
Set Constellation failed
Get Constellation failed
Set Constellation failed
Get Constellation failed
Set Constellation failed
Updating UART1 rate
Updating UART2 rate
Module failed initial config.
Module failed to save.
Failed to configure GNSS module. Trying again.
getNavigationFrequency: zero measRate!
Set rate failed
Get Constellation failed
Set Constellation failed
Get Constellation failed
Set Constellation failed
Get Constellation failed
Set Constellation failed
Get Constellation failed
Set Constellation failed
Get Constellation failed
Set Constellation failed
Get Constellation failed
Set Constellation failed
Module failed initial config.
Module failed to save.
Failed to configure GNSS module.
Accelerometer configuration complete
[  6186][D][menuPP.ino:844] beginLBand(): L-Band not detected
[  6195][D][RTK_Surveyor.ino:610] setup(): Boot time: 6195
```

### Steps to reproduce
upgrade ZED-F9R to firmware version 1.30 ( tried with U-center as well as RTK_u-blox_Update_GUI same results)

### Expected behavior
It should have been initialized without any error. 

### Actual behavior
GNSS Failed and digging deeper seems like an I2c issue. The first indicator was 'Set rate failed' and followed by other errors.

nseidle, 2022-10-26:
Hi @imkishan96 - Thanks for reporting and sorry for the hassle. According to their [v1.30 release notes](https://content.u-blox.com/sites/default/files/documents/ZED-F9R-03B_FW1.00HPS1.30_RN_UBX-22035201.pdf) u-blox has finally started to drop the config messages we've been relying on for many years.

[image]
[image]

(sorry for the bad spacing of the image)

It looks as though UBX-CFG-RATE, as well as many other messages, have been removed. Our system relies on these messages for the moment. We'll need to significantly refactor things to support the 'get/set val' method.

For now, my apologies, but you will need to use ZED-F9R firmware v1.21 or below.

nseidle, 2022-10-28:
Hi - We were able to change over to the new configuration method. Please give the [latest RC](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.6RC) (Oct 28th or more recent) a try. I've tested on ZED-F9R firmware:
* v1.20
* v1.21
* v1.30

with success

PR #365 (2022-12-19): Release candidate
**Note:** If you are currently using version 1.x firmware you will need to load v2.x firmware using a USB connection (SD and WiFi are not supported). We have created a Windows GUI to ease this process. Please see [Updating Firmware from GUI](https://sparkfun.github.io/SparkFun_RTK_Firmware/firmware_update/#updating-firmware-from-gui) for a step by step process. Once you've updated to v2.x firmware, you will be able to load future firmware using any method (SD, WiFi, USB, or GUI).

Changes in this release:

* (Bug fix #345) Decrease UART FIFO to avoid buffer overruns
* (Bug fi [... cut]
closed 2022-12-19 by nseidle