sparkfun/SparkFun_RTK_Firmware #352: Connecting RTK Facet v1.10 with QField App/QGIS

stewielandformprojects, 2022-10-26:
Hello

I have been having a number of issues with setting up the RTK Facet v1.10 for a small business in New Zealand that I work for. I have never used an RTK before so please forgive my ignorance. 

We use QGIS and QField for all of our mapping. We are unable to use SW Maps or the other software specified on the website as this is in WGS84 and we require our CRS to be EPSG2193

I have scoured the internet on forums, youtube and the sparkfun website/forums for any information on how to connect the RTK to QField but I have had no luck. 

We do have one of our engineers who has an older model of the RTK who has given me the following instructions 
- Change the mock location on the android phone to NTRIP
- Connect the RTK to NTRIP with the same settings he had (with the exception of the login details from LINZ)
- Open QField and ensure that the device used is '"internal device" which is actually the RTK.... as the mock location is via NTRIP. 

From this, it should work but it doesn't seem to be working and has horrible accuracy. 

I am hopelessly out of ideas and under pressure to get this connected and working 

I have gone to this website https://docs.sparkfun.com/SparkFun_RTK_Firmware/gis_software/#sw-maps) It has advised that "if there is other GIS software that you'd like to see configuration information about, please open an issue on the [RTK Firmware repo](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues) and we'll add it."

Well!!! Here I am opening an issue here and hoping someone can help.

tonycanike, 2022-10-26:
I'm not a QField user, but I have used a couple different s/w packages to collect data in the field from the Facet, so I'll try to help. I have also used QGIS  to make maps from data I've collected.

It would be great to have a step-by-step set of instructions for QField in the SparkFun documents, so I'd like to see you succeed.

Here are things I would try:

1.  Upgrade the firmware on your Facet.  V1.10 is pretty old.   V2.5 is the most recent release, and many of us who are following along here on GitHub are beta testing candidates for v2.6.  Maybe features have been added and bugs fixed.
2.   Get NTRIP and SWMaps working first.  Setting up SWMaps is well documented.  There are many people who can help you configure SWMaps and get it working. This will debug many issues and be a good learning step.  The first couple times getting everything configured and connected can be confusing. There are a number of subtle indicators on the OLED display of the Facet to learn how to read.  Getting SWMaps working will be a good accomplishment and you will learn a lot about how it all works.
3. I recently posted on the SparkFun user forum a little guidance for someone trying to get SWMaps and NTRIP working.  Some of the pointers I made about the indicators on the Facet display seemed to help the person a lot.   https://forum.sparkfun.com/viewtopic.php?f=116&t=58481
4.  Also post your question on the user forum asking for help with QField - there are more people reading the forum and maybe there's a user who has already done exactly what you need to do.
5.  QGIS can transform from WGS84 data to just about any datum and projection.  QGIS can read WGS84 data collected in SWMaps and transform to your projection.  So you could gather data with SWMaps and transform it as another intermediate step to get everything working and test your system.   It would also show intermediate progress to your bosses :)    Though I totally understand the need to use QField.  I use collection s/w (Carlson SurvPC) mainly because my colleagues and I use Carlson desktop software.  Once it's working it's very convenient and powerful.
6.  I didn't understand your three steps about using a mock device&location on the Android, so I looked it up.  The first result from my google search led me to a StackExchange discussion that (a) **mock locations are no longer needed** needed and (b) QField can now connect directly to Bluetooth GPS/GNSS receivers.     I would suggest reading https://gis.stackexchange.com/questions/362463/using-qfield-to-connect-to-external-bluetooth-gps       Scroll down to the "QField 1.8.0 there's now a native support built into QField, i.e. you can connect to Buetooth-paired devices from within QField" comments.
7.    If you can connect QField directly to the Facet via Bluetooth, you'll still need to feed the NTRIP datastream to the Facet.  There are many different ways of doing this.
8. Please take notes and send them in;so the SparkFun documents can be updated for future users!

Tony.

gdt, 2022-10-26:
I have managed to make it work.  As @tonycanike says, upgrade the firmware.  This is absolutely required.  Then, configure the facet to do NTRIP over wifi, and then connect qfield via bluetooth.  QField does not have NTRIP support.  That's pretty much it.   I am overdue for documenting this.

As for datums:
  - you cannot assume that your RTK output data is in WGS84.  It depends on the frame of your reference station
  - WGS84 is an ensemble but your data is not in the ensemble; no actual data ever is
  - proj mistransforms WGS84 to NAD83 because of ensemble confusion, and hence qgis and I think qfield do too

however the datum issues are not about the facet; the sparkfun devices just provide the F9P output, and all the processing belongs in the data collector.

gdt, 2022-10-26:
The mock location scheme is simply alternate plumbing to get data to an app which does not support obtaining NMEA from bluetooth or TCP.   An app which does do that gets it, and injects coordinates to the OS via a debugging interface, and then all apps simply see those from the standard "get location" API.   This is useful for causing everything to use the high-accuracy coordinates.

And, I suspect pat of the issue is that the OP may not have a clear answer to "Where is your RTK reference data coming from, and how are you getting it into the Facet?  What datum is it in?".

I continue to find the "data from the Facet is in WGS84" boggling as a general belief.  It's just not generally true.

stewielandformprojects, 2022-11-14:
Thanks for your advice Tony, 

I have made a couple of posts on the QField/QGIS forums on the internet but yet to get a response. 

I have managed to get the RTK to work with QField by completing the following;

- Opening NTRIP client and connecting to the RTK, leaving this running in the background.
- Mock location turned on
- Opening QField and setting the device as ''internal device'' (which is the mock location and thus the RTK) 

I did not end up trying SW Maps. I only work a limited number of hours a week so needed to priortise my time to getting QField to work. 

I did try to do what you said in #7 but this did not seem to work for me. It does say on a couple of those links that mock location is still required if you do need to connect with NTRIP Client. If you do have any advice on how to feed the NTRIP DataStream to the facet that would be awesome to know nonetheless. 

One thing to note as I see there are comments about my mention of WGS84 and I don't think I explained this well enough as to why we cannot use this in the field. The most important thing we (hope) to use the RTK for is obtaining R.L's in **real time** for the points and line features we create. These levels need to be in NZVD2016. So I need the map to be in EPSG2193 so I could convert the Ellipsoid height (NZGD2000) into Orthometric using a geoid file in Qfields "vertical grid shift" function. I am yet to get this all set up but hoping its going to work. Trying again to scour the internet for advice on this.

To answer gdt's comment on not having a clear answer to "Where is your RTK reference data coming from, and how are you getting it into the Facet? What datum is it in?". You are right. I don't have an idea. I am completely self taught so don't know the minute details on how everything works. 

- What do you mean by frame of your reference station
- What do you mean by project mistransforms and ensemble confusion?

gdt, 2022-11-15:
Using an NTRIP client and mock locations is a reasonable method. The other method is to hotspot for the Facet and to configure it to do NTRIP over wifi, and then use bluetooth from QField.   That's what I managed to do earlier.   I have not tried it on recent ESP loads; that's on my TODO list.

As for the other things, it's sort of off topic for this bug, but underdocumented in RTK Express/etc. land compared to what would be ideal.  I think that you are going to have to understand how RTK works and reference frames in order to succeed, but I'll give a few clues.

RTK is fundamentally relative positioning from the reference station to you.   The F9P computes a vector, based on carrier-phase observations at the reference and your receiver.  It then takes the reference station's coordinates (which are in the RTCM3 stream) and adds the vector to get your coordinates.    For me, the reference station coordinates are in "NAD83(2011) epoch 2010.0" because that's the US National Spatial Reference System.   These are not the same as the WGS84 coordinates of the same station, differing by a meter or two.   So the coordinates I get for a point are in NAD83, not WGS84.  That's good, because NAD83 is (mostly) a crust-fixed frame, and NAD83 coordinates are stable over time (except in CA, insert usual joke).  WGS84 is not crust-fixed, and points near me have 2-3cm/y velocities.

You are in NZ, not US, but the situation as far as I can tell is the same.  [NZGD2000](https://www.linz.govt.nz/guidance/geodetic-system/coordinate-systems-used-new-zealand/geodetic-datums/new-zealand-geodetic-datum-2000-nzgd2000) is a 3D datum that is plate fixed.   Except that it has a deformation model, because not all of NZ moves at the same rate (similar to US west coast).

When you do RTK, you are getting reference data from someplace.  That's what you configure in NTRIP.   That data is feeding you carrier phase measurements and coordinates *in some datum* of the place where those measurements were made.  So when the F9P does the math, the output coordinates are in the datum of the reference station.   Your reference station service should document this.   If you are using a NZ government source, I would be almost shocked if it were not in NZGD2000.    And, you should be careful to use reference stations that have very little motion between them and where you are; RTK across a fault line is great for monitoring the fault and not so great for surveying.

Next, you have to tell QField what datum your coordinates are in.   This is an open ticket, where I've tried to explain the need, and there are some responses that to me feel off point: https://github.com/opengisch/QField/issues/2855.

The next step is that when QField deals with transformations, it has to get them right.   proj has null transforms from WGS84 to NAD83, which in a twisted sense can be said to be right, but is in practice in the modern world simply wrong.   I have no idea what proj has for NZGD2000 to WGS84(ensemble).    And finally, the idea of WGS84 as an ensemble is problematic. 

If this doesn't make sense enough for you to follow or tell me where I'm confused, you may need to call a geodesy hotline.  This is very complicated stuff.

tonycanike, 2022-11-15:
What gdt said about RTK.  I was simply going to say contact the network operator and find out what datum their reference stations are using, as that's what you'll get out of your rover.

nseidle, 2022-12-23:
QField works fine for me. I'll write up docs next week hopefully.

The only gotcha I found was that QField parser gives up if anything other than NMEA messages are detected. So be sure to use the default 5 NMEA messages during BT SPP and you'll be fine.

gdt, 2022-12-24:
Great to hear.   I am still intending to get that setup, and I will try to file bugs in QField about non-NMEA.

nseidle, 2022-12-27:
[Docs have been added](https://docs.sparkfun.com/SparkFun_RTK_Firmware/gis_software/#qfield-qgis). Please have a look and let me know if anything doesn't make sense.

closed 2022-12-27 by nseidle