sparkfun/SparkFun_RTK_Firmware #353: No SD card log after reset

Eric-FR, 2022-10-26:
RTK Express with Oct26 RC (but already observed with a previous RC).

After Wifi configuration while asking for 'exit', the device is restarting as expected but no data are recorded on the SD card. One can see on the device display that SD card is not detected. If looking at serial, one can see error messages related to the SD card:

[image]

If we switch off and start again, it's fine again. SD card is shown on the display and data are correctly logged.

Same happened after firmware update.

tonycanike, 2022-10-27:
I had a similar--but probably different--problem today with the 26-Oct RC   I put my Facet on a pole, used the setup button to enter WiFi Config mode, raised it high to get over a bush, measured the height, and used the WiFi config to setup logging.   I saved the config and then exit&reset via the buttons at the bottom of the WiFi config pages.  The webpage said the RTK device was rebooting. Couldn't see the OLED with the sun and the height of the Facet.

Hours later, I lower the Facet and see the OLED is displaying that it's still in WiFi config mode.  Who knows what happened,  Maybe I got confused.  But that's the whole point -- who knows what's going on with the Facet??  I'd like to know it's logging before I waste half a day only to find out it wasn't.

Gosh I've spent 100s of hours since May trying to get these things to write a decent log file.  

We really need a way to know it's logging when we can't see the OLED display.

I opened this issue:   #354 
And this one:     https://github.com/sparkfun/SparkFun_RTK_Facet/issues/10

nseidle, 2022-12-19:
While we can't directly reproduce, we've added the following:

* Gracefully recover if SD card is removed/detected. If the SD card is either removed or is electrically 'lost' we can detect the loss and restart logging successfully. In other words (and it's not recommended but) you can now hot swap SD cards.
* Add additional SPI/SD interface flushing. This should allow partially frozen SD cards to correctly restart without a system reset.
* Allow the system to reset if a card is detected but will not initialize. This was your problem: the SD card was detected as present but it would not start talking. We have the new flushing in place that may fix your original problem, but if it doesn't, you can now force the system to reset. This option has been added under the logging menu (option '6').

Please give the RC a try and let me know if you continue to see this problem.

nseidle, 2023-01-19:
I believe we fixed this in [v3.0 firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.0). Please let us know if you continue to see this issue.

closed 2023-01-19 by nseidle