sparkfun/SparkFun_RTK_Firmware #354: Be able to stop logging when the Facet is high in the air.

tonycanike, 2022-10-27:
### Subject of the issue
It is not possible to turn off logging (or even turn off the device to stop logging) when the device is out of reach.

### Your workbench
26-October-2022 RC

### Description of Issue
While performing static logging, it is sometimes necessary to have a GNSS receiver 2.5m or more above the ground to clear low-level vegetation or other obstacles.  In this situation, it is not possible to stop logging without moving the Facet, necessitating manual trimming of the RINEX file.

I can start a new log file by have the Facet in WiFi Config mode before I raise it up in the air.

I know I'm asking a lot and this may never get implemented.  But this is one reason to implement some form of command channel and protocol, perhaps over Bluetooth.   Even if it started out by being just one command that put the device in WiFi config mode :)

My observations are that most commercial survey GNSS receivers have some sort of command protocol that can be used to stop and start logging, turn internal radios on and off, set the coordinates of a RTK base unit, etc.

Given that there is this statement "If you are a big fan of SurvPC please contact your sales rep and ask them to include SparkFun products in their Manufacturer drop-down list." in the manual at https://docs.sparkfun.com/SparkFun_RTK_Firmware/gis_software/#survpc  I believe there would need to be such a command channel and protocol in place for s/w developers to even consider doing this.

amlago, 2022-11-02:
What Tony proposes is something to be implemented without a doubt.
Having access to WIFI settings at any time without having to touch a button would be fantastic.
How to implement it I don't know, but it could be through bluetooth as indicated before.
This is very good in particular when we want to access a device that is in base mode and most likely bluetooth is not in use.
Although the bluetooth connection is only as a trigger for the wifi configuration to open.

amlago, 2022-12-19:
Going back to the point.
Is it possible via bluetooth to tell the Facet to go into WIFI Config mode?
It would be necessary to generate an Android app that connects to the Facet via Bluetooth and acts as an intermediary with the WIFI Config page.
Is this possible?

nseidle, 2022-12-23:
It would be feasible to pipe the entire serial terminal interface over Bluetooth. You'd need to use a Bluetooth terminal app like [Android Serial Bluetooth Terminal](https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en_US&gl=US&pli=1), but then you could enter any menu and do any control (really) at any time (probably). I didn't think many users would want to type letters/numbers into a terminal in the field like this. But we could open it up.

Alternatively, could we just leave WiFi on? Perhaps. This leads to a few problems:

* Authentication. The purpose of requiring a user to press the button until you get to 'Config' is to prevent other people from connecting to the device. We'd need to add a login layer to the WiFi config page.
* I'm not entirely sure we can log while the AP Config section of the code is running. With the addition of the file manager, only one part of the code can access the SD card without things going wrong. We will likely have to turn off logging while a user is connected over WiFi. Is that ok?

I'll work on the Bluetooth approach first.

tonycanike, 2022-12-23:
Interesting! 

I think a documented behavior to cleanly close the log file when the WiFi AP is started is fine. 

Being able to start the WiFi AP over BT would be great. 

I'm now configuring the F9P via Bluetooth, so how would making the serial menu available over BT affect that?  Will the F9P commands still go to the F9P? 

SurvPC supports configuring F9P devices, there's a drop-down built in for the F9P. I'm learning what it can do. 

I have recently started using SurvPC to set the coordinates of the fixed base Facet.  I first use the WiFi AP to configure everything and put the Facet into base mode.  Then I use SurvPC to change the base coordinates.

nseidle, 2022-12-27:
I'm not going to lie, this was a lot of fun to implement! A fun but straightforward challenge. And I think it's a really handy feature.

'Bluetooth Echo' mode is everything you get with a wired USB connection, without the wire. 

To enter the 'Bluetooth Echo' mode:
* Open a bluetooth serial app such as [Serial Bluetooth Terminal for Android](https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en_US&gl=US&pli=1)
* Connect to the RTK Device
* Send the string +++

To exit the 'Bluetooth Echo' mode:
* Disconnect Bluetooth or select 'b' from the configuration menu

Docs are here: https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_with_bluetooth/

Once you've connected over Bluetooth, a ton of NMEA data will be scrolling past. Simply send +++ to stop the NMEA data and enter Bluetooth Echo mode.

[image]

Once connected, anything you send will be received by the system. This makes using the config menu easy.

While in Bluetooth echo mode, Bluetooth send/receive gets priority. The means if you've *also* got a USB connection, anything you type won't get through. Any local serial sent from the RTK to a computer is printed, but anything sent from the computer to the RTK device is ignored.

Please give it a whirl and let me know what you think.

amlago, 2022-12-28:
Facet v2.7 12/26
A great advance for not touching the power button.

1) The connection of the Facet to the terminal was very easy and fast. The messages appear passing very quickly.
2) The sad thing is that the sending of messages at high speed does not stop when entering +++.
  So I can't get into the setup menu.

What am I doing wrong?

amlago, 2022-12-28:
I am using two cell phones, one with Android 9 and the other with Android 12.

amlago, 2022-12-28:
Could it be that version 2.7 has not been uploaded where bluettoth serial communication was enabled with the +++ command?
With version 2.7 of 12/26 I have not been able to enter the configuration in any way.
Thank you

nseidle, 2022-12-28:
Who has two thumbs and forgot to update the release binary? This guy. That joke will probably not translate :)

Sorry! I forgot to update the binary. Updated now. Please test [v2.7-Dec 28](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.7RC).

amlago, 2022-12-28:
ha ha, it was understood yes.
I try the new version and comment the results.
regards

amlago, 2022-12-28:
It works excellent. very good work.
An additional comment:
You can add texts to send as shortcuts on the M1----M7 buttons and even add more shortcut buttons in the application settings.
You can even change the name of the shortcut for example from M1 to Menu.
Claps for you Nathan.

amlago, 2022-12-28:
I upload sample capture
[image]

nseidle, 2023-01-19:
This has been implemented in [v3.0 firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.0).

closed 2023-01-19 by nseidle