sparkfun/SparkFun_RTK_Firmware #359: Problem with correction transmission through BT

Eric-FR, 2022-11-05:
RTK Express

Connected to Android smartphone with BT using whatever Lefebure TRIP, BT_GNSS os SW_Maps.

I discovered that age of differential was unstable and was quite often increasing until the limit of 60 s, loosing RTK correction. Looking at previous records, this is new (fall 22).

Last July:
$GNTXT,01,01,03,v2.2-Jun  8 2022*6C
$GNTXT,01,01,04,HPG 1.30*36

[image]

Last month:
$GNTXT,01,01,03,v2.6-Sep 30 2022*64
$GNTXT,01,01,04,HPG 1.32*34

Same setup, same road at the beginning.

[image]

Looking at other files, the problem should have started with:
$GNTXT,01,01,03,v2.5-Sep 21 2022*67
$GNTXT,01,01,04,HPG 1.32*34

In Lefebure, one can clear see that Lefebure is well receiving correction data.

I made a cross check with my home made logger that is also using BT for corrections. It doesn't suffer from such lags.

Eric-FR, 2022-11-05:
Typically, what I observed with Lefebure (antenna on top of the car without moving, in urban area with good GNSS and 4G reception):

[image]
[image]
[image]
[image]

nseidle, 2023-03-02:
@Eric-FR - Thank you for reporting! My apologies for the delay. 

I have attempted to replicate this issue and I'm not yet able to. Below is a capture to replicate your ~200k of data received and RTK lost.

[image]

I'm able to have both solid reception within Lefebure and on the unit (RTK fix the entire time). I am not driving anywhere but if driving affects the reception of NTRIP corrections, there's nothing really that we can do.

My tests were done with v3.1 firmware. I could go back in time to further try to replicate the issue when you reported it on v2.6 but since it's working now, I'm hesitant to do so.

I'm going to close this as solved for now. Please let me know if you see anything like this again and we'll keep digging.

closed 2023-03-02 by nseidle