sparkfun/SparkFun_RTK_Firmware #360: Rover ZED-9FR Stops recording after 2.5 days thought the rover is configurated to record for a longer period of time

IngridBerru, 2022-11-09:
Hello All,

I have an Express Plus Rover with Firmware version 2.5 and ZED-9FR version HPS 1.21. I configurated the rover using TeraTerm to have a large testing time (See below ) because the rover will be used to monitor a landslide, however, the rover stops recording the data after 2.5 days, though the rover was configurated to have a "Log time" of 438000 minutes (10 months) (See below). I have other rovers which have the same configuration however I do not have any issues with them and these rovers recorded the data for 7 days. 

I'm using an SD card of 32 GB and the rover's log data is 70 MB for 2 days ( I configurated the units to record every minute -  NMEA and RMX messages are recording every 60 seconds). I do not know what is happing to the rover and any help will be highly appreciated. Thank you so much in advance.

TERA TERM CONFIGURATION
Menu: GNSS Menu
1) Set measurement rate in Hz: 1.0000
2) Set measurement rate in seconds between measurements: 1.00
3) Set dynamic model: Portable
4) Set Constellations
5) Toggle NTRIP Client: Disabled
x) Exit
Menu: Logging Menu
microSD card is online
1) Log to microSD: Enabled
2) Set max logging time: 438000 minutes
3) Set max log length: 2880 minutes
4) Write Marks_date.csv file to microSD: Disabled
x) Exit
Menu: System Menu
GNSS: Online - ZED-F9R firmware: HPS 1.21
SIV: 0, HPA (m): 1050.048, Lat: 53.52226070, Lon: -113.47257052, Altitude (m): 646.0
Display: Online
Accelerometer: Online
Fuel Gauge: Online - Batt (98%) / Voltage: 4.18V
microSD: Online
Bluetooth (9562): Online
WiFi MAC Address: 34:AB:95:5C:95:60
System Uptime: 0 00:14:37.141 (Resets: 0)
f) Display microSD Files
d) Configure Debug
z) Set time zone offset: 00:00:00
b) Set Bluetooth Mode: Classic
c) Enable/disable WiFi NMEA client (connect to phone): Disabled
n) Enable/disable WiFi NMEA server: Disabled
r) Reset all settings to default
B) Switch to Base mode
R) Switch to Rover mode
W) Switch to WiFi Config mode
S) Shut down
x) Exit

[image]

Best regards,
Ingrid Berru

nseidle, 2022-11-29:
This issue was originally posted here: https://forum.sparkfun.com/viewtopic.php?f=116&t=58574

This repo is the best place for the conversation. 

```
Hi Ingrid,

My apologies for the slow reply! I'm sorry it's not working as expected.

You have three units you are using. Is there any difference between them? For example, are they all Express Plus units?

Have you tried swapping the microSD card from a unit that works to the unit that does not? I would like to rule out the microSD card as the problem.

My suspicion is the microSD card. If it stops communicating correctly with the ESP32 or becomes corrupt, the system does not have a method for power cycling or resetting the SD card. A power cycle of the unit might be the only solution. 

I will look into other software methods for reopening communication with the microSD card if it fails.

nseidle, 2022-12-19:
As the firmware is currently written, if logging is enabled then we attempt to start communication with a microSD card. We issue the CMD0 command which is a software reset command up to 5 times. Testing with all the microSD cards we can find (about 10 different variants) if a card is present, the card will respond.

It seems your SD card is not responding after 2.5 days. The only other thing we could do is power cycle the card. Currently, our hardware does not support this but we can add it to a future revision. 

In the mean time, can you give us feedback on changing the microSD cards to a different manufacturer and how that affects logging over more than 3 days?

nseidle, 2022-12-19:
This may be related to #353.

While we can't directly reproduce, we've added the following:

* Gracefully recover if SD card is removed/detected. If the SD card is either removed or is electrically 'lost' we can detect the loss and restart logging successfully. In other words (and it's not recommended but) you can now hot swap SD cards.
* Add additional SPI/SD interface flushing. This should allow partially frozen SD cards to correctly restart without a system reset.
* Allow the system to reset if a card is detected but will not initialize. We have the new flushing in place that may fix your original problem, but if it doesn't, you can now force the system to reset. This option has been added under the logging menu (option '6').

Please give the RC a try and let me know if you continue to see this problem.

nseidle, 2023-01-19:
We've had a few units operating for 5+ days without problems using the new release. Please try [v3.0 firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.0) and let us know if you still see this issue.

closed 2023-01-19 by nseidle