sparkfun/SparkFun_RTK_Firmware #362: Slowness in sending ntrip corrections v2.6 28/10/2022

amlago, 2022-12-04:
Facet v2.6 10/28/2022

With the latest version of update v2.6 on 10/28/2022, the speed of sending ntrip corrections slowed down and I have had occasions where it stopped completely.
The work area is the same and the quality of the Wi-Fi connection is identical.
I went back to version 2.6 on 10/13/2022 and everything was fixed.
The sending speed is normal and never stops.
I leave the concern raised.
Cheers

nseidle, 2022-12-19:
> the speed of sending ntrip corrections slowed down

Help me understand which part of the system you are referring to: Are you saying the NTRIP corrections over WiFi to the RTK device are intermittent? Are you using Bluetooth at the same time to connect to the device?

Are you connecting WiFi from RTK to your phone or a home AP?

amlago, 2022-12-19:
Clarification:

1) Test with Sparkfun facet v2.6 10/28.

2) Sending of ntrip corrections to Rtk2go via WIFI of the cell phone.
It is the same base that I have used many times, and the Internet connection through the cell phone's Wi-Fi works very well in the work area.

3) I tried several times and the same thing happened to me.
The issuance of corrections slowed down and sometimes stopped.

4) I'm only sending ntrip fixes and I'm not connected via bluetooth.

5) I went back to version 2.6 on 10/13 and it worked fine for me.
I have continued working with this version 2.6 13/10

amlago, 2022-12-20:
1) I just tried version 2.7 12/20 and sending ntrip fixes keeps failing. It connects to wifi and starts issuing ntrip fixes.
2) At the beginning the rtcm count is normal, after a few tens of seconds it slows down and sometimes stops altogether.
3) Today for the second time an additional thing happened to me. When trying to exit the base mode the button did not respond and did not go through the mode.
I left it pressed for a while and the equipment turned off.
4) Again I went back to version 2.6 10/13 and everything works fine.

It would be necessary to review the differences between v2.6 10/13 and the current ones.
regards

amlago, 2022-12-21:
Facet v2.7
Continuing with the tests, I detected something strange.

1) If the Facet is detecting few satellites, the speed of sending corrections is normal.
2) When the number of visible satellites increases, then the speed of sending ntrip corrections slows down and stops and what I described before happens.
3) I tried without saving the file to the SD and the same thing happens.

4) When I return to v2.6 on 10/13, everything is solved as I have already done on several occasions.

amlago, 2022-12-23:
Fault update. Facet v2.7 12/22
Regarding what was stated before with version 2.7 12/22, the rtcm count and slowdown failures continue.
When trying to use the Facet button it seems not to respond. It's like the Facet hangs.

Going back to v2.6 10/13 everything works fine.

nseidle, 2022-12-23:
This is very good analysis! Thank you. We'll have a look.

nseidle, 2022-12-23:
Thank you for narrowing down the difference between Oct 13 firmware and Dec 22 firmware. We made a [big change](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/356) to how the u-blox module is configured. I suspect this is the problem but I have not found the problem yet.

Have you turned on any other messages other than the default 5 NMEA? You are not using an SD card so that's not it. 

I am able to push RTCM data to RTK2Go over my cell phones Hotspot:

[image]

I don't see a slow down or any change. I'm having a hard time replicating.

[image]

Just to be sure, what version of firmware is on your ZED-F9P? I'm on v1.32. I don't think it matters but want to be sure.

The RTCM comes in over I2C from the ZED and gets pushed to WiFi.

amlago, 2022-12-24:
Regarding your questions, let me clarify the following.

1) I am recording some additional message to the 5 basic messages.
I record 7 or 8 Nmea and 2 raw messages.
2) I have firmware 1.30 on the ZED-F9P

With this configuration and with the old version it works very well.

amlago, 2022-12-24:
Another strange thing.
1) Looking at the summary of the Rtk2go station on the Rtk2go page it seems that the rtcm messages are arriving without problems to them.
2) It may be that the Facet is sending messages but it is hanging and that causes the slow count on the screen. And the slow response of the button.

I'll pass this data on in case it's relevant to your analysis.

amlago, 2022-12-24:
Hello, I try again with v2.7
1) Update to Zed-f9p 1.32
2) Remove the SD
3) Leave only the 5 basic Nmea messages

Results

1) The same. It starts with normal rtcm countdown and after a few seconds it starts to slow down the countdown speed on screen, until the countdown seems to stop and the power button becomes unresponsive.
To turn it off you have to leave it pressed for 15 seconds.
2) I reiterate that if the number of satellites is low, the count on the screen goes somewhat better, I don't know why.

As I have told you, I return to v2.6 10/13 and everything is fine.

nseidle, 2022-12-26:
Thanks again for the great testing! I will continue to try to replicate the problem.

nseidle, 2022-12-26:
Please give v2.7 Dec 26th a try: https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v2.7RC

There may have been an issue setting the ZED-F9P's port settings correctly.

nseidle, 2022-12-26:
> I reiterate that if the number of satellites is low, the count on the screen goes somewhat better, I don't know why.

I regularly see ~30 sattellites. What number of satellites (approximately) works and what number causes the newer firmware to stop sending RTCM packets to RTK2Go?

amlago, 2022-12-26:
Great.
Facet v2.7 from 12/26
Without SD and only with the 5 basic messages.

The v2.7 of 12/26 fixed the emission of ntrip corrections by WIFI.

1) The rtcm count on the screen advances at the rate of one unit per second and does not slow down.
2) The equipment does not hang either and the power button works very well.
3) I am issuing ntrip corrections and around 30 satellites can be seen.

So this seems to be fixed.

amlago, 2022-12-26:
update
1) Add the SD and still all good
2) Add extra messages and still all good

so eureka

nseidle, 2022-12-27:
Very good! Glad to hear it.

closed 2022-12-27 by nseidle