sparkfun/SparkFun_RTK_Firmware #364: Allow assisted start via cellular or WiFi

nseidle, 2022-12-13:
From [the review](https://www.sparkfun.com/products/18380) by customer #24286.

> There's not an obviously way to "hot start" the unit at a known location, 

We have the ability to [do a very fast assisted start](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/main/examples/AssistNow) using the u-blox AssistNow system. We just need the ability to push the acquired data to the unit. This can be done over cellular or WiFi. The trick will be the app on the phone. Perhaps we can ask SW Maps to integrate the feature someday.

gdt, 2022-12-13:
Is it possible to have an AssistNow client over wifi, similar to how it gets NTRIP?

nseidle, 2022-12-14:
Yep - AssistNow over WiFi is very straightforward. My concern is getting lock in the field. Perhaps we require use to run WiFi hotspot?

gdt, 2022-12-14:
Require is too strong, but I would say that any feature you can add, that's reasonable, to help is progress, and people can choose a system configuration based on what works.   If AssistNow can easily work over wifi, without messing up the RAM budget, then it seems good to add it, even though there will be some people that  won't be helped.

The other path is something else fetching, and for that the important thing is to document what the interface is.   If it's the case that AssistNow data can be fetched, and then sent over Bluetooth SPP, and this will just be fed to the F9P, and then one could start sending RTCM3, then it would be good to just say that, and various programs one runs on phones etc. could add that as a feature.

Personally, I'm ok with the wifi approach as that's necessary to use something like QField or Vespucci, as I try very hard to avoid non-Free software as someting I don't have any long-term ability to ensure access to or to improve.

Also, I'm not sure how important this truly is.   The F9P gets ephemeris pretty quickly with a good sky view.  And without a good sky view, RTK is iffy.   So I view this as about quick autonomous position from poweron, not about RTK, and while it's always better that things work better, it seems to be for a use case that's a bit off the main path.   I suppose it would be nice to have RTK lock in 15s rather than 75, though.

nseidle, 2023-03-02:
Thanks @gdt and @tonycanike! Your thoughts are greatly appreciated.

I'm going to close this as it's a lot of work for a few seconds of lock time savings, maybe. And it has a monthly subscription that adds to the diminishing returns.

> My impression is that the SparkFun F9P units keep standby power to the F9P and thus they don't lose ephemeris.

Yes, we keep backup power applied to the ZED-F9P so that it can warm start as quickly as possible.

> Would this speed up a base surveying itself in?

Generally, maybe? Survey-in requires a HPA of less than 1m before it starts (a SparkFun applied requirement), then 60s of wait (u-blox recommended time). So if your unit has been on for more than a minute or two during site setup, I suspect its HPA would be better than 1.0m, and the survey in would start immediately. An assisted start would remove the couple minute initial wait, but the 60s survey-in is generally a requirement and the limiter in this case.

closed 2023-03-02 by nseidle