sparkfun/SparkFun_RTK_Firmware #366: WiFi Config: Display file name when new log is created

nseidle, 2022-12-19:
Request from #344:

> If the logging to the SD card is still running 'in the background' when WiFi AP config is active, then it'd be really cool to see the live log file name. (I'll open a separate issue if you want.)

nseidle, 2022-12-23:
Logging is not able to be active during AP Config. Pressing the new log button closes the current log. A new one is created once the device exits the AP Config mode and resets. We don't know the name of the future log file so we can't currently present the file name.

closed 2022-12-23 by nseidle

tonycanike, 2022-12-23:
One thing - the source of this request was to get some feedback when the "Start New Log" button was pressed.   I'm never sure if my press was accepted if there's no feedback.   Toss in gloves that sometimes work with touch screens, rain hitting the touch screen, etc., and feedback is important.

Would it make sense to message something like "current log file closed, new log file will be created on restart"  ?

Maybe that feedback already has been added.  I'm a bit behind on the firmware, still running a 2.6-DEV version.  I just got home from 10 days in the field surveying and I stuck with running a version that worked for me.

From #344 
> @nseidle is there any way we could get feedback after we push the "Start New Log" button in the WiFi config?
> 
> I finally got around to trying it out (testing the logging fixes in the 26-Oct build this morning). I think I pushed the button but not sure it did anything. Maybe display the new logging file name, or just a message that a new log file will be created after reboot, etc. I'm not exactly sure if pushing the button immediately closes the old and opens a new file, or if that happens after reboot.
> 
> If the logging to the SD card is still running 'in the background' when WiFi AP config is active, then it'd be really cool to see the live log file name. (I'll open a separate issue if you want.) And the filename could change when someone presses the "Start New Log" button.