sparkfun/SparkFun_RTK_Firmware #368: Failures in copying base coordinates from SD and downloading files. V2.7 12/22

amlago, 2022-12-23:
Facet v2.7 12/22
WIFI configuration

1) Basic Menu. With version 2.7 12/22 the display of saved base coordinates flashes.
The coordinate list is cleared from the screen and appears intermittently.
2) When a saved base is marked, the LOAD button does not load the values

3) The new file menu is an excellent idea, but despite being able to see the list of files I couldn't download them to the phone's memory. I'll try again and see if it works tomorrow.
But it is something fantastic to be able to see the files on the cell phone.

amlago, 2022-12-26:
Upgrade
Facet v2.7 from 12/26

1) Base menu. When setting the base with WIFI AP, the list of saved base coordinates keeps flashing on the screen.
2) When marking a station from the list, the LOAD button does not load the values.

amlago, 2022-12-28:
Upgrade
Facet v2.7 from 12/28

With version 2.7 12/28 the same issue remains the same.
Just to keep updating depending on the version.

tonycanike, 2022-12-30:
What's the status of the base coordinate loading issues @amlago reported (points 1 and 2 in the original post)?

I've been keeping to my current FW version ( v2.6-Oct 26 2022) as these are critical capabilities for me, so I can't help with testing 2.7.  Been busy with fieldwork the past couple months and could not interrupt my work.   Those features work great in v2.6-Oct 26 2022.    But I have some time to test and would like to upgrade the FW.

Thanks.

> Facet v2.7 12/22 WIFI configuration
> 
>     1. Basic Menu. With version 2.7 12/22 the display of saved base coordinates flashes.
>        The coordinate list is cleared from the screen and appears intermittently.
> 
>     2. When a saved base is marked, the LOAD button does not load the values

nseidle, 2022-12-30:
@tonycanike - WIP. I hope I can get to it today.

Hi amlago - I always appreciate the bug reports! Thank you for all your time testing. 

> despite being able to see the list of files I couldn't download them to the phone's memory

I have moved the above issue into #370. Please open a new issue for each problem so we can track them accurately.

I'll work on the coordinate issues today.

nseidle, 2022-12-30:
Ah, it's #349. The dynamic coordinates are pulling the focus away from the list and causing the user's choice to deselect. I'll see what I can do to fix it.

Btw - I don't doubt the phone is preferred in the field, but with the latest release (v2.7) connecting to the device over a local WiFi network and using its browser is a breeze (compared to using the phone). 

[image]

amlago, 2022-12-30:
I thank you for the review.  The list of base coordinates is fundamental to my work.  In the field, in general, we only carry the cell phone or the notebook, it is left in the vehicle very far away.  So yes.  Using the cell phone is almost essential.  I keep testing the new firmwares and comment.  regards

nseidle, 2022-12-30:
Ok I believe we got it fixed. Please review and let me know what else needs fixing!

amlago, 2022-12-31:
The base coordinates list worked correctly again.  Fine.

nseidle, 2022-12-31:
Glad to hear it!

closed 2022-12-31 by nseidle