sparkfun/SparkFun_RTK_Firmware #370: File manager doesn't download to cell phone

nseidle, 2022-12-28:
v2.7 RC candidate. Originally from issue #368.

> The new file menu is an excellent idea, but despite being able to see the list of files I couldn't download them to the phone's memory. I'll try again and see if it works tomorrow. But it is something fantastic to be able to see the files on the cell phone.

nseidle, 2022-12-28:
Ok, I cannot replicate yet. But I may have found a few problems with the JS. And we have a new mechanism for downloading the file that may work better. I'll get an RC ready soon.

Note: We designed the file manager to download multiple files at once. Some browsers or phones will block this. (I will take a screen shot the next time I see it) You need to allow permission to download multiple files.

nseidle, 2022-12-30:
@amlago - Please give the RC and try and let me know if you are able to download files or not.

amlago, 2022-12-30:
I'll try it tomorrow and comment.  Thank you

amlago, 2022-12-31:
With Chrome Android at the beginning I get an error page but then it downloads without problems, even downloading several files at the same time.
I will try other android web browsers and comment.

nseidle, 2023-01-01:
Glad to hear it's working. If you get a chance, please capture the error with a screen shot or share any detail you can.

amlago, 2023-01-01:
Understood, as soon as I can upload the error capture.
Happy New Year

nseidle, 2023-01-19:
I'm going to close this issue as I think it's working in v3.0 firmware. Please open a new issue if you still see problems.

closed 2023-01-19 by nseidle