sparkfun/SparkFun_RTK_Firmware #372: Failed in Wi-Fi configuration Ap

amlago, 2022-12-31:
Facet v2.7 12/30
After configuring the facet with wifi AP, when saving, the Facet disconnects and the screen remains indicating an IP 0.0.0.0.

nseidle, 2022-12-31:
Do you have a valid WiFi SSID in your WiFi list?

The device is not able to connect to your WiFi so it's not able to get an IP address, so it's displaying 0.0.0.0.

But you raise a good point, we should probably display an error if no wifi is available, then fall back to AP mode for config.

amlago, 2022-12-31:
1) I discovered that the ip 0.0.0.0 is because the Facet is placed in WIFI network configuration mode when saving the data while in AP configuration mode.
As it is not connected to any WIFI network, the screen logically shows ip 0.0.0.0
2) I think the AP/WIFI choice is not respected when AP is chosen.

nseidle, 2023-01-19:
I believe this has been fixed in [v3.0 firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.0). Please open a new issue if you see any further problems.

closed 2023-01-19 by nseidle