sparkfun/SparkFun_RTK_Firmware #374: Save base coordinates obtained with Survey In to file

amlago, 2023-01-02:
Facet v2.7 12/30

When we use a method to obtain the base coordinates like Survey In, the coordinates are NOT saved in the base points file.

Suggested method

1) Add to the Survey In menu the entry of AH and ARP and name of the base
2) Once the values of the base coordinates have been obtained with Survey In, obtain the HAE Mark by subtracting the values of AH and ARP from the ellipsoidal height (HAE ACP) obtained with Survey In.
3) Automatically save the values of the mark coordinates to the file. (latitude, longitude and HAE Mark)

amlago, 2023-01-02:
By adding what is requested, I think that the tasks related to the base stations are facilitated, in the most common forms of use.

amlago, 2023-01-02:
HAE Mark = HAE APC -AH-ARP

tonycanike, 2023-02-08:
This would be very useful.  It is often necessary to know where the base "thought" it was.

USE CASE A:
1.  I set a mark in the ground.
2. I setup the Facet over it.
3. I set the Facet to log lots of good stuff and survey itself it.
4. The Facet surveys itself in.
5. I write down the HI of the base (very important!)
6. I RTK all day long.
7. I go back to the office, and get accurate coordinate for my mark is by using the base data log and an positioning service.
8. I calculate a vector from where the base thought it was to where it actually was.
9. I translate all my rover points by that vector to correct them.

USE CASE B:
1.  I setup the Facet over a mark in the ground I have good coordinates for already.
2.  I enter the good coordinates for the mark, along with HI and APC offset.
3. I write down the HI of the base (very important!)
4.  I RTK all day long.
5. I go back to the office and wonder "did I enter the coordinate coordinates correctly".  Or my boss asked for documentation.
6.  I look on my phone for a screen-capture of the configuration.
7. I have it or I don't.

 
Hmmm, would using the "mark" button feature save the coordinates to the SD card?   I presume the file of marks gets the APC coordinates; hopefully I wrote down the HI of the Facet.  

Tony.

amlago, 2023-02-09:
Ideally, the mark coordinates should be saved.
Hence the HA and ARP entry.

¨Suggested method

   1) Add to the Survey In menu the entry of AH and ARP and name of the base
   2) Once the values of the base coordinates have been obtained with Survey In, obtain the HAE Mark by subtracting the values of AH and ARP from the ellipsoidal height (HAE ACP) obtained with Survey In.
    3)Automatically save the values of the mark coordinates to the file. (latitude, longitude and HAE Mark)¨

They are ideas to share.
Greetings

amlago, 2023-02-09:
Tony, check out the #347 we've been driving for a while.
I proposed, if possible, to centralize all the basic configuration modes in a dropdown and from there the corresponding interface and mathematical presentations will be made.
I know it's a total redesign of the base configuration.

tonycanike, 2023-02-09:
Ah, yes. Good point! We want the coordinates of the mark saved. 

And the mark button doesn't save the coordinates of the mark!  :)

tonycanike, 2023-02-13:
Other vendors have a log file in which configuration entries and changes are documented for future reference and documentation.  

Perhaps every time a base is started a record of the setup could be written to a file. I believe most commercial GNSS vendors do this in some fashion.  

There is at least one originally-proprietary formats that have become a de facto industry standard, eg .RAW/.RW5. This format is used to capture information that a surveyor might write in a paper book.  Total station backsight, prism constants, GNSS base height, GNSS points recorded, etc. This format is interesting because if you enter an incorrect base coordinate or antenna height in the field, there are a number of commonly-used software products that will allow you to edit the erroneous entry and "reprocess" the "raw" file, correcting all the surveyed point coordinates. Survey office staff and supervisors use this to correct errors made by field crews.  You can edit a base setup field made in the morning and fix all the coordinates captured all day long. 

There is a newer LANDXML open format that also documents survey field data (and other data,) This might be more in line with the goal of open source projects. Check out the GPSSetup tag in the schema at landxml.org. 

I think Emlid writes the base info to a CSV format they have developed to capture GNSS coordinate info.

tonycanike, 2023-02-13:
Can we change the title to save the coordinates and info used for every base setup, and not just survey in? The documentation would be very helpful. 

And practically, it's pretty common for me to setup my base receiver on a tripod, turn it on, and then go into WiFi configuration to set the base coordinates, heights, etc.  

So it's very possible it might survey itself in before I start the WiFi base configuration.  Having only the survey-in info in the file would be very confusing. 

If three days later I'm sorting out the data and I only had the survey-in data written to a file and not the manual base configuration I entered a couple minutes later, it would be very unfortunate.

amlago, 2023-02-16:
In v3.1, the base and HA and ARP coordinates are already being saved in the csv file.

It would be necessary to add what is related to Survey In and it would be fantastic to unify all the modes in a dropdown, choose the input method of base coordinates and do the mathematics that correspond to each method.

See #347

tonycanike, 2023-02-24:
Note that implementing #405 would help the use case if the user's data collector stores the base reference or if the user is logging.

nseidle, 2023-03-08:
I am ready to look into this. Please list what changes you need made. This is what I think I heard:

1. Add to the Survey In menu the entry of AH, ARP and name of the base
2. Once the values of the base coordinates have been obtained with Survey In, obtain the HAE Mark by subtracting the values of AH and ARP from the ellipsoidal height (HAE ACP) obtained with Survey In.
3. Automatically save the values of the mark coordinates to the 'Commonly Used Coordinates' file. (latitude, longitude and HAE Mark)¨

amlago, 2023-03-08:
What you summarize is correct.

Point 1) and 2) I have no doubts.

Regarding point 3) I am not sure whether to save the coordinates automatically or add a check to indicate whether they are saved or not.
I think I don't like the idea of adding a check indicator for saving coordinates.

amlago, 2023-03-08:
I liked adding a check because sometimes it is not necessary or convenient to save the coordinates.
So I am inclined to add a check indicator to save the coordinates in the file.

rrodas64, 2023-03-08:
Why not just add the coordinates obtained with the Survey-in to the list of Commonly Used Coordinates with a specific Nickname? Let's say Surver-in_mmddyyyy_hhmmss.

amlago, 2023-03-09:
@rrodas64 
The idea is to follow the line of work already established in the base configuration, where the base station is given a name that represents the area where it is located or the work you are doing.