sparkfun/SparkFun_RTK_Firmware #375: New method for updating firmware

nseidle, 2023-01-06:
We've added the ability to update the firmware via *pull* from both the AP config page and firmware serial menu. This is what it looks like:

[image]

In summary, you can update the firmware on a unit over WiFi without the need for downloading to an SD or a phone. You can update the unit's firmware by checking for the latest firmware from SparkFun (aka the [RTK Binaries Repo](https://github.com/sparkfun/SparkFun_RTK_Firmware_Binaries)). This should make it easier to upgrade devices in the future.

Please give it a try and let us know if you have any problems.

amlago, 2023-01-12:
Very good idea, as soon as a new version appears I will try it and comment.

nseidle, 2023-01-19:
This has been implemented in [v3.0 firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.0).

closed 2023-01-19 by nseidle