sparkfun/SparkFun_RTK_Firmware #377: Regarding bluetooth serial configuration: details add or correct

amlago, 2023-01-12:
Facet v2.7 01/06/23
bluetooth serial configuration

1) Enter 3) Base Configuration and then 4) Set Antenna Height
If I enter negative values, which is not usual but sometimes I have needed it, it does not enter the value correctly, for example if I enter
-1650 actually places 63886

2) Enter 5) Configuration System and then f) Display SD file.
It does not show me the list of existing files on the SD.
If I can see them by other existing methods.

nseidle, 2023-01-19:
Please do one issue per issue :)

Good find!

I can confirm. Antenna height is not printing as positive (Bluetooth connection doesn't matter in this case).

[image]

It's a variable (uint16_t should be a int16_t). We'll get it into the next RC.

[image]

2) Moved to separate issue

nseidle, 2023-02-04:
This was fixed in [v3.1](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.1). Please let us know if it is still an issue.

closed 2023-02-04 by nseidle