sparkfun/SparkFun_RTK_Firmware #381: Add confirm delete to File Manager and Common Coordinates list

nseidle, 2023-01-19:
From #378 

In the File Manager tab, when deleting a file with the Delete button, it would be good to add a confirm option, to avoid accidental deletion.

In the Base Configuration tab, in the list of saved base coordinates, it would be good to also add an option to confirm, when deleting a base, to avoid accidental deletion.

nseidle, 2023-02-04:
This was fixed in [v3.1](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.1). Please let us know if it is still an issue.

closed 2023-02-04 by nseidle

amlago, 2023-02-04:
¨In the Base Configuration tab, in the list of saved base coordinates, it would be good to also add an option to confirm, when deleting a base, to avoid accidental deletion.¨

This was not contemplated in the latest version of the firmware.

The confirmation to delete files from the SD if it works very well.