sparkfun/SparkFun_RTK_Firmware #383: Copy settings between profiles

nseidle, 2023-01-19:
Moved from issue #378 

> Regarding the list of saved base coordinates, it would be good if it could be copied from profile to profile.

The common coordinates are stored in a file assigned to the profile #. What you are asking is the ability to copy certain things between profiles. I'll see what I can do. It may be awhile.

tonycanike, 2023-02-08:
Could this be done on one's phone?  I haven't tried, but just wondering.

Download this file from the SD card to your phone:   StationCoordinates-Geodetic_0.csv   (Profile1 I believe)
On your phone copy that file to StationCoordinates-Geodetic_1.csv
Upload the file from your phone to the SD card:  StationCoordinates-Geodetic_1.csv  (Profile2 I believe)

amlago, 2023-02-09:
Tony: I'll try to carry out your advice.
I've never created coordinate files by hand for the Facet, but it looks very useful.

nseidle, 2023-02-24:
Do we still need these feature? I'm really leaning away from it. Tony's solution is a good one and a fine way to set coordinates for various profiles.

amlago, 2023-02-24:
Thank you. I think there are other more important things before this. I would say don't listen to this.
I'll deal with the coordinate files manually for now.
Greetings

nseidle, 2023-03-02:
I appreciate your understanding. Thank you! Closing.

closed 2023-03-02 by nseidle