sparkfun/SparkFun_RTK_Firmware #386: "Assisted Base" resource don't work

eduardobuenof, 2023-01-26:
### Subject of the issue
Describe your issue here. Additionally, screenshots are easy to paste into github.

I am trying to use the "Assisted Base" resource (https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_base/#assisted-base), but it is not connect to the NTRIP Client, I believe the access credentials are correct because in "Rover" mode it works perfectly, however when switching to "Base" mode it does not connect to the server by NTRIP. Is this a FW bug? I'm using the latest version (3.0)

### Your workbench
* What version of RTK firmware are you running? RTK_Surveyor_Firmware_v3_0
* 
* This can be found when the serial menu is opened (also in the settings.txt file, and in the serial output at power up).

**###TERMINAL COM###**

Display started
ZED-F9P firmware: HPG 1.32
LittleFS Started
SparkFun RTK Express v3.0-Jan 19 2023
microSD: Online
Profile 'GPS_portable' loaded
MAX17048 configuration complete
Batt (87%): Voltage: 4.14V Discharging: -1.66%/hr Green
GNSS configuration complete
Accelerometer configuration complete
System time set to: Thursday, January 26 2023 13:20:30
*WIFI_OFF
*NTRIP_CLIENT_OFF
NTRIP Client start
NTRIP_CLIENT_ON
State: Rover - No Fix, 2023-01-26 13:20:33.106
Batt (87%): Voltage: 4.14V Discharging: -1.66%/hr Green
WIFI_CONNECTING
NTRIP_CLIENT_WIFI_STARTED
State: Rover - Fix, 2023-01-26 13:20:33.183
Connecting WiFi... SD card size: 7.2 GB / Free space: 7.2 GB

WIFI_CONNECTED
WiFi IP address: 172.17.1.45 RSSI: -41
NTRIP_CLIENT_WIFI_CONNECTED
Rover Accuracy (m): 296.8208
Batt (87%): Voltage: 4.14V Discharging: -1.66%/hr Green
Log file name: SFE_Express_230126_132039.ubx
NTRIP Client connecting to gps-ntrip.ibge.gov.br:2101
NTRIP Client connected
NTRIP Client sending credentials: -I hid this information-
NTRIP Client serverRequest size: 107 of 515 bytes available
NTRIP Client sending server request:
GET /UFPR0 HTTP/1.0
User-Agent: NTRIP SparkFun_RTK_Express_v3.0
Authorization: Basic ZWR1YmY6Mzc2MDA2

NTRIP_CLIENT_CONNECTING
Log file size: 151 - Generation rate: 0.0kB/s
NTRIP_CLIENT_CONNECTED
Rover Accuracy (m): 6.2680
Rover Accuracy (m): 5.5900
Batt (87%): Voltage: 4.14V Discharging: -1.66%/hr Green
Log file size: 14268 - Generation rate: 2.8kB/s
Rover Accuracy (m): 5.1403
Rover Accuracy (m): 5.0615
State: Rover - RTK Float, 2023-01-26 13:20:49.406
Batt (87%): Voltage: 4.09V Discharging: -1.25%/hr Green
Log file size: 36496 - Generation rate: 4.4kB/s
Rover Accuracy (m): 2.8656
WiFi IP address: 172.17.1.45 RSSI: -39
Rover Accuracy (m): 1.9042
Rover Accuracy (m): 1.5496
State: Rover - Fix, 2023-01-26 13:20:54.112
Batt (87%): Voltage: 4.09V Discharging: -1.25%/hr Green
Log file size: 61919 - Generation rate: 5.1kB/s
Rover Accuracy (m): 1.5709
Rover Accuracy (m): 1.6116
[... 236 lines cut ...]
Time elapsed: 45 Accuracy: 31 SIV: 20
Time elapsed: 46 Accuracy: 31 SIV: 20
Batt (87%): Voltage: 4.11V Charging: 5.20%/hr Green
Log file size: 500107 - Generation rate: 1.6kB/s
State: Shut Down, 2023-01-26 13:22:52.654
Log file closed
microSD: Offline

**###TERMINAL COM###**

* What radios are you using: Bluetooth, WiFi, and/or ESP-Now? What app are you using to connect over Bluetooth? Are you transmitting NTRIP back to the device?
In Rover mode: I am using WiFi to connect to NTRIP and receive RTCM.

And in Base mode: I send the RTCM signal by SIK telemetry radio

* Are there any additional details that may help us help you?

Setup:
[image]

Rover Mode:
[image]

Base Mode:
[image]

nseidle, 2023-01-26:
Nice description, photos, and setup. Thanks!

The debug output is superb and very helpful.

I am zeroing in on this section:

> Rover Accuracy (m): 0.2488
Batt (87%): Voltage: 4.11V Charging: 1.87%/hr Green
Log file size: 373340 - Generation rate: 5.5kB/s
Rover Accuracy (m): 0.2381
State: Display Setup, 2023-01-26 13:21:56.469
Batt (87%): Voltage: 4.11V Charging: 1.87%/hr Green
Log file size: 399283 - Generation rate: 5.2kB/s
WiFi IP address: 172.17.1.45 RSSI: -41
State: Base - Not Started, 2023-01-26 13:22:00.573
WIFI_OFF
NTRIP Client connection to caster was broken
NTRIP_CLIENT_ON
*WIFI_OFF
RTCM parser error maxLength: 61 bytes
RTCM parser error maxLength: 71 bytes
RTCM parser error maxLength: 86 bytes
State: Base-Temp - Settle, 2023-01-26 13:22:02.125
Waiting for Horz Accuracy < 0.50 meters: 0.24
Resetting survey
Survey started. This will run until 60 seconds have passed and less than 1.000 meter accuracy is achieved.
State: Base-Temp - Survey Started, 2023-01-26 13:22:08.058
Batt (87%): Voltage: 4.11V Charging: 1.87%/hr Green

Above, the Rover accuracy is 0.2381m. It has not yet achieved RTK fix. So the base will begin a survey-in with ~0.2m accuracy. I recommend waiting for a RTK Fix before starting an assisted survey in for maximum accuracy.

Now, the root of the matter is whether we continue to send NTRIP data to the Base while it's surveying in. 

[image]
Above: I just ran a test on an RTK Express. The Rover got RTK Fix within a few seconds, then I switched to Base mode. Above is the device re-connecting successfully to the NTRIP Caster. I have a lot of debug statements turned on so don't be surprised by the extra prints.

[image]

Above: A few seconds later, the during survey in, the device disconnects from WiFi. This is not good. I'll look into it. 

[image]
Above: However, my survey-in completed with RTK fix in place and HPA of 0.014m so assuming you enter Survey-In with RTK precision, your survey-in should maintain it for ~30 seconds, and then complete with RTK float. 

We'll figure out why the NTRIP Client is disconnecting during Base Survey-In.

eduardobuenof, 2023-01-26:
Thanks for the quick response!

Yes, this data I posted was just to explain the issue in which the RTK Express cannot obtain data from the NTRIP client in Base Survey-In mode, the point is this excerpt:

NTRIP Client connecting to gps-ntrip.ibge.gov.br:2101
NTRIP_CLIENT_ON
**NTRIP Client failed to connect to caster**. Trying again in 15 seconds.

But I did tests in an outdoor location with a precision 14mm, but after entering Base Survey-In mode the Wi-Fi is disconnected and can't obtain NTRIP data, and only starts transmission after a minimum 60s, which is time enough that I lost my 14mm accuracy, and sometimes don't start with 14mm, reset to values above 10m.

nseidle, 2023-01-26:
Ok - I think we have a fix. 

If:

* NTRIP Client is enabled
* Base type is temp
* NTRIP Client is enabled
* Base is started

Then:
* Keep WiFi and NTRIP Client running

Once survey in is complete, NTRIP Client is turned off, and WiFi does whatever WiFi needs to do (stays on for NTRIP Server, etc).

nseidle, 2023-01-26:
Please give the [release candidate](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.1RC) a try.

eduardobuenof, 2023-01-26:
Now I think it's working, but would be somewhere can I see when it pulls data from the NTRIP Client? To make sure it's applying RTCM during Survey-In.

Because in Rover mode it acquired 14mm but during the 60s of the Survey-In preparation it got 26mm HPA.

It is possible to change these ranges:

Menu: Base
1) Toggle Base Mode: Use Survey-In
**2) Set minimum observation time: 60 seconds**
**3) Set required Mean 3D Standard Deviation: 5,000 meters**
**4) Set required initial positional accuracy before Survey-In: 0.100 meters**
6) Toggle NTRIP Server: Disabled
11) Select survey-in radio: Bluetooth
x) Exit

for 2) and 4) I would like to put smaller values

commit 8c7c7e89 (2023-01-26): Keep WiFi on during Temp Base Survey In

Fix #386
closed 2023-02-03 by nseidle