sparkfun/SparkFun_RTK_Firmware #387: BLE pairing attempt from Android crashes ESP32 and reboots at handleGNSSData()

edgecase14, 2023-01-31:
### Subject of the issue
Anticipating the increasing difficulty of using Bluetooth Classic, I was evaluating feasibility of using BLE on Andriod.

### Your workbench
* What version of RTK firmware are you running? This can be found when the serial menu is opened (also in the settings.txt file, and in the serial output at power up).

SparkFun RTK Surveyor v3.1-Jan 26 2023 github binary release

* What radios are you using: Bluetooth, WiFi, and/or ESP-Now? What app are you using to connect over Bluetooth? Are you transmitting NTRIP back to the device?
* Are there any additional details that may help us help you?

This is a non-standard setup, with just an esp32 connected via i2c to ZED-F9P and NEO-D9S.  It works well in Bluetooth classic mode, with PointPerfect L-band corrections, in conjunction with QFIELD.

Phone is Google Pixel 4a.  Pairing was done from Android bluetooth screen.

### Steps to reproduce
Tell us how to reproduce this issue. Please post any log files from serial output that may have been generated.

### Expected behavior
Tell us what should happen
Device should appear in list of "known" devices.

### Actual behavior
Tell us what happens instead
Phone times out with "Couldn't pair with Surveyor Rover-C43A because of an incorrect PIN or passkey." and USB serial console shows crash dump:

Batt (10%): Voltage: 3.70V Discharging: 0.00%/hr Yellow
[ 28418][D][menuPP.ino:728] pushRXMPMP(): Pushing 528 bytes of RXM-PMP data to GNSS
[ 28489][D][menuPP.ino:800] checkRXMCOR(): L-Band Eb/N0[dB] (>9 is good): 9.62
Rover Accuracy (m): 0.0287
[ 30109][D][menuPP.ino:728] pushRXMPMP(): Pushing 528 bytes of RXM-PMP data to GNSS
[ 30188][D][menuPP.ino:800] checkRXMCOR(): L-Band Eb/N0[dB] (>9 is good): 10.25
Rover Accuracy (m): 0.0279
[ 30881][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
BT client Connected
[ 30882][D][BLEServer.cpp:365] onConnect(): BLEServerCallbacks
[ 30887][D][BLEServer.cpp:366] onConnect(): BLEServerCallbacks
[ 30893][D][BLEServer.cpp:367] onConnect(): BLEServerCallbacks
[ 30997][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 30997][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
[ 30998][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 31004][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[ 31011][D][BLECharacteristic.cpp:790] onNotify(): >> onNotify: default
[ 31016][D][BLECharacteristic.cpp:791] onNotify(): << onNotify
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception).
Debug exception reason: Stack canary watchpoint triggered (handleGNSSData)
Core  1 register dump:
PC      : 0x4009538c  PS      : 0x00060036  A0      : 0x40085124  A1      : 0x3ffec160
A2      : 0x00000000  A3      : 0x3ffec414  A4      : 0xffffffff  A5      : 0x3ffec230
A6      : 0x3ffec5b0  A7      : 0x00000008  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000001  A11     : 0x3f476bfd  A12     : 0x000000ff  A13     : 0x0000ff00
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x40092d61  LEND    : 0x40092d71  LCOUNT  : 0xffffffff

Backtrace:0x40095389:0x3ffec1600x40085121:0x3ffec530 0x4010f915:0x3ffec5f0 0x4010a055:0x3ffec720 0x401078fd:0x3ffec7b0 0x400d4235:0x3ffec7d0 0x40107a96:0x3ffec7f0 0x400d41e7:0x3ffec820 0x400d6795:0x3ffec840 0x400ed5eb:0x3ffec860

Core  0 register dump:
PC      : 0x4009133a  PS      : 0x00060033  A0      : 0x8009706c  A1      : 0x3ffebdb0
A2      : 0x00000000  A3      : 0x3f468b7e  A4      : 0x00000014  A5      : 0x00060023
A6      : 0x00060023  A7      : 0x00000001  A8      : 0x3f468b7e  A9      : 0x00000000
A10     : 0x00000001  A11     : 0x00000001  A12     : 0x00060021  A13     : 0x00000000
A14     : 0x007bff88  A15     : 0x003fffff  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x40091358  LEND    : 0x40091362  LCOUNT  : 0x00000000

Backtrace:0x40091337:0x3ffebdb00x40097069:0x3ffebdd0 0x4009547c:0x3ffebe00 0x4009542c:0x00000000  |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
[HH▒▒3][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
Display not detected
ZED-F9P firmware: HPG 1.32
LittleFS Started
[   280][D][esp32-hal-adc.c:210] __analogReadMilliVolts(): eFuse Vref: Supported
[   280][I][esp32-hal-adc.c:240] __analogReadMilliVolts(): ADC1: Characterized using eFuse Vref: 1100

[   286][D][Begin.ino:10] beginBoard(): Board ADC ID: 301
SparkFun RTK Surveyor v3.1-Jan 26 2023
[   325][D][NVM.ino:1101] loadProfileNumber(): Using profile #0
Reset reason: ESP_RST_PANIC
[   808][D][NVM.ino:409] loadSystemSettingsFromFileSD(): Config file read failed: SD offline

nseidle, 2023-02-01:
Thanks for reporting! You may be the first person trying to get RTK firmware to load on a non-RTK device. Congrats!

If I 'Forget' the SPP device on my phone, then scan for and attempt to connect to the BLE enabled RTK device, I get the same crash as you report. I'll dig in.

nseidle, 2023-02-01:
Older [v2.5 firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware_Binaries/tree/main/PreviousVersions) is working for me, v2.6 fails.

nseidle, 2023-02-03:
We've got BLE working again with a few caveats:

* All release candidate firmware sets the output level to 'Debug'. When BLE is connected, the number of messages at this level is so large that it will cause the ESP to reset. 
* Connecting over BLE and doing menu config is mostly working. When Debug output is turned off (using a full release firmware for example) BLE print functions are different enough from standard prints that a few menus are not printing correctly (System and Messages NMEA for example). We're still putting fixes in places for these, but it's very usable.

With a full release, BLE works 100% for transmission of NMEA and UBX to a data collector.

I'll see if I can get a full release made shortly.

nseidle, 2023-02-04:
We've got this added in [v3.1](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v3.1). Please let us know if it is still an issue.

You should be able to use BLE in full releases from here on, but I cannot say if BLE will work, in general, in release candidate firmware due to the nature of our workflow and the need to have debug messages enabled.

Closing the issue. Please let us know if you spot any other problems.

closed 2023-02-04 by nseidle