sparkfun/SparkFun_RTK_Firmware #391: Base coordinates deletion confirmation button from the list

amlago, 2023-02-04:
¨In the Base Configuration tab, in the list of saved base coordinates, it would be good to also add an option to confirm, when deleting a base, to avoid accidental deletion.¨
#381
This was not contemplated in the latest version of the firmware.

The confirmation to delete files from the SD if it works very well.

_Originally posted by @amlago in https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/381#issuecomment-1416878997_

amlago, 2023-02-05:
It would also be good if when saving coordinates with the ADD button there would be a saving confirmation notice and/or an overwrite notice if the Base name already exists.

It happened to me that I saved a base by pressing the Add button and I didn't do it because I pressed the button wrong.
It has also happened to me to overwrite a base with the same name.

nseidle, 2023-02-24:
Right! Sorry I missed that. Should be in soon.

commit 468f005b (2023-02-24): Add delete confirmation to commonly used coordinates

Fix #391
closed 2023-03-08 by nseidle