sparkfun/SparkFun_RTK_Firmware #392: Duplication of base coordinates saved in file

amlago, 2023-02-05:
When switching profiles, the base coordinates were multiplied multiple times in the list of saved coordinates.
Delete them one by one until you have only one value of each saved base.
It's the second time it's happened to me and I really don't know the sequence of events that triggered it.

amlago, 2023-02-07:
Facet v3.1 I forgot to clarify

nseidle, 2023-03-02:
I believe this was occurring because the of the corruption of settings from the browser. See [this comment](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/382#issuecomment-1444693185). I believe this has been resolved in the latest RC but please let me know if you see anything like this again.

nseidle, 2023-03-02:
I forgot to say: thanks for reporting!

nseidle, 2023-03-08:
This should be fixed in [v3.2](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases).  Please let us know if this is still a problem.

closed 2023-03-08 by nseidle