sparkfun/SparkFun_RTK_Firmware #394: Slow issues and save issues on WIFI AP v3.1

amlago, 2023-02-07:
Facet v3.1

Slow issues and save in WIFI AP v3.1

1) In v3.1 the access to the Wi-Fi Ap configuration has become very slow and sometimes the page does not fully load. Tested with the wifi of a cell phone and the wifi of my home router which is very good.
In v3.0 I have rarely had problems like this, although from time to time a hang appears.

2) When the page is completely loaded (v3.1) and changes are made, when saving most of the times it remains forever saving and the process does not end.

amlago, 2023-02-08:
Update 02/08/2023

1) The Ap configuration was done with a Huawei cell phone connected to the hotspot of a first Xiaomi cell phone, which is the one that provides internet to the Facet to send the corrections to Rtk2go. The failures that I mentioned occurred with this configuration.

2) I have tried to configure the Facet v3.1 directly with the Xiaomi cell phone that provides it with internet and it seems that the Ap configuration does not get stuck and save the changes correctly.

I keep reporting if there is any change

nseidle, 2023-02-24:
Setup #1: 

Ok, so you are using WiFi Config in WiFi mode, not AP mode. Is that correct? Like shown below?

[image]

I tried to replicate your setup this way:
* Firmware v3.1 on RTK Facet L-Band
* My phone is a hotspot with network called 'spark', and is providing internet.
* I connected my laptop to WiFi network spark
* I configured my RTK Facet, over serial, to use network 'spark', and to use config in WiFi mode
* I entered Config mode via the setup button on the Facet
* I entered the IP address seen on Facet into my laptop (192.168.179.222)
* My laptop took 5 seconds but loaded the RTK Config page into the browser
* I downloaded a 15MB file from the Facet in about 45 seconds

Everything seems to be working nicely for me.

Can you give me the exact model of Xiaomi cell phone you're using? I may be able to locate one for testing.

Setup #2: This sounds like the WiFi Config using the AP setting is working fine.

nseidle, 2023-02-24:
I have been doing a lot of testing today with v3.1, Google Pixel 4a, Lenovo Laptop, and all is connecting and working well.

I know the WiFi page can sometimes freeze/not load. But it's been working well. Help me replicate your issue.

amlago, 2023-02-24:
It seems that the problem is the slow wifi connection created by one of my phones. If I take the time for the page to load completely, everything seems to work fine.
It just does it very slowly. I keep trying.

nseidle, 2023-03-02:
I believe this is working in the latest RC. Please let me know if you continue to see any issues.

nseidle, 2023-03-08:
This should be fixed in [v3.2](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases).  Please let us know if this is still a problem.

closed 2023-03-08 by nseidle