sparkfun/SparkFun_RTK_Firmware #395: Allow entry of elevation mask

amlago, 2023-02-08:
I would like to enter the angle of the elevation mask.

1) So that, for example, the corrections that are generated are generated only by the satellites above the mask.

2) If the rover is used, the position is only calculated with the satellites above the mask

tonycanike, 2023-02-08:
Good idea!

nseidle, 2023-02-24:
Help me understand this. Are you asking for satellites below a certain elevation to not be used in positional fixes?

[image]

We can add a control for the "Minimum elevation for a GNSS satellite to be used in NAV".

Is that what you're looking for?

nseidle, 2023-02-24:
This is very do-able. Just some work. Please let me know if Min SV Elevation is what you're looking for.

amlago, 2023-02-24:
What you tell me is correct.
The elevation mask is the minimum angle, above which the satellites are used for the calculations.
If the elevation mask is zero, all visible satellites above the horizon are used.
Generally the angle is 10 degrees or 15 degrees.

nseidle, 2023-03-09:
Working well. This will be in next RC.

[image]

[image]

nseidle, 2023-05-05:
This was implemented in [v3.3 of the RTK firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v.3.3). Please let us know if you see any problems.

closed 2023-05-05 by nseidle

nseidle, 2023-05-05:
[Docs added](https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_gnss/#min-sv-elevation-and-cn0).