sparkfun/SparkFun_RTK_Firmware #396: Improve Base Coordinate file handling when "format rules" are violated.

tonycanike, 2023-02-08:
### Subject of the issue

It appears if the StationCoordinates-Geodetic_N.csv file is hand edited and some formatting rules are violated, the file ignored.  If the file is ignored, it gets deleted and replaced when someone adds a new coordinate to it via the WiFI Base Configuration.

I don't use ECEF, but perhaps the same problem is there too.

### Your workbench
FW 3.1

### Steps to reproduce

I have databases and spreadsheets full of coordinates of control points that I want to set up my base over.  So I LOVE the feature to save Fixed Base coordinates to a file.  Thank you!! 

It's much more efficient to create the StationCoordinates-Geodetic_N.csv on my computer with the help of spreadsheets then to sit and type all the little numbers into configuration entry screens. So of course I'm going to hand edit the file.

By trial and error, it seems that
1.  Mark names are limited to 7 characters
2.  Lat and Lon are limited to 10 dig to the right of the decimal.  (at least for 38.1234567890 and -79.1234567890)
3.  Never had a height with more than 3 dig to the right of the decimal.  Didn't try more.
4.  I recall that in the past I could not leave any blanks in the file.  Didn't try this time.
5.  It seems if these rules are violated, the file is totally ignored.  Ok, I get it, hand-editing config files is not a published interface.
6.  If the file is ignored, no coordinates are loaded into the WiFi base configuration.
**7.  If the file is ignored, and a new coordinate is saved, the whole file is wiped out.**  

I probably missed something or got something wrong.  But after a couple hours of trial and error and following the above rules, I got a file with 15 coordinates in it to be recognized by the firmware.  Which is really sweet!

**It would be really nice if that #7 rule above didn't happen and the file wasn't wiped out.**  Perhaps at the least the file could be renamed instead of being wiped out.   Perhaps rename it to "StationCoordinates-Geodetic_N.csvBadBadDangerDangerSTOPeditingThisFile" or something like that.

### Expected behavior
Tell us what should happen

### Actual behavior
Tell us what happens instead

nseidle, 2023-02-24:
tldr; Would you please share the contents of a CSV file that fails for you?

Sorry! Please ask questions before burning hours trying to figure out my madness. We'll document it for you and everyone else!

Just to [rubber duck](https://en.wikipedia.org/wiki/Rubber_duck_debugging) a bit:

* The unit will read whatever is in stationCoordinateGeodeticFileName, line by line, and sends that to the browser. Each line is prefixed with 'stationGeodetic%d' assigning a record # as we go.
* Browser (JavaScript), upon receiving a line with 'stationGeodetic', shoves the CSVs into the recordsGeodetic array
* JS will create the list of commonly used points based on recordsGeodetic
* JS will validate values when the user hits 'Add' button
* JS removes the following characters from any nickname: ' ' '\' ',' (space, backslash, comma)

I don't immediately see anything that would lead to what you are seeing. Let's try to replicate:

    LongerThan7CharacterTest,40.09035417,-105.18507155,1525.290
    Test2,40.09035417,-105.18507155,1525.290,A,0
    Test3,12.345,-1.234567890123456,1,2,3

The above CSV was loaded on the SD card using VS Code. Note record one is incomplete, record 2 has a letter for the antenna height, record 3 has weird decimals.

[image]

The data comes across to the browser well.

[image]

Loading LongerThan7Chars looks good. Loads all the values it can. No antenna height or ARP.

[image]

Above, Test2 loads with missing antenna height. That's good.

[image]

Above, Test3 loads correctly with very long/short decimals.

Modifying these records a bit: Test2 to antenna height of 500 and Test3 height to 20. 

[image]

Above, pressing the 'Save Config' button I get the blob of text settings including the new geo positions.

Pulling SD I see:

    LongerThan7CharacterTest,40.09035417,-105.18507155,1525.290
    Test2,40.09035417,-105.18507155,1525.290,500,0
    Test3,12.345,-1.234567890123456,1,20,3

Looks good to me.

[image]

Ways the files could be messed up:

* The StationCoordinates-Geodetic_0.csv (same for ECEF) gets overwritten by data from the browser. If this data is corrupt, or incomplete, it will blow away what you have on the SD card.

I re-ran tests with a 9 station log and still no problems. I am actually shocked my code is working as well as it is! But as far as I can see, all is well :)

tonycanike, 2023-02-25:
Thanks, I'm tied up for a few days and will look at this mid next week.

tonycanike, 2023-03-08:
I'm going to close this.  @nseidle 's testing reports lead me to believe there was something else going on.  Family matters are requiring me to postpone my testing.

closed 2023-03-08 by tonycanike