sparkfun/SparkFun_RTK_Firmware #397: Base Fixed WiFi config APC Tool Tip:  include the numbers 61.4 and 43.6 in the tool tip text

tonycanike, 2023-02-08:
### Subject of the issue
Include L1 APC offsets for the Facet (61.4) and the Top106 (43.6) in the tool tip for the Base Fixed WiFi config APC entry field.
(L-Band Facet too maybe?  I'm not familiar with that model.)

### Your workbench

Facet with FW 3.1

### Steps to reproduce

1. Setup and level tripod over the base mark. Set it up nice and high to get a good view of the sky.
2. Install the Base Facet on the puck on the tribrach.
3. Connect all the radio, antenna, and external power cables.
4. Raise that hi-gain antenna I got from Sparkfun way up high using another pole and tripod.
5. Tell the neighbor I'm not stealing their land.
6. Shoo the cows away.
7. Reach up and put the Facet in AP Config mode.
8. Get out my phone, connect to the Facet's RTK Config access point.
9. Open the base config page and set the coordinates of the mark.
10. Measure the HI of the Facet and enter it into the antenna height field.
11. Look at the APC field and go "oh crap what was that again".
12. Kick myself for not writing it on the Facet with a big Sharpie.
13. Try to read the label on the Facet to no avail.
14. sigh loudly and think about beer.

### Expected behavior
Help a brother out with a little hint.

amlago, 2023-02-09:
I'm laughing alone.
It has happened to me a thousand times to forget the ARP values.
Good idea

nseidle, 2023-02-24:
Hilarious (thank you!) and a fine idea.

Where are you getting your TOP106 ARP from? We have the [mfg datasheet](https://cdn.sparkfun.com/assets/b/4/6/d/e/TOP106_GNSS_Antenna.pdf) at 43.6mm and the [ANTCAL](https://geodesy.noaa.gov/ANTCAL/LoadFile?file=SFETOP106_NONE.atx) from NGS at 52.9mm.

nseidle, 2023-02-24:
WIP: 
[image]

In the future, if this list gets too long, we can modify the text based on the unit's platform ID (aka only show Facet info if it's a Facet).

tonycanike, 2023-02-26:
I got the 43.6 from the data sheet. 

https://cdn.sparkfun.com/assets/b/4/6/d/e/TOP106_GNSS_Antenna.pdf

nseidle, 2023-03-08:
This has been added in [v3.2](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases).  Please let us know if this is still a problem.

closed 2023-03-08 by nseidle