sparkfun/SparkFun_RTK_Firmware #399: Now can not use WiFi to configure Facet.

tonycanike, 2023-02-12:
### Subject of the issue
I can not use WiFi to configure the Facet.  I was able to do it once earlier today to start a 3-hour PPP logging session that failed, but that's a separate issue #398 

### Your workbench
3.1

### Steps to reproduce
Power on
Let base or rover start
Use button to switch to WiFi config
Watch it display the SFE logo and reboot 

Repeat endlessly.

### Expected behavior
Would be nice to be able to configure the Facet and try logging again.

### Actual behavior
Can't configure the Facet.

tonycanike, 2023-02-12:
I pulled the SD card and did a factory reset. Delected settings profile from SD card.  Reinstalled SD Card.
Was able to get into WiFi config.  
I saved and exited vie the webui.   Got "RTK device is rebooting..."

BUT it HUNG in WiFi Config for many minutes and never exited.

amlago, 2023-02-12:
facet v3.1
Hi, the same thing happened to me a few days ago. I put the facet in wifi config mode and it rebooted forever. Update several times with other firmwares and the same.
The solution was to restore factory and it has continued to work for me.
Put v3.1 back.
I didn't know what happened.

amlago, 2023-02-12:
It is something similar or the same thing that I warned in #394

tonycanike, 2023-02-13:
@amlago Oh good point!  Our two reports can probably be consolidated.

nseidle, 2023-02-24:
> I pulled the SD card and did a factory reset. Delected settings profile from SD card. Reinstalled SD Card. Was able to get into WiFi config.

Did you have the same issue as [Ingedar](https://forum.sparkfun.com/viewtopic.php?f=116&t=59117)? Perhaps there is a setting between v2.x and v3.0 that causes the WiFi config to get set to WiFi instead of AP?

> BUT it HUNG in WiFi Config for many minutes and never exited.

If possible, the next time you do a WiFi config, please be connected to USB with the terminal open. What you'll see is a big blob of text when 'Save Config' is sent:

[image]

Above is all the settings going from the browser back to the unit.

Similarly, when you press Exit and Reset:

[image]

I'm interested if your browser is ever sending the specific line:

    Parsing: exitAndReset,1,

tonycanike, 2023-02-28:
I'll try to look at this; I have a few pressing non-surveying matters here on my end.

tonycanike, 2023-03-08:
I'm going to close this. I won't be able to provide the feedback requested in any reasonable timeframe.  

If this was a significant bug
other users would have seen it.

closed 2023-03-08 by tonycanike