sparkfun/SparkFun_RTK_Firmware #400: Display shows 1.04m "accuracy" and 0 satellites

tonycanike, 2023-02-12:
Dish antenna is blinking. 
Single crosshairs are solidly on. 

How can this be?  

f/w 3.1

[image]

tonycanike, 2023-02-13:
A reboot sorted it all out.   I'm not terribly worried about it, just thought I'd report it.

nseidle, 2023-02-24:
Ya, I've seen it too. It's very difficult to recreate. Thanks for reporting the Firmware version as we've been trying to mitigate it over the versions but obviously it's still lurking.

nseidle, 2023-05-05:
I *think* this was related to #455. There was a bug in the GNSS library that may have been preventing PMP, PVT, and other messages from being transferred correctly. This was fixed [here](https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/commit/c1c857b2c374027ab664870b1f11afe98968d446) in the GNSS library. This fix was not in RTK firmware version v3.2 and below.

I'm going to mark this 'fixed' because I believe this was the culprit, and haven't seen this bug occur again for a long time, but let me know if it pops up again.

nseidle, 2023-05-05:
This should be fixed in [v3.3 of the RTK firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v.3.3). Please let us know if you still see problems.

closed 2023-05-05 by nseidle