sparkfun/SparkFun_RTK_Firmware #401: Firmware writes to very old log file instead of starting a new log file.  Start new log file button in Wifi doesn't work.

tonycanike, 2023-02-12:
### Subject of the issue
Log data being appended to log log file (from Dec 2022) instead of opening a new log file.
The start new log file button in the WiFi config didn't cause a new log file to be opened.

### Your workbench
f/w 3.1

### Steps to reproduce
Configure in WiFi to log PPP
Start logging.  See the animated P logging icon.
Let it run for 10 minutes
shutdown and pull SD card
No new file.
repeat a couple times.
Pushed that "start new log file" button in the Wifi Config a couple times.  There's no feedback that it actually realized it was pressed.
Check in serial menu and see log file is getting better.
Realize the old old log file is being appended to.

### Expected behavior
I thought this was fixed many months ago.  Just start a new log file.  Why would anyone want data appended to a 2-month-old file? 

### Actual behavior

tonycanike, 2023-02-12:
I've tried the new log file button and reset to factory defaults (with and without SD card), and it just refuses to start a new log file.

amlago, 2023-02-26:
facet v3.1
I confirm what Tony says.
I have tried to create new log files by pressing the new file button and no new file is generated.
The new data is added to the old file and added correctly, so the file size is considerable.

1) I would like that when pressing the new file button a message confirming the closing of the file and the creation of a new file would appear. Show new file name on screen? may be a good idea.
2) Check what is happening with the new file button, because it seems to me that in previous versions it worked fine.

amlago, 2023-02-26:
I clarify that the only file that has been generated, because I have worked on the same base station, works correctly in the post process.
I transform it with RTKLIB and post process it with CHC Geomatic 2, without any problem.

amlago, 2023-02-26:
I just tried Facet v3.2 and the same thing happens. No new file is generated.
And in v3.2 there is no SD card icon on the screen.
Just the vertical bars in place.

amlago, 2023-02-27:
facet v3.1
With the bluetooh serial configuration I have sometimes been able to create new log files.
It's different than wifi setup where the new file button never works.
Not always, but with the bluetooth serial setup I usually get it after a complete shutdown.
Reboots don't seem to work the same way as hard shutdown.
We continue to investigate.
It seems to me that with v2.7 the new log file button worked.

amlago, 2023-02-28:
Update
1)After a complete shutdown, I can create new log files from bluetooth serial settings.
2) From the Wi-Fi configuration, the new log file button does not work.
3) Another topic appeared. Pressing the new log file button in wifi settings disappears the sd card data in the File Mananger tab.
4) If I reload the wifi config page, the File manager tab disappears.

nseidle, 2023-03-02:
We have two issues in one. This is the friendly reminder to try to report one issue at a time.

## Firmware writes to very old log file instead of starting a new log file. 

A new log file *should* be generated at every new power-on. The only time a file is re-used is when a system reset is detected. Usually this is an error or a system panic so we attempt to continue the last log for continuity reasons.

If you power off the device, and power it back on, is a new log started? You should not need to do WiFi config or anything else.

## Start new log file button in Wifi doesn't work.

The current approach is not working. We'll come up with something better.

nseidle, 2023-03-02:
For those following along at home:

[image]

First, GNSS has enough reception to set the system time. Second, the log file is created with the current time/date. A log file is not started until the date/time can be determined.

[image]

Above, in the case that the system experiences a reset (instead of the normal power on/power off) the log will be appended.

nseidle, 2023-03-02:
I believe both of this issues have been addressed. Please check latest RC (available shortly).

amlago, 2023-03-03:
I tried it a couple of times and it worked.
Pressing the new log file button generates a new file and changes the log file name on screen.
Excellent

nseidle, 2023-03-03:
👍 Let me know if there needs to be a more clear indication of a file being generated. The name changes obviously, but it may not be clear enough.

nseidle, 2023-03-08:
This should be fixed in [v3.2](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases). Please let us know if this is still a problem.

closed 2023-03-08 by nseidle

tonycanike, 2023-03-29:
I just spent 7 days in the field using the v3.3-Mar 11 2023 firmware.  It does appear that both of these issues have been resolved.  

The increase in the field usability of the RTK Firmware has increase significantly in the past few months!

Thank you very much for all the work!