sparkfun/SparkFun_RTK_Firmware #402: How to log data to SD card for RTK surveyor

okhsallam, 2023-02-14:
Dear all, I am new on using the RTK Surveyor. 

I am using it as a rover, but I am not sure how to enable logging the data on the SD card ?

Thanks in advance

tonycanike, 2023-02-14:
It's not hard, but you will need to spend some time learning generally how to use the device if you haven't already. 

Here is the manual page on logging.   

https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_data_logging/

Here's the manual page on configuring it via WiFi. 

https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_with_wifi/

nseidle, 2023-02-16:
@okhsallam - Those docs should cover the basics. If you have additional questions or problems, please feel free to open a new issue.

closed 2023-02-16 by nseidle