sparkfun/SparkFun_RTK_Firmware #405: Provide Base Location in Rover data stream and log

tonycanike, 2023-02-17:
### Subject of the issue
Describe your issue here. Additionally, screenshots are easy to paste into github.

### Your workbench
* What version of RTK firmware are you running? This can be found when the serial menu is opened (also in the settings.txt file, and in the serial output at power up).
* What radios are you using: Bluetooth, WiFi, and/or ESP-Now? What app are you using to connect over Bluetooth? Are you transmitting NTRIP back to the device?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post any log files from serial output that may have been generated.

### Expected behavior
Tell us what should happen

### Actual behavior
Tell us what happens instead

tonycanike, 2023-02-23:
I've noticed that when SurvPC 6.16 is connected to a rover using SurvPC's "Generic NMEA" driver, it has a screen to display the location of the base and other info about the base. When connected to a Facet, the base info screen displays but is blank. 

Unless there was some programming oversight, this implies that SurvPC can determine the base location from messages in a NMEA stream, or at least an extended NMEA stream from one receiver. 

In general, SurvPC is very good about disabling screens particular drivers don't support.

nseidle, 2023-03-08:
A fine idea but we've not decoded RTCM before. It's possible, will just take some time.

tonycanike, 2023-03-15:
It has occurred to me that perhaps my data collector can decode the RTCM message types regarding the base (aka the reference station) if the rover would forward them from the base to the data collector in the NMEA data stream.  Eg RTCM MTs 1006,1033, etc.  

There is some logic to this; why invent some new rover-to-data collector base-info-protocol and write code for the rover to translate RTCM to such a new protocol, when instead the rover could more simply forward a few RTCM message types. 

If I can figure out how to monitor that data stream from another GNSS receiver rover I have (which somehow does provide base info to my data collector) to my data collector perhaps I could test this theory. 

It also might be simpler for the RTK product to include selected RTCM messages in the rover log to document 

This is not yet a well-formed request but only some thoughts about how we might proceed.

nseidle, 2023-05-01:
RTCM can come in from a few sources:

* BT from a datacollector software. 
* WiFi from a caster
* Radio connection to a base

BT: In this case, the DC should already be aware of the RTCM stream as the DC is getting RTCM over NTRIP (over it's own connection cellular, WiFi, etc). I'd leave the RTCM decoding up to the DC.

WiFi: We could parse the incoming packets. This is still on my todo list.

Radio: The RTCM comes in over an unknown radio and is directly dumped into the ZED-F9P. I don't know of an immediate way to intercept these messages.

tonycanike, 2023-05-01:
Thanks Nathan, agreed.  

I am using the Radio case.  I have a theory that other GNSS receivers do this (when acting as a rover, forward RTCM 1005, etc. from the base to the data collector) and that various data collector software uses the RTCM, but I am not sure.  

I'd like to test that the data collector will actually properly use the RTCM 1005, etc. before anyone does a lot of work.  If you have any pointers on how to create a test data set with NMEA (from a rover) and RTCM (from a base) messages, and play it back to my data collector at an appropriate rate, I'll run a test before we pursue further.

tonycanike, 2023-05-02:
Hmmm what if a ZED rover was configured to send RTCM 1005 - would it just forward what it received?